merbanan/rtl_433 #1100: undefined reference to `template'

fme34, 2019-07-06:
Hi Falks,
Looking for help on compiling rtl_433 on Rapsberry Pi 3b+ running Raspbian Buster fresh install. I got the following error at make step:

/usr/bin/ld: rtl_433.o: in function `main':
/home/pi/rtl_433/src/rtl_433.c:1120: undefined reference to `template'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:931: rtl_433] Error 1
make[2]: Leaving directory '/home/pi/rtl_433/src'
make[1]: *** [Makefile:497: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/rtl_433'
make: *** [Makefile:383: all] Error 2

Any Idea ?
Regards

zuckschwerdt, 2019-07-06:
Yes, sorry, the last commit was incomplete. Please `git reset --hard HEAD^ ; git pull`

closed 2019-07-06 by zuckschwerdt

fme34, 2019-07-06:
Worked, thanks !!