merbanan/rtl_433 #1101: Latest WS4945 Not Decoding

lesreaper, 2019-07-08:
I have some old WS4945's here that decode using RTL_433 under `-R 23`. I believe they're a little bit older models since they were here in the house when I moved in.

However, I ordered brand new one, and RTL_433 isn't decoding the signal properly. Do I need to adjust something? I tried the `-X` decoder it recommended, but that didn't help. I also looked here but didn't understand the conversation: #645 

Here's what I'm getting from the output of this command:
```
rtl_433 -F "mqtt://localhost:1883,user=username,pass=password,retain=0" -a -A -R 23 -X 'n=name,m=OOK_PWM,s=264,l=2540,r=2252,g=744,t=908,y=0' -F json -M utc | mosquitto_pub -t rtl_433 -l -u username  -P password
```
I know MQTT is working because the other WS4945 sensors get picked up and open or closed status sent in the package. Here's the output:
```
Detected OOK package	2019-07-08 22:04:43
Analyzing pulses...
Total count:   26,  width: 26.06 ms		( 6514 S)
Pulse width distribution:
 [ 0] count:    1,  width: 2532 us [2532;2532]	( 633 S)
 [ 1] count:   25,  width:  264 us [260;276]	(  66 S)
Gap width distribution:
 [ 0] count:   13,  width:  232 us [224;244]	(  58 S)
 [ 1] count:    7,  width:  736 us [732;744]	( 184 S)
 [ 2] count:    2,  width: 2484 us [2236;2732]	( 621 S)
 [ 3] count:    3,  width: 1240 us [1236;1244]	( 310 S)
Pulse period distribution:
 [ 0] count:    3,  width: 2756 us [2504;3000]	( 689 S)
 [ 1] count:   12,  width:  500 us [496;508]	( 125 S)
 [ 2] count:    7,  width: 1000 us [996;1008]	( 250 S)
 [ 3] count:    3,  width: 1504 us [1500;1508]	( 376 S)
Level estimates [high, low]:  15996,     29
RSSI: -0.1 dB SNR: 27.3 dB Noise: -27.4 dB
Frequency offsets [F1, F2]:   17872,      0	(+68.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_width: 264, long_width: 2532, reset_limit: 2736, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=264,l=2532,r=2736,g=748,t=904,y=0'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 6 
[00] { 7} 7e        : 0111111
[01] { 1} 80        : 1
[02] { 1} 80        : 1
[03] { 3} e0        : 111
[04] { 8} ff        : 11111111 
[05] { 6} fc        : 111111

*** signal_start = 9108926, signal_end = 9135435, signal_len = 26509, pulses_found = 26
Iteration 1. t: 347    min: 67 (25)    max: 627 (1)    delta 1
Iteration 2. t: 347    min: 67 (25)    max: 627 (1)    delta 0
Pulse coding: Short pulse length 67 - Long pulse length 627

Short distance: 101, long distance: 308, packet distance: 622

p_limit: 347
bitbuffer:: Number of rows: 3 
[00] { 7} 80        : 1000000
[01] { 2} 00        : 00
[02] {17} 00 00 00  : 00000000 00000000 0
*** signal_start = 9178856, signal_end = 9246932, signal_len = 68076, pulses_found = 52
Iteration 1. t: 349    min: 67 (50)    max: 632 (2)    delta 5
Iteration 2. t: 349    min: 67 (50)    max: 632 (2)    delta 0
Pulse coding: Short pulse length 67 - Long pulse length 632

Short distance: 57, long distance: 287, packet distance: 35051

p_limit: 349
bitbuffer:: Number of rows: 2 
[00] {26} 80 00 00 00 : 10000000 00000000 00000000 00
[01] {26} 80 00 00 00 : 10000000 00000000 00000000 00
Detected OOK package	2019-07-08 22:04:44
Analyzing pulses...
Total count:    2,  width: 1.10 ms		(  274 S)
Pulse width distribution:
 [ 0] count:    1,  width:  840 us [840;840]	( 210 S)
 [ 1] count:    1,  width:   40 us [40;40]	(  10 S)
Gap width distribution:
 [ 0] count:    1,  width:  216 us [216;216]	(  54 S)
Pulse period distribution:
 [ 0] count:    1,  width: 1056 us [1056;1056]	( 264 S)
Level estimates [high, low]:   1000,     28
RSSI: -12.1 dB SNR: 15.4 dB Noise: -27.5 dB
Frequency offsets [F1, F2]:     797,      0	(+3.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed gap
Attempting demodulation... short_width: 40, long_width: 840, reset_limit: 220, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=40,l=840,r=220,g=0,t=320,y=0'
pulse_demod_pwm(): Analyzer Device 
bitbuffer:: Number of rows: 1 
[00] { 2} 40        : 01
```

zuckschwerdt, 2019-07-09:
Use one of (not both in one command)
- `rtl_433 -R 23 -F "mqtt://localhost:1883,user=username,pass=password,retain=0"`
- `rtl_433 -R 23 -M utc -F json | mosquitto_pub -t rtl_433 -l -u username  -P password`

Use this for testing:
`rtl_433 -a -A -R 0 -X 'n=DSCtest,m=OOK_PCM,s=250,l=500,r=5000' -F json`

If this doesn't work record a sample (`rtl_433 -S all`) and upload here.

lesreaper, 2019-07-10:
So, interestingly enough, one of the 99 loaded decoders using `rtl_433 -S all` worked, just not the official DSC one. You can see the image here: https://i.imgur.com/EjDl3Vq.png

However, using the `-G`, which you would think would load all the decoders like `-S all`, did nothing for the proper decoding and sending of data. I tested it with the working Door Sensors and the new one simultaneously.

The other two commands, and the testing command, did nothing other then recognizing but not decoding the signal.

I attached the zip from the command `-S all` command: 

[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/3375416/g001_433.92M_250k.zip)

zuckschwerdt, 2019-07-10:
The decoding in the picture you posted is the DSC decoder, protocol 23.

Note that `-S` won't add additional decoders, just dumps the received signal to file.
You don't need `-G`, it will only add the remaining blacklisted decoders. You should be fine with the default set, or even just `-R 23`.

The sample you attached decodes fine, doesn't this work?
- `rtl_433 g001_433.92M_250k.cu8`

You can even refine that to
- `rtl_433 -R 23 -M newmodel -M protocol -M level g001_433.92M_250k.cu8`

to get the new model names and more details:
```
Reading samples from file: g001_433.92M_250k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244928s   Protocol  : 23
model     : DSC-Security id        : 2182508
closed    : 1            event     : 1             tamper    : 1
battery_ok: 1            xactivity : 1             xtamper1  : 1
xtamper2  : 0            exception : 0             esn       : 214d6c
status    : 162          status_hex: a2            mic       : CRC
Modulation: ASK          Freq      : 0.0 MHz
RSSI      : -0.1 dB      SNR       : 28.8 dB       Noise     : -28.9 dB
```

lesreaper, 2019-07-10:
So, these do give me the readouts:
```
rtl_433 g001_433.92M_250k.cu8
```
and
```
rtl_433 -R 23 -M newmodel -M protocol -M level g001_433.92M_250k.cu8
```

However, when I run this: 
```rtl_433 -R 23 -M utc -F json | mosquitto_pub -t rtl_433 -l -u username -P password```

It doesn't recognize this brand new device. It only recognizes the older WS4945's in the house. 

Should I create a new model or something with these new settings from the `.cu8` file?

lesreaper, 2019-07-10:
I'm also open to anybody who might know a door contact sensor that is 100% compatible with this library.

zuckschwerdt, 2019-07-10:
I tested
- `rtl_433 -R 23 -M utc -F json g001_433.92M_250k.cu8 | mosquitto_pub -t rtl_433 -l`
- `rtl_433 -R 23 -M utc -F mqtt ~/Desktop/g001_433.92M_250k.cu8`

both work fine. To confirm run
- `mosquitto_sub -t '#' -v`

lesreaper, 2019-07-11:
I'm not saying those don't work fine, I know they do when loading the file. Definitely, and they look nice on the RTL_433 side.

What's happening is that when I run RTL normally, with the ```rtl_433 -R 23 -M utc -F json | mosquitto_pub -t rtl_433 -l -u username -P password``` it picks up all the other WS4945's in the house AND I get their device ID's, whether they're closed or not, etc in a JSON object like this:

```
{'time': '2019-07-11 21:26:21', 'model': 'DSC Contact', 'id': 1111111, 'closed': 1, 'event': 1, 'tamper': 0, 'battery_low': 0, 'xactivity': 1, 'xtamper1': 0, 'xtamper2': 0, 'exception': 0, 'esn': '209ef2', 'status': 163, 'status_hex': 'a3', 'mic': 'CRC'}
```

I don't get the JSON object on the new WS4945 on the client side. Wherever that difference may be, is where I'm looking to modify. Since the other devices work, I have to assume there's something server side on this particular device.

zuckschwerdt, 2019-07-12:
You should run just `rtl_433 -R 23 -M utc -F json` and watch the output carefully. Make sure it shows both your older and newer sensors. Problem is either acquiring the data or getting the data through MQTT to the consumer. I can tell which right now?

rct, 2019-07-12:
Is the new sensor actually from DSC?

Given their installed base, it seems pretty unlikely that they've changed something about the protocol in a recent batch.  Not impossible, but seems unlikely.

Looking at this now, I'm having a hard time following what's been going on. Do you have a recording where you've managed to capture the new sensor?   When you read that file back it, does the decoded ESN match the new sensor?

Also, I completely agree with Christian, just try to isolate the problem with only rtl_433, without MQTT etc.   Figure out whether your SDR and rtl_433 are hearing the new device and if there is failure to decode it, why.    Of course, turning on debugging might help.

lesreaper, 2019-07-14:
It's definitely a DSC, the only difference is the new ones don't have a `Z83` number on them. Go tit brand new from Amazon.

I recorded the sensor going off using the `-S all`. Was able to recognize and decode everything, and saved the file.

However, when running it live with `-a  -A`, not not with the `-S all`, it hears the signal, but doesn't decode it. All of my other DSC sensors work great with the system.

By debugging, do you mean using more verbosity? Something else?

rct, 2019-07-15:
Why are you running with the analyzers ( `-a` or `-A` ) enabled when you just want to see if it decodes?

My understanding could be out of date as so much has changed, but the  `-a` analyzer used to disable the decoders.  `-R 0` also disables the built in decoders.

Yes, by debugging, I mean enable verbose output so you can see if the DSC decoder is getting called when you trip the sensor.   Set the level to 2,  You might want to give the source a quick skim after you see the verbose output to see how far it is getting.

I'm not sure what you mean by the `Z83` number, can you explain?

lesreaper, 2019-07-21:
It tries to decode, but fails. It even identifies it as a `DSC Decoder` using the `-vv` verbosity, but can't send the data anywhere.

I'm going to buy an older DSC sensor and start over. I'm not sure why the new ones aren't working, but I'd be happy to help create a new decoder for it if someone knows how.

closed 2019-08-01 by lesreaper