merbanan/rtl_433 #1104: CurrentCost EnviR Help

neilco, 2019-07-12:
I've started looking into adding support for the CurrentCost EnviR transmitter (as per #247), but I could do with some guidance. 

I've captured transmissions with the CT clamp sensor in each of the three sensor ports on the bottom on the transmitter. The signals seems to be quite clean, as you can see below:

[image]

Next, I eyeballed the first 32 bits of the reading from Sensor 1:

[image]

Unfortunately, this were I've hit a roadblock. According to [this site](https://github.com/JackKelly/rfm_edf_ecomanager/wiki/Technical-details-of-Current-Cost-RF-protocol) linked from #247, I should expect to see either:

[image]

or

[image]

However, that's not what I'm receiving. You can also see that the first 45 bits are the same for all three sensors. This got me thinking it could be [the init pattern](https://github.com/merbanan/rtl_433/blob/8fdaf27d6562b4c01682fceec5e49a6c256e990c/src/devices/current_cost.c#L10) but the first 24 bits of that are:

[image]

So that doesn't fit either. 

What am I doing wrong?

zuckschwerdt, 2019-07-12:
That looks like Manchester code. If you can get IQ sample data then use rtl_433 to decode the stream using the flex decoder. First try PCM to get the data you have above. Then try the MC or decode further by hand: `01` is a 0-bit, `10` is a 1-bit.

neilco, 2019-07-12:
```shell
$ rtl_433 -X 'm=FSK_PCM'
rtl_433 version 18.12-235-g8bae8c8 branch master at 201907080940 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/neil/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Bad flex spec, missing modulation!
Use -X <spec> to add a general purpose decoder. For usage use -X help
```

> Bad flex spec, missing modulation!

But…. but… isn’t `m` modulation?!

neilco, 2019-07-12:
Also, if `01` is a 0-bit and `10` is a 1-bit, doesn’t that mean the data I’ve captured is junk because of that string of nine `1`s?

zuckschwerdt, 2019-07-13:
For flex, the names always goes first (compat to the old positional format).
Ah ok, you already did a Manchester decode, sorry missed that. But 0xCC is 0x55 after MC decode.

neilco, 2019-07-13:
> compat to the old positional format

You assume I know what that means when I don’t. 

> But 0xCC is 0x55 after MC decode.

So either I’m doing something wrong or the data is junk. There is no 0x55, 0xAA, or 0xCC in the signals I’ve captured above. What am I doing wrong?

zuckschwerdt, 2019-07-13:
> You assume I know what that means when I don’t.

The original flex spec format was positional, thus the name field always needs to be first. Simply, always start a flex spec with `n=myname,m=…`

> What am I doing wrong?

Your decoding looks right. Could be that the data is a different protocol (Though using `000…` as a preamble is a poor choice for a protocol as the MC `101010…` is ambiguous). Try to acquire more data, put it in a [BitBench](http://triq.net/bitbench?c=y0000000000000000110100011111111101&f=8h%20) and then play with shift and invert to see if some pattern emerges.

neilco, 2019-07-18:
Yes! 

[image]

zuckschwerdt, 2019-08-26:
Support now has been added, thanks!

closed 2019-08-26 by zuckschwerdt