merbanan/rtl_433 #1106: order of parameters matters for at least -d

dankamongmen, 2019-07-14:
Using

`./rtl_433 -f 303.866M -v -d0 -a -M newmodel -X 'n=Minka-Aire,m=OOK_MC_ZEROBIT,s=360,l=0,r=660,repeats>=3' -A -M hires -M level -ddriver=hackrf`

I'm able to access my HackRF via Soapy, but if the `-d` is moved to the front of the argument list, I instead get my RTL-SDRv3, no matter what I supply (including `""`, `driver=hackrf`, and `21` (the USB device number).

This was maddening =]. I'm gonna drop you a PR to fix it up, if that's cool?

dankamongmen, 2019-07-14:
```
[schwarzgerat](0) $ ) $ ./rtl_433 -d 21 -f 303.866M -d0 -a -M newmodel -X 'n=Minka-Aire,m=OOK_MC_ZEROBIT,s=360,l=0,r=660,repeats>=3' -A -M hires -M level
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dank/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 100 out of 126 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-126 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 303.866MHz.
Allocating 15 zero-copy buffers
^CSignal caught, exiting!
[schwarzgerat](0) $ 
```

dankamongmen, 2019-07-14:
```
[schwarzgerat](0) $ ./rtl_433 -f 303.866M -d0 -a -M newmodel -X 'n=Minka-Aire,m=OOK_MC_ZEROBIT,s=360,l=0,r=660,repeats>=3' -A -M hires -M level -ddriver=hackrf
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/dank/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 100 out of 126 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-126 ]
[INFO] Opening HackRF One #0 87c867dc2c4b2b5f...
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 303.866MHz.
^CSignal caught, exiting!
[schwarzgerat](0) $ 
```

Note the different output depending on the location of `-d`.

zuckschwerdt, 2019-07-14:
Parameters are mostly positional, yes. But `-d` isn't. It will only take the last value (why are you using multiple `-d`?) and then open that device after all arguments are parsed. Look at the last `-d` given in each invocation, that's what's opened.

dankamongmen, 2019-07-14:
Oh man, I *am* providing two -d's. I'm stupid. Apologies. :/

closed 2019-07-14 by dankamongmen

zuckschwerdt, 2019-07-14:
Happens. No harm in asking. Have fun!

dankamongmen, 2019-07-14:
Well thanks for the PEBKAC fix. I'm gonna be tossing you a decoder for Minka-Aire remotes later today. Ought I create a bug for them, or just open a PR?

zuckschwerdt, 2019-07-14:
Sounds great! Just the PR.