merbanan/rtl_433 #1107: Add support for Oregon scientific THGR328N and RTGR328N

dgrostoto, 2019-07-14:
I wrote some code to add support for the two sensors.
THGR328N is a 5 channel temperature plus hygro.
RTGR328N is somewhat very interesting since it is one of the rare that reports RF clock DCF77 (for europe it is emitted from Francfort and coverage is thousands of kilometers, i guess there is other points of emission around the world), it report temperature and hygro too. Operation of this sensor is a little uncommon since it use 7 identifiers (5 for temperature and hygro and 2 for RF clock)
RF clock reference : https://fr.wikipedia.org/wiki/DCF77
 I give as attached the file i've modified (src/devices/oregon_scientific.c), i also made some minor modification concerning CM180 support (math precision mainly) and added debug information i needed to create support for this support (verbose mode of decoder).
Every seems to work well at home with half of dozen different sensor types.
Hope you can integrate this modification as soon as you can that i don't have to report modifications on my side each time i clone the git... :)

Base file oregon_scientific.c 
[oregon_scientific_lastest_withmymodif.c.txt](https://github.com/merbanan/rtl_433/files/3389867/oregon_scientific_lastest_withmymodif.c.txt)
used for modification was cloned from git 2019 july 13 (yesterday).

dgrostoto, 2019-07-14:
feel free to ask any questions if you have

zuckschwerdt, 2019-07-14:
Thanks, looks good, I'll clean this up (should mask out some bits instead of having 8 IDs…) and integrate sometime next week probably.

A branch with these changes is now here: https://github.com/zuckschwerdt/rtl_433/tree/feat-dgrostoto
Diff at https://github.com/zuckschwerdt/rtl_433/commit/5312dee

dgrostoto, 2019-07-14:
thanks in advance for your merge work. You're right concerning IDs .

zuckschwerdt, 2019-07-17:
This is merged now (without the debugging code). Reopen if something is missing.

closed 2019-07-17 by zuckschwerdt