merbanan/rtl_433 #1108: Time missing in MQTT output

gdrapp, 2019-07-14:
The new MQTT output is great but I've noticed that time data is explicitly removed.  I use the time to tell when the state of the device was last updated.  Would you consider adding the time back in?  I'd be happy to submit a pull request.  Thanks!

https://github.com/merbanan/rtl_433/blob/8fdaf27d6562b4c01682fceec5e49a6c256e990c/src/output_mqtt.c#L355

zuckschwerdt, 2019-07-14:
Do you use MQTT to retain ("store") data? Probably best to feed the messages to an TSDB and run queries there. Could be useful to quickly "pull" the last state from MQTT, but it's best used to transfer events, not store state?

gdrapp, 2019-07-15:
I use rtl_433 with MQTT to make data available to openHAB for home automation.  I use retain in MQTT so that my openHAB instance has sensor data immediately after a restart instead of having to wait for a sensor to report in again.  With time included I know immediately how stale the data is.  Is there harm including time?  Seems like those that don't need it can ignore it and it would be there for those that do.

merbanan, 2019-07-16:
Time should be kept. While I myself don't use it the option should be there for those who want it.

zuckschwerdt, 2019-07-17:
Ok, thanks. I had the assumption that subscribers would timestamp themselves. I would still advice to use a TSDB or logging and not rely on MQTT retaining for and extended period.

Filter on `time` is removed with 6d76581. Note that you can adjust the time format with `-M`, you'll likely want `-M time:iso:usec`.

closed 2019-07-17 by zuckschwerdt

gdrapp, 2019-07-17:
Thanks!  And I would agree about using a TSDB.  I use influxdb for trending and long term storage.