merbanan/rtl_433 #1112: Add support for Generic 21 key rf Remote

chaos511, 2019-07-16:
can you add support for a 21 key remote like [this](https://www.ebay.com/itm/DC-5-24V-21-Key-RF-Remote-Controller-for-WS2812b-WS2811-Pixel-LED-Strip-Light/183753705835?hash=item2ac893096b:g:01AAAOSw4ixchsYp) it uses OOK
byte 0,1,3 are always the same header/remote id?
byte 2 is the button being pressed like this
fe  fd  fc
fb  fa  f9
f8  f7  f6
f5  f4  f3
f2  f1  f0
ef  ee  ed
ec  eb  ea

the command and the output
`rtl_433  -A -R 0 -g 10

Detected OOK package    2019-07-16 17:22:48
Analyzing pulses...
Total count:   25,  width: 39.71 ms             ( 9927 S)
Pulse width distribution:
 [ 0] count:   19,  width:  264 us [256;296]    (  66 S)
 [ 1] count:    6,  width: 1060 us [1056;1064]  ( 265 S)
Gap width distribution:
 [ 0] count:   18,  width: 1372 us [1352;1380]  ( 343 S)
 [ 1] count:    6,  width:  596 us [588;600]    ( 149 S)
Pulse period distribution:
 [ 0] count:   24,  width: 1640 us [1624;1660]  ( 410 S)
Level estimates [high, low]:   7042,     12
RSSI: -3.7 dB SNR: 27.3 dB Noise: -31.0 dB
Frequency offsets [F1, F2]:    6583,      0     (+25.1 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with fixed period
Attempting demodulation... short_width: 264, long_width: 1060, reset_limit: 1384, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=264,l=1060,r=1384,g=0,t=316,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {25} ff ca f6 80 : 11111111 11001010 11110110 1`

zuckschwerdt, 2019-07-17:
I have a few of those too. They can be added with flex decoders. Have a look at the `conf/` folder for examples.

chaos511, 2019-07-17:
thanks i got it to work with a flex decoder

zuckschwerdt, 2019-07-17:
If you like post the working conf here (or PR) and we'll include it as example.

chaos511, 2019-07-17:
This is the config i am using i do not know if id is correct because i only have one remote 

`decoder {
        name=Generic-21-key-Remote,
        m=OOK_PWM,
        s=264,
        l=1060,
        r=5084,
        g=0,
        t=316,
        y=0,
        bits=25,
        invert,
        get=@8:{8}:id,
        get=@16:{8}:button
}`

merbanan, 2020-01-07:
config file added.

closed 2020-01-07 by merbanan