merbanan/rtl_433 #1113: conf file should not override command line args

ZeroChaos-, 2019-07-18:
The example conf file includes some great stuff, like setting device and freq.  That said, if the command line args conflict with the config file, the command line should win.

zuckschwerdt, 2019-07-22:
Command line args are read left to right. Be sure to use args to alter conf settings *after* you read the conf file.

closed 2019-07-22 by zuckschwerdt

ZeroChaos-, 2019-07-22:
Interesting theory, but it doesn't seem to require a command line arg to read the config file. it does it by default.

zuckschwerdt, 2019-07-22:
The default conf file is the very first thing parsed, command line options then later override that.

ZeroChaos-, 2019-07-22:
no sir, no they don't

```
theprophet rtl_433 # rtl_433 -f 900M
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Reading conf from "rtl_433.conf".
Protocol [31] "TFA-Twin-Plus-30.3049, Conrad KW9010, Ea2 BL999" does not take arguments ", Conrad KW9010, Ea2 BL999"!
Protocol [33] "WT450, WT260H, WT405H" does not take arguments ", WT260H, WT405H"!
Protocol [40] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning" does not take arguments ", 5n1 Weather Station, 6045 Lightning"!
Protocol [42] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor" does not take arguments ", Humidity, Wind and Rain Sensor"!
Protocol [74] "Acurite 00275rm,00276rm Temp/Humidity with optional probe" does not take arguments ",00276rm Temp/Humidity with optional probe"!
Protocol [75] "LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor" does not take arguments ", TFA Dostmann 30.3155 Temperature/Humidity sensor"!
Protocol [78] "Fine Offset Electronics, WH25, WH32B, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor" does not take arguments ", WH25, WH24, WH65B, HP1000 Temperature/Humidity/Pressure Sensor"!
Protocol [79] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor" does not take arguments ", WH0530 Temperature/Rain Sensor"!
Registered 104 out of 133 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 130-133 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
```

zuckschwerdt, 2019-07-23:
Multiple `-f` do not override but add for frequency hopping. Try (without config file) `rtl_433 -f 433.92M -f 868M`. Sorry, that's not fixable. But you can always skip the config: `rtl_433 -c null -f 900M`.