merbanan/rtl_433 #1114: WS4945 showing "DSC row 0 invalid bit count 11"

garyvdm, 2019-07-20:
Thank you for the awesome work you have done on this project

I'm trying to pickup a bunch of DCS WS4945 transmitters that were installed at a house I bought.

[image]

Here is a recording from `rtl_433 -S all`. 
[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/3413680/g001_433.92M_250k.zip)

When I run `rtl_433 -vvv -R 23 g001_433.92M_250k.cu8`, I see the following line a few times in the output: 

    DSC row 0 invalid bit count 11

I do not get any decoded signals.

garyvdm, 2019-07-21:
From some info from #645 , I was able to get it to working with this patch:

```
diff --git a/src/devices/dsc.c b/src/devices/dsc.c
index 05bb383..8a08b00 100644
--- a/src/devices/dsc.c
+++ b/src/devices/dsc.c
@@ -220,9 +220,9 @@ static char *output_fields[] = {
 r_device DSC = {
     .name          = "DSC Security Contact",
     .modulation    = OOK_PULSE_PCM_RZ,
-    .short_width   = 250,    // Pulse length, 250 µs
-    .long_width    = 500,    // Bit period, 500 µs
-    .reset_limit   = 5000, // Max gap,
+    .short_width   = 536,    // Pulse length, 536 µs
+    .long_width    = 1072,    // Bit period, 1072 µs
+    .reset_limit   = 6000, // Max gap,
     .decode_fn     = &dsc_callback,
     .disabled      = 0,
     .fields        = output_fields,
```

merbanan, 2019-07-22:
Send a PR for one more r_device with the correct timings. Wrap the callback so that different model names can be used. This device can use "DCS WS4945" as model names.

garyvdm, 2019-07-22:
Ok. I'll give it a try.

dxdc, 2019-11-01:
@garyvdm @merbanan I submitted a PR for this. I'm having some other issues with 2 DSC contacts (others are working fine) so need to do some more troubleshooting. May be unrelated.

merbanan, 2020-01-30:
Fix committed. Regarding the 2 contacts the signal quality is lacking. Maybe try offsetting the frequency a little bit. It was pretty close to dc.

closed 2020-01-30 by merbanan