merbanan/rtl_433 #1117: Unhandled sensor reading akhan / X10

alex1313c, 2019-07-30:
Hello
Sorry for my english translated
I have two problems i use Realtek RTL2832 based DVB dongle to receive 433Mhz I use the latest version RTL, and domoticz version: 4.10602
the first is a remote control key 3 buttons ("beeper X6 R-TX2") which is very well
recognized by the command rtl__433 -G ( or protocol 48)

time      : 2019-04-09 20:27:00
model     : Akhan 100F14 remote keyless entry      ID (20bit): 0x6e890
Data (4bit): 0x4 (Mute)
_ _ _ _ _ _ _ _ _ _ _ _

time      : 2019-04-09 20:27:07
model     : Akhan 100F14 remote keyless entry      ID (20bit): 0x6e890
Data (4bit): 0x1 (Lock)
_ _ _ _ _ _ _ _ _ _ _ _

time      : 2019-04-09 20:27:08
model     : Akhan 100F14 remote keyless entry      ID (20bit): 0x6e890
Data (4bit): 0x2 (Unlock)
_ _ _ _ _ _ _ _ _ _ _ _

and on domoticz:

2019-04-09 20:28:57.341 Status: Rtl433: Unhandled sensor reading, please report: (time,msg,codes,model,id,channel,state,unit,group,battery,
temperature_C,humidity,brand,sid,data,sensor_id,battery_low,battery_ok,sequence_num,message_type,wind_speed_mph,wind_speed_kph,wind_avg_mi_h,
wind_avg_km_h,wind_dir_deg,rain_inch,rain_in,temperature_F
2019-04-09 20:28:57.341 |2019-04-09 20:28:56,,,Akhan 100F14 remote keyless entry,452752,,,,,,,,,,0x4 (Mute),,,,,,,,,,,,,
2019-04-09 20:28:57.341 ) 

-------------------

2019-04-09 20:29:05.254 Status: Rtl433: Unhandled sensor reading, please report: (time,msg,codes,model,id,channel,state,unit,group,battery,
temperature_C,humidity,brand,sid,data,sensor_id,battery_low,battery_ok,sequence_num,message_type,wind_speed_mph,wind_speed_kph,wind_avg_mi_h,
wind_avg_km_h,wind_dir_deg,rain_inch,rain_in,temperature_F
2019-04-09 20:29:05.254 |2019-04-09 20:29:05,,,Akhan 100F14 remote keyless entry,452752,,,,,,,,,,0x1 (Lock),,,,,,,,,,,,,
2019-04-09 20:29:05.254 ) 

-------------------

2019-04-09 20:29:12.066 Status: Rtl433: Unhandled sensor reading, please report: (time,msg,codes,model,id,channel,state,unit,group,battery,
temperature_C,humidity,brand,sid,data,sensor_id,battery_low,battery_ok,sequence_num,message_type,wind_speed_mph,wind_speed_kph,wind_avg_mi_h,
wind_avg_km_h,wind_dir_deg,rain_inch,rain_in,temperature_F
2019-04-09 20:29:12.066 |2019-04-09 20:29:11,,,Akhan 100F14 remote keyless entry,452752,,,,,,,,,,0x2 (Unlock),,,,,,,,,,,,,
2019-04-09 20:29:12.066 ) 

-------------------

My second problem and protocol x10 (motion detector X10 MS13E and wall remote X10 SS13) ( Protocol 22 )

time      : 2019-04-09 20:33:40
model     : X10-RF       houseid   : B             deviceid  : 2
state     : ON           data      : 708f10ef
_ _ _ _ _ _ _ _ _ _ _ _

log domoticz

2019-04-09 20:34:27.044 Status: Rtl433: Unhandled sensor reading, please report: (time,msg,codes,model,id,channel,battery,temperature_C,humidity,brand,
sid,data,sensor_id,battery_low,battery_ok,sequence_num,message_type,wind_speed_mph,wind_speed_kph,wind_avg_mi_h,wind_avg_km_h,wind_dir_deg,rain_inch,rain_in,
temperature_F
2019-04-09 20:34:27.044 |2019-04-09 20:34:26,,,X10-RF,,,,,,,,1888424175,,,,,,,,,,,,,
2019-04-09 20:34:27.044 ) 

on the other hand if I activate the protocol 96 it works except that all my equipment X10 controls the Lightning 2 (DMX10 RF)

2019-04-09 20:35:32.234 (RTL433) Lighting 2 (DMX10RF)

I also have 2 remote controlled KR18 and KR21 that are not visible with the command rtl__433 -G, but that's another problem
Do you have a solution to save it's equipment thank you in advance

zuckschwerdt, 2019-07-30:
`Unhandled sensor reading, please report (… to Domoticz)` ;)

closed 2019-07-30 by zuckschwerdt

alex1313c, 2019-07-30:
It's already done but no answers :(
https://github.com/domoticz/domoticz/issues/3144
thanks anyway

zuckschwerdt, 2019-07-30:
A change in Domoticz is needed. We really can't help with that, sorry.