merbanan/rtl_433 #1125: Strange WG-PB12V1 package always following WH2 Temperature/Humidity sensor data

petterreinholdtsen, 2019-08-16:
Hi.  I just tested rtl_433 in Debian, and ran into something strange.  I see several entries like this:

```
time      : 2019-08-16 02:37:08
model     : Fine Offset Electronics, WH2 Temperature/Humidity sensor         ID        : 148
Temperature: 19.3 C      Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:08
model     : WG-PB12V1    ID        : 1
Temperature: 196.8 C     Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:12
model     : Fine Offset Electronics, WH2 Temperature/Humidity sensor         ID        : 237
Temperature: 29.4 C      Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:12
model     : WG-PB12V1    ID        : 6
Temperature: 339.3 C     Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:22
model     : Fine Offset Electronics, WH2 Temperature/Humidity sensor         ID        : 197
Temperature: 19.3 C      Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:22
model     : WG-PB12V1    ID        : 1
Temperature: 275.2 C     Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:22
model     : Fine Offset Electronics, WH2 Temperature/Humidity sensor         ID        : 245
Temperature: 19.1 C      Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-16 02:37:22
model     : WG-PB12V1    ID        : 31
Temperature: 352.0 C     Integrity : CRC
```

Notice how a WG-PB12V1 measurement always follow a Fine Offset measurement, and how the temperature is amazingly high.  Could it be that the real radio message is the size of both these messages combined and interpreted incorrectly?  Perhaps the second part is humidity?

I am not sure how to debug this, but thought it best to report the issue for you to have a look.

zuckschwerdt, 2019-08-16:
Yes, false positives is always an issue we need testers  and reports to figure out :)
Here both protocol have the same length (6 bytes) and the same checksum (CRC-8 0x31). We would need sample data (`rtl_433 -S all`) to figure out if there is something to discern both. Verify with `rtl_433 FILENAME.cu8` that you have a sample that triggers this and then post a zip here.

petterreinholdtsen, 2019-08-16:
Attached file with two such strange entries.
[g001_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/3508743/g001_433.92M_250k.cu8.zip)

petterreinholdtsen, 2019-08-19:
Is the attached file sufficient to debug this, or is more data needed?

zuckschwerdt, 2019-08-19:
The recording is very nice (see yourself in http://triq.net/iqs ). Will take a few days to get around to analyze this.

zuckschwerdt, 2019-08-20:
This is fixed with f7c259a now. Somebody with a WG-PB12V1 please reopen if your device isn't decoded properly, i.e. if there are other message type values besides 0x3.

closed 2019-08-20 by zuckschwerdt

petterreinholdtsen, 2019-08-20:
[Christian W. Zuckschwerdt]
> This is fixed with f7c259a now. Somebody with a WG-PB12V1 please
> reopen if your device isn't decoded properly, i.e. if there are other
> message type values besides 0x3.

Thank you very much.  I'll test it when I am near the problematic
sensors.  Did you figure out what the information in the "trailing
package" was?

-- 
Happy hacking
Petter Reinholdtsen

zuckschwerdt, 2019-08-20:
It's just the 6 byte WH2 package, which is nearly identical to a 6 byte WG-PB12V1 package. Both could read a humidity value, but it is set to "invalid" (0xff).

petterreinholdtsen, 2019-08-21:
I tested the git version, and the "extra" package seem to be silently ignored now, instead of printed as a WH2 package.