merbanan/rtl_433 #1126: New Bresser 5 in 1 sensor?

mrbgfromv, 2019-08-16:
So I bought myself a [Bresser 5 in 1 weather station](https://www.bresser.de/Wetter-Zeit/Wettercenter/BRESSER-XXL-Wetter-Center-JC-mit-5-in-1-Aussensensor.html) and thought I could use rtl_433 to decode the data, because the sensor looks exactly like the on [in this thread](https://github.com/merbanan/rtl_433/issues/719), but I got nothing:

`rtl_433 -f 868300000` -> no results.

So I captured sound snippets and decoded them via

`rtl_433 -f 868300000 -r g*.cu8 -X n=bresser,m=FSK_PCM,s=120,l=120,r=2500`

which after some filtering to codes like this:

`{223}eaaaaaaaaaa2dd4101a188000c018ffffff3158148292fff0f200000`

Things I noticed:

- Messages are shorter than expected

mrbgfromv, 2019-08-16:
Meanwhile: 
Checked the model number at the sensor and it is: 7002590 (and is different to 7002510 which was reversed engineered by [andreafabrizi](https://github.com/andreafabrizi/BresserWeatherCenter) )

mrbgfromv, 2019-08-18:
For reporting and archiving purposes:
Spent the whole Saturday on sampling, hacking with c and decoding messages.
In the end I couldn't find other values than humidity and temperature. 
I think the messages are wrong and I am chasing ghosts: Maybe the sdr hardware is of poor quality and the sampling got wrong, then some wired decoding errors happened and finding patterns would be futile.
At this time I will zip the whole project, put a big stone on it an archive it on a safe place on my PC.
If in the future anyone ever reopens this issue I am happily going to contribute my findings ...

closed 2019-08-18 by mrbgfromv

zuckschwerdt, 2019-08-19:
Interesting story. Vendors and shops often talk about the "weather station" and just list the part numbers of the head units. rtl_433 can only listen to the sensors and it's guess work which sensors fit with some station.
Not sure if 7002590 and 7002510 are much different. Could be something simple like a color option. If you can upload a sample (zip) here we can take a look and compare the signal.

mrbgfromv, 2019-08-19:
Just to be clear: I'm blaming no one here.
You asked for some samples: [Download them here.](https://github.com/mrbgfromv/rtl_433)
They are not filtered. I think 50% of them lead to false values in my example code, which I added there, too.

zuckschwerdt, 2019-08-20:
No worries, it's always interesting to see if a protocol can be generalized to include different revisions.
It looks like the modulation, preamble and sync word is the same. I'll have a look how the payload differs.

zuckschwerdt, 2019-08-20:
Can you put the fields you indentified into a BitBench? Something like [this for the original 5n1](http://triq.net/bitbench?c=%7B208%7Dea897fe12fdfef7fff6dfbfeff1576801ed02010800092040100%20%5B7002510%5D&c=%7B208%7D8355188000c018ffddff1028184291fff0a20000000000000000%20%5B7002590%5D&f=CHK%3A104h%20U%3A8h%20ID%3A8d%20%3F%3A4h%20GUST%3A4h4h4h%20DIR%3A4h%20WIND%3A4h4h4h%20%3F%3A8h%20TEMP%3A%3C16h%20HUM%3A4h4h%20RAIN%3A%3C16h%20TS%3A4x4h).
(Just add a second format line, that way we can easily spot similarities.)

mrbgfromv, 2019-08-20:
I tried, but failed.
This is a original message:
`00aaaaaaaaaa2dd4e8317ff78ff6efefff88f7ddff17ce8008700910100077082200000 (91% H)`
and this is the "new" one:
`caaaaaaaaaa2dd4dcaa188000c018ffeeff1358185291fff04e00000 (91%H , 18.5°C)`
They are quite different.
I spotted no wind speed, rain, wind direction or a checksum. 
Note the original messages include a lot more human readable data.
I still believe that the message itself is already wrong.

zuckschwerdt, 2019-08-20:
The signal looks ok, but it's strange that the end is always just a big chunk of `0` (no FSK data, the signal stays at the lower frequency).
The preamble, sync word, and length match expectations. The payload then does not. Sounds reasonable that the transmitter is bad/broken.

wildstray, 2020-01-10:
I'm in the same situation. Bresser model 7002580
Messages received (coherent with messages received by mrbgfromv):
eaaaaaaaaaa2dd456101880173418ffffff2488ffeeb7ff01b700000
caaaaaaaaaa2dd4fd941880173418ffffff2488220231fff01700000
d55555555545ba9fb2831002e6831fffffe4910440463ffe02e00000
Current parameters on the display: temp 22.0, hum 31%, wind speed 0, wind direction might be "b" because is a step between SW and W, rain 10.4mm (and pressure 1032mb... but where is the sensor, insider or outside?)
It's driving me crazy, they changed encoding for sure...

mrbgfromv, 2020-01-10:
Just a note: Mine developed over time increasing transmission errors (display showed "Err" and "---") and so I returned it back. Good luck with this thing.

ArneNaessens, 2020-02-20:
Any success on extracting data from this device? I have the same 7002590 device.

peteakalad, 2020-04-15:
I've purchased a new 7002510 this week - 7002590 ought to be the same transmitter protocol as currently.
This is mine;
https://www.amazon.co.uk/Bresser-Weather-Station-Control-German/dp/B0170A47DI

Try running current master like follows, works for me;
rtl_433 -f 868300000 -R 119 -Y classic -s 250k

I would suspect the -Y classic will help.

Note: I have a PR/fork with some wind measurement fixes.

jrp13149, 2020-12-26:
My partner got a Bresser 5-in-1 yesterday. I spent a bit of time banging a post into the garden this afternoon, and setting it up. She's happy with the display device, but I wanted to see if I could do better. I just happened to have a Raspberry PI, and a USB DVB-T lying in the garage. so, after a bit of messing, I got rtl_433 installed

`rtl_433 -f 868.3M`
 
Sees the USB device, but no data reported.

`rtl_433 -f 868300000  -a 4`

gets me:

Registered 146 out of 176 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 ]
Detached kernel driver
Found Fitipower FC0013 tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
*** signal_start = 4784648, signal_end = 4839567, signal_len = 54919, pulses_found = 1
Distance coding: Pulse length 34918

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 34918
bitbuffer:: Number of rows: 0 
*** signal_start = 16783281, signal_end = 16838155, signal_len = 54874, pulses_found = 1
Distance coding: Pulse length 34873

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 34873
bitbuffer:: Number of rows: 0 
*** signal_start = 28781896, signal_end = 28836715, signal_len = 54819, pulses_found = 1
Distance coding: Pulse length 34818

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 34818
bitbuffer:: Number of rows: 0 

`rtl_433 -f 868300000  -X n=bresser,m=FSK_PCM,s=120,l=120,r=2500`

gets me: 

time      : 2020-12-26 16:27:55
model     : bresser      count     : 1             num_rows  : 1             rows      : 
len       : 259          data      : d6b76bf16eae7983ff8b7f7ef9c7fc27ffffe38df003d80c086300ef080200000
codes     : {259}d6b76bf16eae7983ff8b7f7ef9c7fc27ffffe38df003d80c086300ef080200000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-12-26 16:28:07
model     : bresser      count     : 1             num_rows  : 1             rows      : 
len       : 237          data      : d7557ff7ea71a3ff0e7fb7ff3ff5ffffffef81f703906600ee0801000000
codes     : {237}d7557ff7ea71a3ff0e7fb7ff3ff5ffffffef81f703906600ee0801000000

I'm not seeing much evidence of transmission problems. More that the defaults in the rtl_433 aren't quite right.

Is there documentation anywhere of any previous Bresser message strings? I'm interested in getting this working, and have a little bit of time in the next week.

Thanks!

peteakalad, 2020-12-26:
Don't see the normal preamble for the payload in your codes, even shifting them, the other bresser protcols all start with 2d d4

If you add -Y classic -s 250k does it do the same?

If it is different it may be the same/similar to the newer 6in1 protocol.

In terms of documentation have a search through the issues for bresser, and in the comments at the top of the bresser decoders in:
https://github.com/merbanan/rtl_433/tree/master/src/devices