merbanan/rtl_433 #1127: Signal level for decode. Not decoding

kevtheskin, 2019-08-17:
Hello there, For some reason I can no longer decode anything. I am running windows build rtl_433-18.12-218-gc234929-win-x64 (wins10 pro)and nothing. What is the lowest snr that the decoder can decode. It wont decode my own weather station at 20ft?. Uhf signals in sdr# are good.

Cheers Kev

merbanan, 2019-08-17:
Try increasing the gain (-g). The issue report is lacking details so it is hard to give any better advice.

kevtheskin, 2019-08-17:
Hiya peeps, Thanks for getting back to me. I have tried various gain settings and different antenna's and a pre-amp . I think maybe my nividia gpu maybe causing noise on the 433 ism band. My old school weather station rx picks up the sensor ok so I know its tx-ing. Kev

kevtheskin, 2019-08-17:
Ps if required how do I generate more info for you. Cheers K

merbanan, 2019-08-17:
Run with -A and -D -D then post the output.

kevtheskin, 2019-08-18:
Hello again Merbanan, Its seems to be working again ?

kevtheskin, 2019-08-18:
time      : 2019-08-18 15:18:06                    brand     : OS
model     : THN132N      House Code: 228
Channel   : 1            Battery   : OK            Celsius   : 17.40 C
Modulation: ASK          Freq      : 434.0 MHz
RSSI      : -12.1 dB     SNR       : 9.0 dB        Noise     : -21.2 dB

zuckschwerdt, 2019-08-28:
Reopen if this comes up again and there is more info.

closed 2019-08-28 by zuckschwerdt