merbanan/rtl_433 #1129: MQTT Two Devices - Two Instances - MQTT Client Id Clash Param Order

peteakalad, 2019-08-18:
I've two devices and was trying to start two instances of rtl_433 connecting to my mqtt broker, like so;

rtl_433 -F mqtt://localhost:1883 -M utc -f 433887500 -d 0
rtl_433 -F mqtt://localhost:1883 -M utc -f 868221000 -d 1

Both instances would continually connect and disconnect from the broker.

The mosquitto log stated;
Client rtl_433-65c8ffff already connected, closing old connection.

Doing some debugging eventually identified that the cause was the order of the parameters, if the -d(evice) parameter is after the -F mqtt parameter, the -F is processed first before the -d has initialised the dev_query value used as an input to generate the MQTT client id.  As a result the client id is always reported as the same version even though the device id is different.

Simple workaround is to ensure to put the -d parameter before the -F like so;
rtl_433 -d 0 -F mqtt://localhost:1883 -M utc -f 433887500
rtl_433 -d 1 -F mqtt://localhost:1883 -M utc -f 868221000

zuckschwerdt, 2019-08-19:
Good analysis. We should probably document that. I'll take a look soon.

merbanan, 2021-03-13:
@zuckschwerdt have you had the time to look at it yet ?

zuckschwerdt, 2021-03-13:
Oops. Yeah we still should document that parameters are read left to right. Very likely `-d` should always be first.

helix26j, 2023-07-13:
I'm trying to output two different MQTT outputs in the same rtl_433 config file to the same server for the home assistant auto-discovery in two different formats but the Client-ID conflicts even if I use different user names.

This is also causing 500 KB/sec of network traffic on the current nightly while the nightly branch from 202302271156 is barely measurable.

Could an index of the MQTT outputs be added to the client ID as well?  (like the current ID + 0 for the first output and + 1 for the second, etc)?

zuckschwerdt, 2023-07-13:
Interesting. Yes, that's a use-case we should support.

I'm leaning towards adding the full `-F mqtt...` parameter list to the client-id hash instead of a sequence number. It might even help people using the same hostname on multiple instances (well if anything about the parameters is different...).

I'll try that asap.

zuckschwerdt, 2023-07-13:
Change `src/output_mqtt.c`like this, recompile, run and it should work, right?
```diff
@@ -500,8 +500,9 @@ struct data_output *data_output_mqtt_create(struct mg_mgr *mgr, char *param, cha
     // generate a short deterministic client_id to identify this input device on restart
     uint16_t host_crc = crc16((uint8_t *)mqtt->hostname, strlen(mqtt->hostname), 0x1021, 0xffff);
     uint16_t devq_crc = crc16((uint8_t *)dev_hint, dev_hint ? strlen(dev_hint) : 0, 0x1021, 0xffff);
-    char client_id[17];
-    snprintf(client_id, sizeof(client_id), "rtl_433-%04x%04x", host_crc, devq_crc);
+    uint16_t parm_crc = crc16((uint8_t *)param, param ? strlen(param) : 0, 0x1021, 0xffff);
+    char client_id[21];
+    snprintf(client_id, sizeof(client_id), "rtl_433-%04x%04x%04x", host_crc, devq_crc, parm_crc);
```

helix26j, 2023-07-13:
This did resolve the issue on both versions.   The MQTT server shows both connections at once with unique client IDs and neither disconnect while receiving data.
Thanks

zuckschwerdt, 2023-07-14:
I'll run some further checks add likely add that code soon.

gdt, 2023-09-25:
Is the fix in?  This issue seems almost closable.

zuckschwerdt, 2023-09-26:
[MQTT 3.1.1](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/csprd01/mqtt-v3.1.1-csprd01.html#_Toc376954341) specifies that the broker MUST accept clients id's between 1 and 23 characters.
I don't see any reason not to implement this change. Only the exact same hostname/device/mqtt-params will allow a transparent MQTT reconnect -- but changing any of that should mean you're not interested in continuing a session.

commit 63a049bf (2023-09-26): Change MQTT client ID to contain all params (closes #1129)
closed 2023-09-26 by zuckschwerdt