merbanan/rtl_433 #1132: Support ESIC EMR7379 power meter?

petterreinholdtsen, 2019-08-23:
Hi.  I got a power usage sensor from ESIC Technologies ltd. named "Typ.: EMR7370 Art.nr.: 36-3981", bought on Clas Ohlson.  I would very much like to see rtl_433 support it, but am unsure how to identify the radio signal from the unit.

The meter measure volt and ampere on a power connector.  I assume it is using 433 MHz, but am unsure how to verify this.  The meter seem to transmitt updates several times a minute, possibly every 30 or 15 seconds (estimate after looking at its display).

I've tried to run 'rtl_433 -S all -T 120' and tried to correlate the files created with the meter updates, but it is not obvious to me which of the files correspond to the sensor readings.  A complicating factor is that there is a lot of sensor readings in the area, so I can not really know that the signals I am seeing is coming from this meter.

Can you provide some guidance?

zuckschwerdt, 2019-08-23:
Remove the antenna and put the receiver at 10 cm to the unit. Also use CubicSDR or Gqrx for a first impression where/when there are signals.

petterreinholdtsen, 2019-08-23:
I've continued to research this meter, and discovered that the original model name I used was the receiver, not the sender.   The receiver is described in http://hobbyelektronik.org/w/index.php/EMR7370 and the sender in https://wiki.volkszaehler.org/hardware/channels/meters/power/emt7110 .  The protocol details seem to be described in German in https://www.seegel-systeme.de/2015/09/07/das-funk-energiekosten-messgeraet-emt7110/ :

* Mittenfrequenz 868,28 MHz
* Modulation: FSK
* Frequenzhub: +/- 90 kHz
* Datenrate: 9,579 kbit/s
* Synchronwort: 0x2DD4

petterreinholdtsen, 2019-08-23:
Based on the above data, I tried to run 'rtl_433 -f 868.28M -s 1024k -S all -T 120' with the antenna on top of the sensor (not easy to place the receiver on top of the power connector), and got a lot of files (size range 128KiB - 3MiB), none of which rtl_433 could understand.  How can I figure out which one is needed?

gqrx could se several signals.

zuckschwerdt, 2019-08-23:
The spectrogram at http://triq.net/iqs can help identify files with interesting signals.

petterreinholdtsen, 2019-08-23:
I had a look, at they all look interesting to me.  What am I looking for?

zuckschwerdt, 2019-08-23:
Upload a smaller file (as zip) here and we can take a look why rtl_433 can't detect the data.

petterreinholdtsen, 2019-08-23:
Attached is a zip file created using 'rtl_433 -f 868.28M -s 1024k -S all -T 120' in one shell, while running 'echo signal > "date-$(LC_ALL=C date --iso=second)"' in the same directory every time I saw the volt information change on the receiver display.  Thus, the data packages should be in the files preseding the date-* file when sorting by creation time.

[testrun.zip](https://github.com/merbanan/rtl_433/files/3535841/testrun.zip)

For the record, this is the ordering I see:

```
g001_868.28M_1024k.cu8
g002_868.28M_1024k.cu8
g003_868.28M_1024k.cu8
date-2019-08-23T12:52:34-03:00
g004_868.28M_1024k.cu8
g005_868.28M_1024k.cu8
g006_868.28M_1024k.cu8
g007_868.28M_1024k.cu8
date-2019-08-23T12:52:48-03:00
g008_868.28M_1024k.cu8
g009_868.28M_1024k.cu8
g010_868.28M_1024k.cu8
g011_868.28M_1024k.cu8
date-2019-08-23T12:53:04-03:00
g012_868.28M_1024k.cu8
g013_868.28M_1024k.cu8
g014_868.28M_1024k.cu8
g015_868.28M_1024k.cu8
g016_868.28M_1024k.cu8
date-2019-08-23T12:53:18-03:00
g017_868.28M_1024k.cu8
g018_868.28M_1024k.cu8
g019_868.28M_1024k.cu8
g020_868.28M_1024k.cu8
date-2019-08-23T12:53:33-03:00
```

petterreinholdtsen, 2019-08-23:
I came across <URL: https://github.com/mhop/fhem-mirror/blob/master/fhem/FHEM/36_EMT7110.pm > with perl code that seem to decode these packages.

zuckschwerdt, 2019-08-23:
Great work. It decodes well. Try e.g.
```
rtl_433 -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=107000' g003_868.28M_1024k.cu8
```
You get
```
{132}555516ea12b52a43a000000065a00025f
{132}555516ea12b52a43a000000066a00024f
{131}aaaa2dd4256a548740000000cd400049e
{132}555516ea12b52a43a0000000662000257
{131}2aaa2dd4256a548740000000cc40004ae
{132}555516ea12b52a43a0000015e5200010f
{131}aaaa2dd4256a54874000002bca400021e
{132}555516ea12b52a43a0000000662000257
```
And then you can sync for `2dd4` like [in this BitBench](http://triq.net/bitbench?c=%7B132%7D555516ea12b52a43a000000065a00025f&c=%7B132%7D555516ea12b52a43a000000066a00024f&c=%7B131%7Daaaa2dd4256a548740000000cd400049e&c=%7B132%7D555516ea12b52a43a0000000662000257&c=%7B131%7D2aaa2dd4256a548740000000cc40004ae&c=%7B132%7D555516ea12b52a43a0000015e5200010f&c=%7B131%7Daaaa2dd4256a54874000002bca400021e&c=%7B132%7D555516ea12b52a43a0000000662000257&f=SYNC%3A16h%20DATA%3A%208h%20%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208h%208x&a=Sync&m=2dd4) to get:
```
2d d4 25 6a 54 87 40 00 00 00 cb 40 00 4b 7
2d d4 25 6a 54 87 40 00 00 00 cd 40 00 49 7
2d d4 25 6a 54 87 40 00 00 00 cd 40 00 49 7
2d d4 25 6a 54 87 40 00 00 00 cc 40 00 4a 7
2d d4 25 6a 54 87 40 00 00 00 cc 40 00 4a 7
2d d4 25 6a 54 87 40 00 00 2b ca 40 00 21 7
2d d4 25 6a 54 87 40 00 00 2b ca 40 00 21 7
2d d4 25 6a 54 87 40 00 00 00 cc 40 00 4a 7
```

petterreinholdtsen, 2019-08-23:
Thank you, but I am not quite sure where to go from here.  Is this helping?

```
$ rtl_433 -f868.28M  -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=107000'  
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/caninos/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 104 out of 133 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-133 ]
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 1024000 S/s.
Tuner gain set to Auto.
Tuned to 868.280MHz.
Allocating 15 zero-copy buffers
Detected Kernel usbfs mmap() bug, falling back to buffers in userspace
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:18
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 132          data      : 555516ea12b52a43a000000069a00021f
codes     : {132}555516ea12b52a43a000000069a00021f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:18
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 132          data      : 555516ea12b52a43a000000069a00021f
codes     : {132}555516ea12b52a43a000000069a00021f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:19
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 18           data      : ffe10
codes     : {18}ffe10
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:19
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 20           data      : ffe14
codes     : {20}ffe14
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:19
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 18           data      : ff810
codes     : {18}ff810
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:19
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 19           data      : ffe28
codes     : {19}ffe28
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:24
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 19           data      : ffe08
codes     : {19}ffe08
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:24
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 19           data      : ffc28
codes     : {19}ffc28
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:29
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 19           data      : ffc28
codes     : {19}ffc28
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:29
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
[... 54 lines cut ...]
codes     : {18}00050
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:44
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 18           data      : ffc10
codes     : {18}ffc10
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:48
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 132          data      : 555516ea12b52a43a000000069a00021f
codes     : {132}555516ea12b52a43a000000069a00021f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:48
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 132          data      : 555516ea12b52a43a000000069a00021f
codes     : {132}555516ea12b52a43a000000069a00021f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-23 19:25:50
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 18           data      : ffc10
codes     : {18}ffc10
^CSignal caught, exiting!
Reattached kernel driver
%
```

zuckschwerdt, 2019-08-24:
If you want to get the data values directly you can use:
```
rtl_433 -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=107000,bits>=128,preamble=aa2dd4'
```
This removes the spurious short lines and syncs to the `2dd4` preamble.
The next step is to relate the data to measured values and decode it. The link you posted helps but it does not seem like a direct fit. (i.e. `256a` should be the ID and then `5487` the voltage, which I can't really see.)

petterreinholdtsen, 2019-08-24:
[Christian W. Zuckschwerdt]
> If you want to get the data values directly you can use:
> ```
> rtl_433 -s1024k -X 'n=EMT7110,m=FSK_PCM,s=104,l=104,r=107000,bits>=128,preamble=aa2dd4'
> ```

Right.  It gave me these lines:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-24 09:35:18
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a54874000002bd140001ae
codes     : {99}256a54874000002bd140001ae
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-24 09:35:18
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a54874000002bd140001ae
codes     : {99}256a54874000002bd140001ae
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-24 09:35:33
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a548740000000d1400045e
codes     : {99}256a548740000000d1400045e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-24 09:35:33
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a548740000000d1400045e
codes     : {99}256a548740000000d1400045e

> This removes the spurious short lines and syncs to the `2dd4`
> preamble.  The next step is to relate the data to measured values and
> decode it. The link you posted helps but it does not seem like a
> direct fit. (i.e. `256a` should be the ID and then `5487` the voltage,
> which I can't really see.)

The reported voltage is around 230, +- 1.5 V, and definitely not
5487. :) The ampere is 0, as nothing is connected to the sensor.

-- 
Happy hacking
Petter Reinholdtsen

zuckschwerdt, 2019-08-24:
The voltage could be scaled (e.g. by ten, thus 2300) and the value is hex, 0x5487 = 21639 -- but I don't think thats 216.39 V.
Perhaps connect some different known loads (plain incandescent lights are good) and observe the changes in the data to spot the Ampere and Watt fields.

petterreinholdtsen, 2019-08-25:
I agree the perl description do not seem to match the radio format.  The one from seegel-systeme.de seem to be closer.  Anyway, here are a few more example strings, annotated with the reading from the display at the time.

```
25 6a 54 87 40 02 00 27 d4 40 00 19 e  0.3 A, ~230 V
25 6a 54 87 40 00 00 2b d1 40 00 1a e  0.0 A, ~230 V
25 6a 54 87 40 00 00 00 d1 40 00 45 e  0.0 A, ~230 V
25 6a 54 87 40 02 00 27 da 40 01 12 e  0.3 A, ~230 V
25 6a 54 87 40 02 00 2b db 40 01 0d e  0.4 A, ~230 V
25 6a 54 87 40 02 00 2c d9 40 01 0e e  0.4 A, ~230 V
25 6a 54 87 40 00 00 07 da 40 01 34 e  0.0 A, ~230 V
25 6a 54 87 40 00 00 07 d8 40 01 36 e  0.0 A, 236,5 V, 0 W
25 6a 54 87 40 02 00 2c da 40 01 0d e  0.4 A, ~230 V, 10 W
```

zuckschwerdt, 2019-08-26:
Oops, I didn't look careful enough ;) Straight forward then. Support added with 8e1e26e.

closed 2019-08-26 by zuckschwerdt

petterreinholdtsen, 2019-08-26:
Thank you very much.  It seem to work for at least some of the fields:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:21:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.041 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:21:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.041 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:17
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:17
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:32
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:32
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:22:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:23:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:23:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:23:17
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.043 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:23:17
model     : ESIC-EMT7110 Sensor ID : 256a5487
[... 18 lines cut ...]
time      : 2019-08-26 08:23:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 227.5 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:24:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 08:24:02
[... cut ...]  ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 228.0 V       Energy    : 0.0 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM

But the Power and Energy values seem to be off.  For example,
0.043A*228.0V = 9.8040W, while the Power entry is 0.0.  I expected it to
be 10.0 W, like it is on the display.

I notice the sensor data always come i pairs.  Is it useful to supress
duplicates?

-- 
Happy hacking
Petter Reinholdtsen

zuckschwerdt, 2019-08-26:
The duplicates are too far apart (14 ms) to merge, sorry.
I'll double check, but I guess a possible confusion could be power (W) vs. energy (Wh). And there is a digit missing, should be 0.01 kWh for 10 Wh, I guess.

zuckschwerdt, 2019-08-26:
I've now added the digit in the `kWh` reading. If you test with e.g.
```
rtl_433 -y 'aaaa 2dd4 256a5487 4 002 0027 da 4 001 12'
```
You'll get the expected: Power     : 1.0 W        Current   : 0.039 A       Voltage   : 237.0 V       Energy    : 0.01 kWh

petterreinholdtsen, 2019-08-26:
[Christian W. Zuckschwerdt]
> I've now added the digit in the `kWh` reading. If you test with e.g.

I tested the latest git, and now get this:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:11:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 229.0 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:11:02
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 229.0 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:11:17
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.042 A       Voltage   : 229.0 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:11:17
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.042 A       Voltage   : 229.0 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM

Still no power reading, but the energy reading have something.

I ran it again with -X
'n=EMT7110,m=FSK_PCM,s=104,l=104,r=107000,bits>=128,preamble=aa2dd4' and
got this:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:32
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a548740000004ca400246e
codes     : {99}256a548740000004ca400246e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:32
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 229.0 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:32
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a548740000004ca400246e
codes     : {99}256a548740000004ca400246e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:32
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.004 A       Voltage   : 229.0 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:47
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a54874000002ac9400221e
codes     : {99}256a54874000002ac9400221e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.042 A       Voltage   : 228.5 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:47
model     : EMT7110      count     : 1             num_rows  : 1             rows      : 
len       : 99           data      : 256a54874000002ac9400221e
codes     : {99}256a54874000002ac9400221e
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-08-26 10:12:47
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 0.0 W        Current   : 0.042 A       Voltage   : 228.5 V       Energy    : 0.02 kWh
Pairing?  : 0            Connected?: 1             Integrity : CHECKSUM

-- 
Happy hacking
Petter Reinholdtsen

zuckschwerdt, 2019-08-26:
The first three zeros in your captured codes is the power reading, nothing I can do there.
Maybe it's an Power facor (effective vs apparent) issue. Best to put a real load on it, say 100W bulb.

petterreinholdtsen, 2019-08-26:
[Christian W. Zuckschwerdt]
> The first three zeros in your captured codes is the power reading,
> nothing I can do there.  Maybe it's an Power facor (effective vs
> apparent) issue. Best to put a real load on it, say 100W bulb.

Good point.  I put a (led) bulb on, and got this reading:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-08-26 15:01:27
model     : EMT7110      count     : 1             num_rows  : 1
rows      : 
len       : 99           data      : 256a54874011003fc94000fde
codes     : {99}256a54874011003fc94000fde
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-08-26 15:01:27
model     : ESIC-EMT7110 Sensor ID : 256a5487
Power     : 8.500 W      Current   : 0.063 A       Voltage   : 228.5 V
Energy    : 0.00 kWh     Pairing?  : 0             Connected?: 1
Integrity : CHECKSUM

The W, A and V measurements match those on the display, though the
display only show rounted numbers (8.5W, 0.06A, 228.5V).  I'm not sure
where to find the kWh reading on the display, so I can not verify that
it is correct.  I do not know where to find the pairing and connected
settings either.

Anyway, this is working great now.  Thank you very much. :)

-- 
Happy hacking
Petter Reinholdtsen