merbanan/rtl_433 #1134: Lacrosse TX141-BV3 Support in lacrosse_TX141TH_Bv2.c

tristanmoney, 2019-08-25:
Lacrosse TX141-BV3's do not appear to be detected by rtl_433. I used the pulse analyzer and found that the bit length is 33 bits, not 37 bits as is used in lacrosse_TX141TH_Bv2.c. I changed the following two lines and recompiled using Visual Studio which allow BV3's to be detected:

#define LACROSSE_TX141_BITLEN 33 //originally 37

r = bitbuffer_find_repeated_row(bitbuffer, 5, 33); //originally 37

I'm not sure if this will break the BV2 support but it would be nice to include BV3 compatibility in future releases.

zuckschwerdt, 2019-08-25:
Good work. Please upload a few samples and a summary which values you expect in those samples.

tristanmoney, 2019-08-25:
I'm pretty new to this and I'm not sure how to properly upload yet, but here is the output of the recompiled rtl_433 and the sample files. Everything seems to work great (even the test button). I guess it would be nice if it showed that it was a Bv3 sensor but this not a big deal.

time      : 2019-08-25 10:33:51
model     : LaCrosse TX141-Bv2 sensor              Sensor ID : ea
Channel   : 01           Temperature: 16.50 C      Battery   : OK            Test?     : No
*** Saving signal to file g001_433.92M_250k.cu8 (155187 samples, 393216 bytes)

time      : 2019-08-25 10:37:08
model     : LaCrosse TX141-Bv2 sensor              Sensor ID : 22
Channel   : 01           Temperature: 17.00 C      Battery   : OK            Test?     : No
*** Saving signal to file g004_433.92M_250k.cu8 (155361 samples, 393216 bytes)

[Samples.zip](https://github.com/merbanan/rtl_433/files/3538438/Samples.zip)

zuckschwerdt, 2019-08-25:
Thanks, looks good.

zuckschwerdt, 2019-08-25:
Support added with bb2b046.

closed 2019-08-25 by zuckschwerdt