merbanan/rtl_433 #1135: TX35DTH-IT not decoded with sample rate <> 250 kHz

f1oat, 2019-08-27:
With "-s 1024k", all my 868 MHz devices are decoded, except the TX35 sensors.
When using "-s 250k", TX35 decoding is ok.

zuckschwerdt, 2019-08-27:
Can you grab TX35 samples at 1024k? (`-S all`, then filter through those and upload a zip here)

f1oat, 2019-09-03:
Here are 3 samples.
Thank you for your help.
[TX35DTH-IT 1024k.zip](https://github.com/merbanan/rtl_433/files/3571166/TX35DTH-IT.1024k.zip)

zuckschwerdt, 2019-09-04:
Thanks. The samples look good, but the demod won't pick up the FSK.
(Try `sigrok-open.sh g001_868.35M_1024k.cu8 1024` to see that.)
We need to refine the current demod with these samples and use them as baseline for the new demod.

f1oat, 2019-09-05:
I have been able to demodulate @1024k by scaling following sampling rate dependent constants by a factor of 4:
#define PD_MIN_PULSE_SAMPLES (4*10)
#define FSK_DEFAULT_FM_DELTA (6000/4) 
#define FSK_EST_SLOW         (4*64)
#define FSK_EST_FAST         (4*16)  

Now I need to figure out a clean implementation for that.

f1oat, 2019-09-05:
Done, I will push a PR

merbanan, 2021-03-22:
This is superseded by the minmax demodulator.

closed 2021-03-22 by merbanan