merbanan/rtl_433 #1136: Add "protocol" to supported tokens in MQTT format string

curlyel, 2019-08-27:
Would be quite helpful to have the protocol number [protocol] available as token in MQTT format strings, e.g.:

```
-F mqtt://localhost 1883 retain 0 devices sensors/rtl433[/model][/channel:0]
vs.
-F mqtt://localhost 1883 retain 0 devices sensors/rtl433[/protocol][/channel:0]
```
This would help creating much shorter MQTT topics easier to consume/configure in MQTT clients (beside the "-M newmodel" https://github.com/merbanan/rtl_433/pull/986) :

```
sensors/rtl433/Fine_Offset_Electronics__WH0530_Temperature_Rain_sensor/0/temperature_C 35.000000
vs.
sensors/rtl433/79/0/temperature_C 35.000000
```
Current tokens are AFAIK:
https://github.com/merbanan/rtl_433/pull/1079#issuecomment-504760052

zuckschwerdt, 2019-08-27:
Added with faee3a5. Be sure to use `-M protocol`, otherwise that token is not available.

closed 2019-08-27 by zuckschwerdt

curlyel, 2019-08-27:
What??? The issue was just seconds old? You're quicker than light!!!

zuckschwerdt, 2019-08-27:
Maybe not light, but perhaps faster than sound ;) Props if you know the quote:

> At first, the only trace of his passage is a dancing trail of sparks  But once he makes his way out onto a long straight, he begins to leave further evidence: a spume of shattered blue safety glass spraying outward in parallel vanes from all four lanes of traffic as the windows and the windshields of the cars blow out of their frames, spraying into the air like rooster tails behind a speedboat.