merbanan/rtl_433 #1137: CSV output does not work with customary units?

pclov3r, 2019-08-29:
So I am using rtl_433 with customary units and it appears in the file it still wants to attempt to log the temperature in C instead of F. As a result this column remains blank. 

This intended behavior or a bug?

zuckschwerdt, 2019-08-29:
The conversion should work, data is converted before looking at the output. The column header will be wrong though (untranslated). The problem is that column keys are fixed and don't go through a conversion step.

pclov3r, 2019-08-29:
Thanks for the quick acknowledgement of the issue!

zuckschwerdt, 2019-08-29:
As a quick fix, enable one protocol that has native customary output.
E.g. `-R 19 -R 52` for Nexus with the `temperature_F` key from Bresser.

zuckschwerdt, 2019-08-29:
Fixed with f1b3d18.

closed 2019-08-29 by zuckschwerdt