merbanan/rtl_433 #1139: [feature request] Limit output to decodings matching a given list of mappings

curlyel, 2019-08-29:
>The getters need some restructuring anyway. 

_Originally posted by @zuckschwerdt in https://github.com/merbanan/rtl_433_tests/pull/286#issuecomment-526192055_

If this is going to happen, you might consider expanding the getters logic to limit the generated output to just the packets/decodes, where one of the getters map conditions is met, drop all others e.g.:

`get=@0:{20}:code:[123456:REMOTE-A 987654:REMOTE-B 345699:REMOTE-C]:exclusive
`

This would reduce the 'noise' of unwanted decodes being published and prevents potential duplicate decodings from different flex decoders matching on similar settings/timings.

zuckschwerdt, 2019-08-30:
To get something like this right now. How many sensors do you have?
For just a few devices you can list each as a flex decoder on it's own (all in a single `conf` file) with a `match` spec and add a dummy getter (`get=@0:{1}:command:[0:OPEN 1:OPEN]`) to report triggering.
For more than a handful this will be tedious though…

curlyel, 2019-08-30:
Currently I do have just two of them. Just ordered some additional.
But I will never have more than five to ten per installation most probably.

So I can live with it as it is right now :-)

Nevertheless I do share my thougts/ideas/wishes/etc. when they come across.
Sometimes it's easy to implement and some others may benefit from it too ...

So from my side: No need to implementing this right now :-)

zuckschwerdt, 2019-08-30:
Thanks, yes, always great what the applications are. Helps to shape new features!

gdt, 2023-09-25:
@curlyel Is it still useful to have this issue open?  It feels like no, but I don't want to be that heavy handed.

zuckschwerdt, 2023-09-26:
This is a nice feature but unlikely to happen in isolation. I plan to rewrite the flex decoder with some processing logic, someday.
We can close this, the feature will be included in some rewrite but not on it's own.

closed 2023-09-26 by zuckschwerdt