merbanan/rtl_433 #1141: rtl_433 on my Raspberry PI is throwing a usb_claim_interface error -6

juergenschubert, 2019-08-31:
rtl_433 -M 109 or a rtl_433 -r 109 is telling me:
rtl_433 version 18.12-204-gb1b62f2 branch master at 201905030920 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 98 out of 125 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-63 67-71 73-100 102-103 108-116 119 121 124-125 ]
usb_claim_interface error -6

Geoff99, 2019-09-01:
Hi @juergenschubert

I vaguely remember getting to get that error occassionally.

From (hazy) recollection it means that the rtl dongle is already in use by another program.  In my case it turned out to be another application that I had set up to start running automatically when I booted the RPi that was grabbing the rtl dongle and not releasing it.  In the short term I had to find out which program was using the rtl dongle and kill it.  In the longer term I had to remember how I had set up the autostart program and remove it from the boot up sequence. 

It might be something similar in your case.  Maybe an earlier atempt to run rtl_433 is still active in the background?

Hope that helps you track down the cause of the problem.

osborne82, 2019-09-01:
You need to Blacklist the dvb drivers from loading on the pi
blacklist dvb_usb_rtl28xxu
>blacklist rtl2830
>blacklist rtl2832
>blacklist dvb_usb_v2
>blacklist dvb_core

Should be good once rebooted the pi

closed 2019-09-05 by zuckschwerdt