merbanan/rtl_433 #1143: Publish to WU?

jmcmorais, 2019-09-01:
Hi, I'm having trouble posting to WUnderground! The PI test is ok, my Acurite is working well and I see the result of the sensors! 3 years ago I have a WU ID (IMINASGE30) when using a 1080 station with usb access via console. My acurite has no console / PC and so I decided to install rtl_433 in a PI. I already updated the weather.ini file but nothing happens!
I ask you to help me clarify where I can review! I've read the weatherd.py file but didn't want to move it because I'm not an expert on the subject ... hi!
Thank you!
Jo

zuckschwerdt, 2019-09-02:
I think people are using WeeWx-SDR for this. Baybe https://github.com/weewx/weewx/wiki/sdr-rpi-recipe can help.
Closing this as it's not really about rtl_433. But perhaps other users can give additional help.

closed 2019-09-02 by zuckschwerdt

jmcmorais, 2019-09-02:
Interesting about WeeWX. I'll study better but at first glance it seems ideal in my case!
Thank you for your help!