merbanan/rtl_433 #1148: DirecTV RC66RX missing Volume Up, Down and Mute

psorobka, 2019-09-10:
Hi, I have a RC66RX Remote and it's working perfectly, except for Volume Up, Down and Mute. When I press them, I get 

> "model" : "DirecTV-RC66RX", "id" : 1, "button_id" : 91, "button_name" : "DTV: IR ALERT"

I can get the samples, if needed, just tell me how.

zuckschwerdt, 2019-09-11:
The button 91 (0x5B) is listed as "DTV: IR ALERT", not sure what that means. But it's not missing.
https://github.com/merbanan/rtl_433/blob/master/src/devices/directv.c#L156

Samples are not needed. Just list the buttons that output a misleading text with the button ID here.

zuckschwerdt, 2019-09-11:
S.a. https://github.com/merbanan/rtl_433_tests/tree/master/tests/directv#button-code-mapping (click the hider)

psorobka, 2019-09-11:
The same story goes for TV Input, On and Off. 
It seems that those buttons are not supported.

klohner, 2019-09-19:
From my tests with the remote, I found no situation where the remote would generate unique RF codes for the volume buttons.  It seems that by design, the remote will only send out a generic RF code that a volume-related button is being pressed.  It simultaneously sends out the unique volume up/down/mute code only via an IR signal designed for whatever device set up to provide audio power amplification.

Something similar is encountered with the TV Input button.  In all positions of the top DTV/AV1/AV2/TV AV switch, no RF signal is sent at all when pressing the TV Input button.  Instead, there's only an IR code sent.

As for the On/Off button, my remote only sends unique RF signals if the top mode selector switch is set all the way left to DTV.  In the other AV modes, no RF signal is sent at all.

This issue as far as I can determine, is not with the rtl_433 driver, it's that this is the by-design limits of how the remote functions when sending RF signals.

zuckschwerdt, 2019-09-19:
@klohner thanks for the explanation. @psorobka please close if nothing else todo.

merbanan, 2020-01-09:
Closing issue.

closed 2020-01-09 by merbanan