merbanan/rtl_433 #1153: Defaults for 915 operation

bill-orange, 2019-09-20:
I have several 433 Mhz devices whose data I can capture easily.  I also have a Ambient weather Station on 915.  I have had no success capturing anything on 915 Mhz.  I suspect my sampling rate or some other command line parameter is inappropriate for the higher frequency.  

Suggestions?  I suppose it could be a software bug but considering that 433 Mhz is successfully captured that seems unlikely.  Perhaps a 915 Mhz example could be added to the instructions.

merbanan, 2019-09-20:
Usually 915MHz devices use FSK modulation with higher bit rates. The rtl_433 defaults doesn't handle that well. You most likely have to increase the sample rate to 1MS/s. And even then the fsk-demod might not be able to handle the signal bitrate. But the signal grabber should be able to capture the signal. If it does not then it is an issue with reception.

zuckschwerdt, 2019-09-20:
Hands on example: `rtl_433 -f 915M -s 1024k -S all` and then inspect the files with the `sigrok_open.sh` or drop them in http://triq.net/iqs

merbanan, 2020-01-08:
By default rtl_433 now sets a 1MS/s rate if the frequency is higher then 800MHz.

closed 2020-01-08 by merbanan