merbanan/rtl_433 #1155: Noob

itishfrog, 2019-09-22:
I have strangest issue on version 18.05 all Oregon scientific devices are seen but newer versons of rtl_433 only see the  pcr800 

dont know where and how to fix this, and not quite up there with coding so i get lost

unfortunately now that i have updated, even though i go back to old version the missing devices stay gone
devices missing are WGR800 and THGR810

i used the pi image from https://www.youtube.com/watch?v=L0fSEbGEY-Q 
would rather not reload entire image as i have just got node-red running smoothly

Please advise

zuckschwerdt, 2019-09-22:
Can you grab some samples (`rtl_433 -S all`), pick and sort interesting ones and upload as zip here. We'll take a look.

merbanan, 2023-02-08:
No activity in several years.

closed 2023-02-08 by merbanan