merbanan/rtl_433 #1156: Completion of error handling

elfring, 2019-09-23:
Would you like to add more error handling for return values from functions like the following?
- [fprintf](https://pubs.opengroup.org/onlinepubs/9699919799/functions/fprintf.html "Print formatted output") ⇒ [pulse_data_print](https://github.com/merbanan/rtl_433/blob/c234929924513ffe300fc16fc1f81c4f7f16b7fe/src/pulse_detect.c#L26)
- [malloc](https://pubs.opengroup.org/onlinepubs/9699919799/functions/malloc.html "Memory allocation") ⇒ [create_device](https://github.com/merbanan/rtl_433/blob/bd05f662cc9383e6e97acc05b779cc1fc80a04db/src/decoder_util.c#L20)

zuckschwerdt, 2019-09-23:
True, allocs should be guarded. I need to check where we missed that.
Not sure what to do if fprint to stdout/stderr fails?

Mindavi, 2019-10-15:
I don't think not being able to print should be acted upon. I guess there are situations where those pipes are closed (e.g. docker).

elfring, 2019-10-16:
[A few update candidates were left over](https://github.com/merbanan/rtl_433/blob/78a4fbed62b3bd6cc7d0f2d35db1dd42c182e651/src/devices/flex.c#L494 "Update candidate: flex_create_device()").

-----

How do you think about to improve [static source code analysis](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2C_C.2B.2B "Tools for code analysis") also for this software?

-----

* Are you interested to apply [aspect-oriented software development](https://aspectc.org/ "Software development with AspectC++?")?
* Would you like to encapsulate error detection and corresponding exception handling as [a reusable aspect](https://en.wikipedia.org/wiki/Aspect_%28computer_programming%29 "Combination of a pointcut with advice") in this software?

zuckschwerdt, 2019-10-16:
We don't really care about crashing in low-memory situations, but checking or wrapping all heap interacting calls seems like a desirable thing. (I guess that would be a header file with inline functions, but you might call it aspect.)
Did you use an automated checker to spot those calls? Or just (regex-)search and gut-feeling?

elfring, 2019-10-16:
>Did you use an automated checker to spot those calls?

Not for this bug report. - How many open issues can still be detected by the GitHub source code search interface (for example)?

zuckschwerdt, 2019-10-17:
Should be addressed more thoroughly with de8f73 now. Every alloc and strdup has a check and warning output or fatal abort now. Enforced with a (crude) rule in the style-check.

gdt, 2023-09-29:
Calling this done at 4 years.   Feel free to open PRs to improve!

closed 2023-09-29 by gdt

elfring, 2023-09-29:
:thought_balloon: [There are remaining update candidates](https://github.com/merbanan/rtl_433/blob/027240807198f5bf27e2db703f67a01eb83c1f68/src/output_log.c#L43 "Data logging").

gdt, 2023-09-29:
Could you open a PR?

commit 78a4fbed (2019-10-16): Add alloc checks, fixes #1156