merbanan/rtl_433 #1161: Please add the device Lidl Auriol hg02832

OverSave, 2019-10-07:
Hello, Please add the device Lidl Auriol hg02832.

`rtl_433 -r g001_433.93M_250k.cu8 -a -A
rtl_433 version 19.08-18-g8eecdbb branch master at 201909241811 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.
Registered 108 out of 138 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-138 ]
Test mode active. Reading samples from file: g001_433.93M_250k.cu8
Detected OOK package    @0.211772s
Analyzing pulses...
Total count:   44,  width: 40.22 ms             (10056 S)
Pulse width distribution:
 [ 0] count:   21,  width:  252 us [244;308]    (  63 S)
 [ 1] count:    3,  width:  860 us [852;864]    ( 215 S)
 [ 2] count:   20,  width:  616 us [608;620]    ( 154 S)
Gap width distribution:
 [ 0] count:    4,  width:  848 us [840;856]    ( 212 S)
 [ 1] count:   20,  width:  600 us [600;608]    ( 150 S)
 [ 2] count:   19,  width:  236 us [236;248]    (  59 S)
Pulse period distribution:
 [ 0] count:    1,  width: 1148 us [1148;1148]  ( 287 S)
 [ 1] count:    3,  width: 1708 us [1700;1720]  ( 427 S)
 [ 2] count:   39,  width:  852 us [848;860]    ( 213 S)
Level estimates [high, low]:   1676,     17
RSSI: -9.9 dB SNR: 19.7 dB Noise: -29.6 dB
Frequency offsets [F1, F2]:     566,      0     (+2.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 252, long_width: 616, reset_limit: 860, sync_width: 860
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=252,l=616,r=860,g=0,t=0,y=860'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] { 1} 80             : 1
[01] {40} dd ac 5f 92 10 : 11011101 10101100 01011111 10010010 00010000

Detected OOK package    @0.312884s
Analyzing pulses...
Total count:   44,  width: 40.24 ms             (10061 S)
Pulse width distribution:
 [ 0] count:    1,  width:  328 us [328;328]    (  82 S)
 [ 1] count:    3,  width:  860 us [860;860]    ( 215 S)
 [ 2] count:   20,  width:  248 us [244;260]    (  62 S)
 [ 3] count:   20,  width:  616 us [608;624]    ( 154 S)
Gap width distribution:
 [ 0] count:    4,  width:  848 us [844;856]    ( 212 S)
 [ 1] count:   20,  width:  600 us [596;608]    ( 150 S)
 [ 2] count:   19,  width:  236 us [232;248]    (  59 S)
Pulse period distribution:
 [ 0] count:    1,  width: 1176 us [1176;1176]  ( 294 S)
 [ 1] count:    3,  width: 1708 us [1704;1716]  ( 427 S)
 [ 2] count:   39,  width:  852 us [844;864]    ( 213 S)
Level estimates [high, low]:   1756,     16
RSSI: -9.7 dB SNR: 20.1 dB Noise: -29.8 dB
Frequency offsets [F1, F2]:     651,      0     (+2.5 kHz, +0.0 kHz)
Guessing modulation: No clue...

Detected OOK package    @0.414024s
Analyzing pulses...
Total count:   44,  width: 40.24 ms             (10061 S)
Pulse width distribution:
 [ 0] count:    1,  width:  332 us [332;332]    (  83 S)
 [ 1] count:    3,  width:  860 us [856;864]    ( 215 S)
 [ 2] count:   20,  width:  248 us [240;260]    (  62 S)
 [ 3] count:   20,  width:  616 us [608;624]    ( 154 S)
Gap width distribution:
 [ 0] count:    4,  width:  844 us [840;848]    ( 211 S)
[... 400 lines cut ...]
Pulse width distribution:
 [ 0] count:   21,  width:  252 us [244;316]    (  63 S)
 [ 1] count:    3,  width:  860 us [856;868]    ( 215 S)
 [ 2] count:   20,  width:  612 us [608;620]    ( 153 S)
Gap width distribution:
 [ 0] count:    4,  width:  844 us [836;852]    ( 211 S)
 [ 1] count:   20,  width:  600 us [600;612]    ( 150 S)
 [ 2] count:   19,  width:  236 us [236;244]    (  59 S)
Pulse period distribution:
 [ 0] count:    1,  width: 1152 us [1152;1152]  ( 288 S)
 [ 1] count:    3,  width: 1708 us [1704;1720]  ( 427 S)
 [ 2] count:   39,  width:  852 us [844;864]    ( 213 S)
Level estimates [high, low]:   1712,     12
RSSI: -9.8 dB SNR: 21.2 dB Noise: -31.0 dB
Frequency offsets [F1, F2]:    -770,      0     (-2.9 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 252, long_width: 612, reset_limit: 856, sync_width: 860
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=252,l=612,r=856,g=0,t=0,y=860'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] { 1} 80             : 1
[01] {40} dd ac 5f 92 10 : 11011101 10101100 01011111 10010010 00010000
`

[image]

zuckschwerdt, 2019-10-07:
The bits before temperature are likely high bits for temperature. 7 bit isn't enough. common is 12 bit. so 4 bit status (batt...) and 12 bit temp. the last byte is a checksum. seems easy enough, I'll add a decoder once we have samples: https://github.com/merbanan/rtl_433#supporting-additional-devices-and-test-data

OverSave, 2019-10-09:
i added a fork and pull request

zuckschwerdt, 2019-10-09:
I'll add a decoder soon. In the meantime:
```
rtl_433 -X 'n=auriol-hg02832,
  m=OOK_PWM,
  s=252,
  l=612,
  y=860,
  g=750,
  r=62990,
  invert,
  bits>=40,
  get=@0:{8}:id,
  get=@8:{8}:humi,
  get=@16:{4}:flags,
  get=@20:{12}:temp,
  get=@32:{8}:chk' tests/auriol/06/*.cu8
```

zuckschwerdt, 2019-10-09:
To crack the checksum, can you collect a range of unique codes? Just a text file with lines like this:
```
{40}2254009d06
```

merbanan, 2019-10-09:
2254009d06
225300b308
225210b68f
225c10b88f
225c20ba28
225f20c30b
225f00c41a
225f10c1ac
225f20bebf

225210b68f -> 2^6=4
225c10b88f -> c^8=4

Looks linear.

merbanan, 2019-10-09:
@OverSave we need a large amount of messages to be able to figure out the checksum. I think that the id changes when you change the batteries. So run the command for a day, change the battery and repeat. Then post all the messages you get.

OverSave, 2019-10-09:
this command?

rtl_433 -X 'n=auriol-hg02832, m=OOK_PWM, s=252, l=612, y=860, g=750, r=62990, invert, bits>=40, get=@0:{8}:id, get=@8:{8}:humi, get=@16:{4}:flags, get=@20:{12}:temp, get=@32:{8}:chk' tests/auriol/06/*.cu8

zuckschwerdt, 2019-10-09:
Yes, just not the `tests/auriol/06/*.cu8` part. That's offline reading, you want live reception. Also with `-F json:yourlog.txt` to collect the output. And you can remove all the `,get=...` parameters if you don't care about the debug output. Makes the log shorter. Use a different log name (date suffix?) each time to keep track.

merbanan, 2019-10-09:
Just run with rtl_433 -X 'n=auriol-hg02832, m=OOK_PWM, s=252, l=612, y=860, g=750, r=62990, invert, bits>=40, get=@0:{8}:id, get=@8:{8}:humi, get=@16:{4}:flags, get=@20:{12}:temp, get=@32:{8}:chk'

You can redirect the decoded messages to a file.

OverSave, 2019-10-11:
I wanted to create a 24 hour log but after 7 hours it was canceled. I hope the 7 hours are enough

zuckschwerdt, 2019-10-11:
Thanks! Add here as a zip would be ok too. I'll filter the unique codes and see what we get.

zuckschwerdt, 2019-10-11:
Strangely I don't see how the keys are constructed. More unique codes will be helpful. Can you reset the ID (remove and replace the battery) and record more codes? Just drop a zip here.

OverSave, 2019-10-11:
all right

OverSave, 2019-10-12:
okay here are another 20 hours.

[20191011_newbat_log.zip](https://github.com/merbanan/rtl_433/files/3720369/20191011_newbat_log.zip)

zuckschwerdt, 2019-10-12:
The ID didn't change. But here is a try anyway:
In addition to the keys above we got two double bit changes to work with:
- bit  9+10 -> mask 0xbb
- bit 16+17 -> mask 0x47

If we reasonably assume key 10 is the same as key 26, then key 9 must be 0x86^0xbb=0x3d.
If we then expand the observed pattern (poly 0x31) one bit left/up we suspect key 8 = 0x7a.
When we xor key 8 and 9 we get 0x47. This fits The double bit change at 16+17.

In summary. They tried to use a CRC-8 poly 0x31, but shifted the wrong way and also (accidentally?) reset the key every new byte.
The algo now is: CRC-8 poly 0x98, init 0x7a, for each byte then xor all.
(key stream is 7a 3d 86 43 b9 c4 62 31 repeated 4 times.)

OverSave, 2019-10-12:
someone has already worked on that, does that help?

https://github.com/alex-konshin/f007th-rpi/blob/master/SensorsData.hpp

zuckschwerdt, 2019-10-12:
It's all figured out already, decoder coming in a few days or so :) Uhh, that linked code looks messy ;)

OverSave, 2019-10-12:
Okay xD

klohner, 2020-04-30:
This seems to be working now. Issue closed?
```
$ rtl_433 -f 250k -r rtl_433_tests/tests/auriol/Auriol_HG02832D/g001_433.93M_250k.cu8

time      : @0.285064s
model     : Auriol-HG02832                         id        : 34
channel   : 1            Battery   : 1             Temperature: 15.7 C       Humidity  : 84 %          Button    : 0             Integrity : CRC
```

merbanan, 2020-07-20:
Decoder implemented so closing ticket.

closed 2020-07-20 by merbanan

NilsBe, 2021-04-07:
Great Work! Received my SDR today and installed rtl_433. Could receive the readings from my Auriol-HG02832 sensor. Thank you all!

zuckschwerdt, 2021-04-07:
Good to hear! Makes me think, maybe we need a simple feedback form where users can enter their device with a simple works/doesn't work checkbox /counter. A tally of what's used and working vs. needs work.