merbanan/rtl_433 #1162: Support for LaCrosse TX37U-IT

CharlesGodwin, 2019-10-07:
This is a 915Mhz unit.
I tried this but get no result
`rtl_433 -R 8 -R 34 -R 73 -R 75 -R 76 -R 124 -f 915M`

I also tried with only -R 8 and only-R 76, the most likely candidates. Nothing
I am using on a Raspberry Pi
`rtl_433 version 19.08-23-gdc4440b branch master at 201910051118 inputs file rtl_tcp RTL-SDR SoapySDR`

Any suggestions for what to try next?

CharlesGodwin, 2019-10-07:
Should have read the help files first. https://triq.org/

zuckschwerdt, 2019-10-07:
I don't think we have support specific to this device. If it isn't working with one of the other LaCrosse decoders (better to try all decoders or even `-G`) then upload some sample files as zip here. (Check in https://triq.org/iqs before.) We'll take a look.

CharlesGodwin, 2019-10-07:
I tried and failed to figure out a pull request in https://github.com/merbanan/rtl_433_tests. i'm not that skilled.

The TX37U is described [here ](https://www.lacrossetechnology.com/tx37u-it?type=parts)
The attached zip file was taken when the air temperature was about 13 C
[2019-10-07_0800.zip](https://github.com/merbanan/rtl_433/files/3698079/2019-10-07_0800.zip)

zuckschwerdt, 2019-10-07:
A zip also works, no need for a PR. The signals are present but distorted. Can you try again with a higher sample rate (`-s 1024k`)?

merbanan, 2019-10-07:
We should default a higher sample rate for >433MHz.

zuckschwerdt, 2019-10-07:
The samples vaguely looked like FSK. Is FSK common on 915, like on 868? What about 315?

CharlesGodwin, 2019-10-07:
Here is the new file. current temperature about 16 C
[2019-10-07_1120.zip](https://github.com/merbanan/rtl_433/files/3698185/2019-10-07_1120.zip)

merbanan, 2019-10-07:
@zuckschwerdt FSK is common at 868, 915. 315 probably uses OOK.

zuckschwerdt, 2019-10-07:
That's GFSK (gaussian filtered FSK) at about 33 µs bit width. Looks like minimum shift width -- the 101010… preamble is a neat sine in FM.
Our current demodulator can't automatically pick this up :/

merbanan, 2019-10-07:
:( I still have a few projects I need to finish before I can look into this

CharlesGodwin, 2019-10-07:
Take your time. Thanks for the help

merbanan, 2019-10-07:
@zuckschwerdt I took a look at this signal in I/Q Spectrogram and I see 4 clear levels.

merbanan, 2019-10-07:
Or when I also look at the signal in octave.
[signal_demod.pdf](https://github.com/merbanan/rtl_433/files/3698729/signal_demod.pdf)

merbanan, 2019-10-07:
First the signal preamble uses 2 alternating symbols (mid-hi and mid-low), then you can see that the hi-hi symbol is present and then even lots of low-low symbols.

zuckschwerdt, 2019-10-07:
I think multiple f levels are an artifact. Pulseview shows the PCM as expected:
[image]

merbanan, 2019-10-07:
What sample are you looking at?

merbanan, 2019-10-07:
Hmm, ok that makes more sense, what I see is then a filtering artifact

merbanan, 2019-10-07:
With a tweaked butterworth filter I am able to recover the frequency transitions.

[signal_demod2.pdf](https://github.com/merbanan/rtl_433/files/3699063/signal_demod2.pdf)

So it should be doable. Just need run time switching of filter coeffs or some heuristics that can figure out what is needed.

zuckschwerdt, 2019-10-08:
We already untangled both analyzers from the sample processing and now have a (configurable) choice of four steps for sample data: two analyzers, ook demod, fsk demod. It's possible to add more (optional) steps and choose them on demand/startup. Like the fsk demod we could enable a specific demod only if a decoder needs that.
It'll take some more coding to properly abstract it but that's on my list, and we are nearly there.

CharlesGodwin, 2019-11-02:
Any luck on this?
Thanks

CharlesGodwin, 2019-11-20:
you can close this issue. I've moved on to an alternative solution. Thanks for your  efforts.

gdt, 2023-09-30:
This issue has been abandoned by the submitter, and it doesn't seem like we should leave it open.   Before closing, is there information we should capture, and if so where?

closed 2024-06-23 by gdt