merbanan/rtl_433 #1163: Efergy CT   Not reporting current over 33Amps

camokarzi, 2019-10-12:
Hi,

I have two sets of CT clamps on Solar/Usage side (both 3 phase)   When current surpasses approx 33Amps the output stops reporting as seen below.  I believe this is something in the code.

{"time" : "2019-10-12 13:00:37", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 23.027, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:00:41", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 14.163, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:00:47", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.836, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:00:53", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.472, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:00:59", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.963, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:01", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 31.908, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:07", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 23.591, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:11", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.981, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:13", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 22.921, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:17", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.417, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:23", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.745, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:29", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.454, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:35", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.472, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:41", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.545, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:43", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 23.278, "interval" : 6, "learn" : "NO", "mic" : "CHEC$
{"time" : "2019-10-12 13:01:47", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 13.818, "interval" : 6, "learn" : "NO", "m

zuckschwerdt, 2019-10-12:
Can you record a sample where current is more than 33 Amp? Or easier, add
```
bitbuffer_print(bitbuffer);
```
before [`data_make`](https://github.com/merbanan/rtl_433/blob/master/src/devices/efergy_e2_classic.c#L87) and paste the output bitbuffers.

camokarzi, 2019-10-12:
Okay,
            Let me see if I can do that and get back to you. I did do some further reading and saw that the frequency for the efergy ct should be 433550000 rather than my setting of 433500000   not sure if this would account for my issues.

merbanan, 2019-10-12:
@camokarzi Hitting the frequency exactly will make reception work less well. Counter-intuitive but thats the way the hardware works.

merbanan, 2019-10-12:
https://github.com/merbanan/rtl_433/blob/master/src/devices/efergy_e2_classic.c#L82

That code looks bogus and I don't understand the -(int8_t)bytes[6] + 15; step.

merbanan, 2019-10-12:
https://github.com/merbanan/rtl_433/blob/master/src/devices/efergy_e2_classic.c#L45

That part should use the alignment code. Anyway the fact validation code is most likely what needs changing (or just removed).

merbanan, 2019-10-12:
@camokarzi place the

bitbuffer_print(bitbuffer);

call above https://github.com/merbanan/rtl_433/blob/master/src/devices/efergy_e2_classic.c#L65

That way we should get more readings.

camokarzi, 2019-10-12:
Thanks for the quick response . Okay I will try doing this and see how I go.

camokarzi, 2019-10-13:
Hi,

Thanks

I added the line at line number 85

Re-complied and currently running  on freq   533550000   As you can see below  the system is intermittently reading ID 54534 as via Efergy API I receive correct data that the current is approx  43Amps on my Solar input

Question   where do the logs output to so I may upload to you ?

FYI:  Usage CT  ID :  27507
         Solar CT ID:  54534

time      : 2019-10-13 12:19:22
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.34 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:28
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.56 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:34
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.81 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:40
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.84 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:45
model     : Efergy e2 CT Transmitter ID: 54534
Battery   : OK           Current   : 30.70 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:46
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.38 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:52
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.91 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:19:58
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.73 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:04
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.52 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:10
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 19.13 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:16
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.81 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:22
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.70 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:28
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.59 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:34
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 19.16 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-10-13 12:20:40
model     : Efergy e2 CT Transmitter ID: 27507
Battery   : OK           Current   : 18.98 A       Interval  : 6s            Learning  : NO            Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

zuckschwerdt, 2019-10-13:
Doesn't see to work. Did you forget to copy/install the rtl_433 binary? Otherwise call `/yourpath/rtl_433/build/src/rtl_433`
Anyway, logging won't work this way. Better to add this to line 40 then:
```
decoder_output_bitrowf(decoder, bytes, num_bits, "Efergy-e2CT");
```
Now you can log with either `-F kv:mylog.txt` or `-F json:mylog.txt`.

camokarzi, 2019-10-13:
Think I got it working

Is this the expected output below?   I have switched on the air conditioning on ID 27507   which should read approx 41Amps. Switched on at 2019-10-13 20:28:47 then fails to register an output.

{"time" : "2019-10-13 20:28:41", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bf84dafa308"}
{"time" : "2019-10-13 20:28:41", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 30.786, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:46", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf9308ff758"}
{"time" : "2019-10-13 20:28:46", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:47", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bf808cfade8"}
{"time" : "2019-10-13 20:28:47", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 31.862, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:52", "msg" : "Efergy-e2CT", "codes" : "{64}f3f92abf9308ff75"}
{"time" : "2019-10-13 20:28:52", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:53", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfbfd7f9678"}
{"time" : "2019-10-13 20:28:58", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf9308ff758"}
{"time" : "2019-10-13 20:28:58", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:59", "msg" : "Efergy-e2CT", "codes" : "{64}f38c94bfbf0ef99e"}
{"time" : "2019-10-13 20:29:04", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf9308ff758"}
{"time" : "2019-10-13 20:29:04", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:29:05", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfbdd2f9608"}
{"time" : "2019-10-13 20:29:10", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf9308ff758"}
{"time" : "2019-10-13 20:29:10", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:29:11", "msg" : "Efergy-e2CT", "codes" : "{64}f38c94bfbdbff94d"}
{"time" : "2019-10-13 20:29:16", "msg" : "Efergy-e2CT", "codes" : "{64}f3f92abf9308ff75"}
{"time" : "2019-10-13 20:29:16", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:29:17", "msg" : "Efergy-e2CT", "codes" : "{64}f38c94bfbefef98d"}
{"time" : "2019-10-13 20:29:22", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf9308ff758"}
{"time" : "2019-10-13 20:29:22", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:29:23", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfbedff96e8"}
{"time" : "2019-10-13 20:29:28", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf9308ff758"}
{"time" : "2019-10-13 20:29:28", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.851, "interval" : 6, "learn" : "NO", "mic" : "CHECKS

{"time" : "2019-10-13 20:27:46", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf906fffd98"}
{"time" : "2019-10-13 20:27:46", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.872, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:27:47", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfb447fccf8"}
{"time" : "2019-10-13 20:27:47", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 4.732, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:27:52", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf914cffb78"}
{"time" : "2019-10-13 20:27:52", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.865, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:27:53", "msg" : "Efergy-e2CT", "codes" : "{64}f38c94bfb542fccb"}
{"time" : "2019-10-13 20:27:53", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 4.671, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:27:58", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf914cffb78"}
{"time" : "2019-10-13 20:27:58", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.865, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:27:59", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfb606fc908"}
{"time" : "2019-10-13 20:27:59", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 4.623, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:04", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf914cffb78"}
{"time" : "2019-10-13 20:28:04", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.865, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:05", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfb58dfc168"}
{"time" : "2019-10-13 20:28:05", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 4.653, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:28:10", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf914cffb78"}
[... cut ...]
{"time" : "2019-10-13 20:31:40", "msg" : "Efergy-e2CT", "codes" : "{65}f3f92abf914cffb78"}
{"time" : "2019-10-13 20:31:40", "model" : "Efergy e2 CT", "id" : 54534, "battery" : "OK", "current" : 0.865, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}
{"time" : "2019-10-13 20:31:41", "msg" : "Efergy-e2CT", "codes" : "{65}f38c94bfb0acfc308"}
{"time" : "2019-10-13 20:31:41", "model" : "Efergy e2 CT", "id" : 27507, "battery" : "OK", "current" : 4.958, "interval" : 6, "learn" : "NO", "mic" : "CHECKSUM"}

zuckschwerdt, 2019-10-13:
Should be fixed now (efb5104). The copied codes only have 33 Amps though. If it fails with higher current please again try to grab a code.

camokarzi, 2019-10-13:
Just tested the new code and looks to be working now as I have reported AMPs over 40 now, I will monitor over the next few days but I think it is fixed

A big big thankyou to you for editing the code and your quick response to my issue.

Cheers

Paul

merbanan, 2019-10-31:
No more comments for a while. Closing issue.

closed 2019-10-31 by merbanan