merbanan/rtl_433 #1165: Toyota/Lexus TPMS PA66-GF33

m-d-f, 2019-10-13:
I hereby want to open an issue for another Type of TPMS Sensor used in a 2007 Lexus GS (S19). The Sensor is a Pacific PA66-GF33 or compatible.

In a test the sensors were decoded with Module 88 and temperature and pressure are shown (even though I didn't verifiy if exact) and a 8 digit ID:

model     : Toyota       type      : TPMS          id        : fb662aa9 
status    : 128          pressure_PSI: 35.750      temperature_C: 27.000     mic       : CRC

Point is: the car uses 7 digit IDs, very similiar to the issue:
https://github.com/merbanan/rtl_433/issues/706

Loading only module 110 that was introduced after issue 706 does not yield anything.

Actually I was hoping to be able to read the ID of a newly installed OE-Sensor (HUF UVS4042) which was as a replacement for a faulty original one. My problem was/is: the OE sensor has some kind of ID on it but this does is even different form the 8digit one I can receive.

Again similiar to the issue mentioned I am able to provide the 7digit IDs of my working 3 sensors and I also did capture some CU8-files, even though I'm not sure if I will manage to publish them via a pull request in the test repository because I'm not fit with things git but will try.

m-d-f, 2019-10-13:
I decided to simply upload the sample files here:

[samples.zip](https://github.com/merbanan/rtl_433/files/3721927/samples.zip)

m-d-f, 2019-10-13:
These are samples of the 3 working orginal sensors. The y are registered in the vehilce with th following IDs:
1AE2275	
EE78274	
F0C6274	

Of course i can't tell which sample belongs to Which ID.

m-d-f, 2019-10-13:
I hope some of the pro-maintainers can take a look 👍

zuckschwerdt, 2019-10-13:
Could be that the raw codes (as we output them) have some state bits mixed into them. The registered codes look backwards. The last bits are expected to change while the left-most should be more like a "batch number". So reversing (byte,- nibble-, bit-wise?) is likely needed.
Maybe play with [BitBench](http://triq.net/bitbench?c=fb7001a0&c=fb6fdd3a&c=fb662aa9&c=&c=1AE2275&c=EE78274&c=F0C6274&f=8b%20) and see how the codes/bits relate?

m-d-f, 2019-10-14:
To be honest I was hoping that maybe the same logic as with the other type of Pacific-Sensor would apply but looking more closely, you are right, there are differences. 
Also it's not entirely true that you can't tell which raw codes belong to which registered codes, at least with one pair it is obvious they belong together.
But then it is not really clear for me how to approach this problem on the bitstring level and actually this is something I never had to cope with before. 
Would you try to reverse parts of the bitstring first like [here](http://triq.net/bitbench?c=fb7&c=fb6&c=&c=275&c=274&f=8b%20)?

zuckschwerdt, 2019-10-14:
At least at a quick glance I couldn't tell ;) Which pair did you identify? Maybe there is a scheme how to translate the bits?

m-d-f, 2019-10-14:
I think the raw code beginning with "fb7"  and the registered one ending with "275" likely are a match. Look at my link in my previous post, I isolated the parts I would expect to belong together. Both the raw and the registered are only one number apart hexwise and also at the same digit if you assume that the first and last three digits are flipped.

zuckschwerdt, 2019-10-14:
More data always helps to see a pattern. If there is a pattern. You relate
`fb7001a0`to `1AE2275`, and by extension maybe
`fb6fdd3a`: `EE78274`
`fb662aa9` : `F0C6274`
But it's hard to come up with a probable scheme for that. Anything beyond reordering the bits (reflect bits, bytes, nibbles) and invert is rather more like a crypto.

m-d-f, 2019-10-15:
OK, Plot twist:
Today I found some time to go and see what those TPMs of the 2nd wheel-set in the basement would do. They were easy to wake up and to my astonishment they were recognized as PMV-107J and after a closer look also the ID is properly decoded with a leading zero.

Mhhh, interesting. 
I then proceeded retrying with the 1st set of wheels on my car and repeated my measurement from sunday. This time it was fairly hard to receive something at all but after some trying I found the 1st set of wheels also decoded "correctly" as PMV-107J.

So conclusion is that on Sunday I somehow must have received the wrong TPMSs even though I was standing in the open with other cars at least 25 meters away. For some reason it seems i didn't receive my wheels at all back then. To make it perfect confusion the other car seems to have been a toyota obviously and for some reason it also made sense in my eyes to end up with only three receivable signals (assuming that the new sensor as being a different brand and making being already asleep after some minutes).

So sorry for the confusion and time spent for nothing and thanks for your software. I finally was able to get the missing ID into my ECU and did it "on my own". (reading: not going to a tire shop or car dealer, lol).

Regards

closed 2019-10-15 by m-d-f

merbanan, 2019-10-15:
@m-d-f  " I finally was able to get the missing ID into my ECU and did it "on my own"." 

How was this part done?

m-d-f, 2019-10-15:
This was done via a USB-OCBD Adpater and a copy of the Toyota Techstream software which I already got some weeks ago. I was just in need of the 7-digits ID of the new OE-Sensor on my first set of wheels and that's why I ordered myself a Noelec NESDR to begin with. I could have just driven to a tire-shop as well I guess but this seemed not complicated enough..