merbanan/rtl_433 #1168: Add support for Lidl Auriol AHFL 433B2 IPX4

mnhauke, 2019-10-15:
I just made an PR for the samples:
https://github.com/merbanan/rtl_433_tests/pull/310

```
Test mode active. Reading samples from file: g002_433.92M_250k.cu8
*** signal_start = 23308, signal_end = 252774, signal_len = 229466, pulses_found = 224
Iteration 1. t: 188    min: 175 (161)    max: 201 (63)    delta 360
Iteration 2. t: 186    min: 173 (145)    max: 199 (79)    delta 8
Iteration 3. t: 185    min: 173 (136)    max: 198 (88)    delta 1
Iteration 4. t: 184    min: 172 (131)    max: 197 (93)    delta 2
Iteration 5. t: 184    min: 172 (131)    max: 197 (93)    delta 0
Distance coding: Pulse length 184

Short distance: 499, long distance: 1016, packet distance: 2249

p_limit: 184
bitbuffer:: Number of rows: 10 
[00] {42} 9a 00 6a 8a 4e 40 : 10011010 00000000 01101010 10001010 01001110 01
[01] { 1} 00                : 0
[02] {42} 9a 00 6a 8a 4e 40 : 10011010 00000000 01101010 10001010 01001110 01
[03] { 1} 00                : 0
[04] {42} 9a 00 6a 8a 4e 40 : 10011010 00000000 01101010 10001010 01001110 01
[05] { 1} 00                : 0
[06] {42} 9a 00 6a 8a 4e 40 : 10011010 00000000 01101010 10001010 01001110 01
[07] { 1} 00                : 0
[08] {42} 9a 00 6a 8a 4e 40 : 10011010 00000000 01101010 10001010 01001110 01
[09] { 0}                   : 
Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta -727379968
Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
Distance coding: Pulse length 0

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 0
bitbuffer:: Number of rows: 0 
```

The protocol also seems simple.
See [Bitbench decoding](
http://triq.net/bitbench?c=%7B42%7D%20ad%2010%20db%20a4%2040%2080%20%5B%20CH2%2C%2021.9C%2C%2082%25%20%5D&c=%7B42%7D%209a%2000%206a%208a%204e%2040%20%5B%20CH1%2C%2010.6C%2C%2069%25%20%5D&c=%7B42%7D%204d%200f%2051%2078%204e%2040%20%5B%20CH1%2C%20-8.1C%2C%2060%25%20%5D&c=%7B42%7D%2069%200f%205a%207c%2041%2000%20%5B%20CH1%2C%20-9.0C%2C%2062%25%20%5D&c=%7B42%7D%202f%2020%2088%20a0%204c%2040%20%5B%20CH3%2C%2013%2C6C%2C%2080%25%20%5D&c=%7B42%7D%20ad%2000%20f1%209c%2040%2000%20%5B%20CH1%2C%2024.1C%2C%2078%25%2C%20Batt.%20not%20changed%20%5D&c=%7B42%7D%20ad%2000%20f2%209a%204f%20c0%20%5B%20CH1%2C%2024.2C%2C%2077%25%2C%20Batt.%20not%20changed%20%5D&c=%7B42%7D%20ad%2000%20f2%20a6%204f%2000%20%5B%20CH1%2C%2024.2C%2C%2083%25%2C%20Batt.%20not%20changed%20%5D&f=ID%3A8b%20%3F%3Abb%20CH%3A2d%20TEMP_SIGN%3A4b%20TEMP_C12%3Ad%20HUM%3A7d%20bbbbbbbbbbb%0A%0A)

unknowns for now:
* Low battery indication
* checksum

KamilKopaczyk, 2021-01-14:
I found that this device is very similar to https://github.com/merbanan/rtl_433/blob/master/src/devices/gt_wt_02.c
The comment in `gt_wt_02.c` even mentions following:
> A Lidl AURIO (from 12/2018) with PCB marking YJ-T12 V02 has two extra bits in front.
while this device is using PCB marked as YJ-T22 V02.

The main difference is that each row contains 42 bits instead of 37 or 39, although the row format is almost identical.

To support this new device should separate C file with it's own name or maybe `gt_wt_02.c` file should be adjusted?

As a proof of concept I managed to modify `gt_wt_02.c`to read correct values from `Lidl Auriol AHFL 433B2`, but checksum comparison is broken. Could anybody help with this?
I can provide some more samples if needed.

merbanan, 2021-01-15:
Signal sample are needed for support. And we need lots of messages to figure out the checksum. When we have those we can decide where to put the code.

on1arf, 2021-03-19:
Is this issue still valid?
I have such a device so I can provide samples, if needed

merbanan, 2021-03-19:
Yes, rtl_433 -S unknown and upload them to the ticket or send a PR to the rtl_433_tests repo.

on1arf, 2021-03-19:
ok, I'll leave it running for some time and capure the samples.

Two questions:
- I can connect it to my lab powersupply and lower the voltage to simulate low-battery . Any idea at what voltage the 'battery low' bit should get triggered?

- Am I correct toi assume that the .cu8 files are unsigned int 8 I/Q samples?
(So I just need to convert them to float32 I/Q to load to be able to load the this data in inspectrum?)

merbanan, 2021-03-19:
Inspectrum should handle cu8 files just fine without conversion.

on1arf, 2021-03-19:
61 samples in file.
Notes:
1st sample was recorded after manually pushing the 'trx' button on the device. The other transmissions where initated by the dongle.
Temperature around 24 degrees, humidy around 50 to 60 %

the 'low battery' test not yet done.

[AHFL433.tar.gz](https://github.com/merbanan/rtl_433/files/6172837/AHFL433.tar.gz)

merbanan, 2021-03-19:
> * I can connect it to my lab powersupply and lower the voltage to simulate low-battery . Any idea at what voltage the 'battery low' bit should get triggered?

This depends from device to device.

on1arf, 2021-03-19:
25 more samples:
g001-g017: 1.6 V
g018-g025: 3.0 Volt
[AHFL433_lowbatt.tar.gz](https://github.com/merbanan/rtl_433/files/6173547/AHFL433_lowbatt.tar.gz)

on1arf, 2021-03-20:
did some more experiments with it:
bit 0 to 7 are the address. Changes after a battery rest and sometimes after a manual TX
bit 8 is battery low (0 is ok, 1 is battery-low)
bit 10,11 are channel: ch1: 0 0, ch2: 0 1, ch3: 1 0
I have not managed to get bit 9 to flip.

Pushing the TX button does not make a difference compaired to a transmission initiated by the device
Choosing C or F on the device does not change anything neither (as was already mention in another post by somebody)

I also found that also the bits8 and 10-11 are error-protected. (changing only these bits also changes the error-correction pattern)

I don't know if this helps;
when changing bit 8: (battery low)
0 1 0 0 0 0 0 1 1 0 1 0 0 0 0 0 1 1 1 0 1 1 0 0 0 1 0 1 1 0 0 0 0 1 0 0 1 1 1 0 1 0 
0 1 0 0 0 0 0 1 0 0 1 0 0 0 0 0 1 1 1 0 1 1 0 0 0 1 0 1 1 0 0 0 0 1 0 0 1 1 0 0 1 0 

when changing bit 10: (channel 1 vs 3)
0 0 0 1 0 0 0 0 0 0 0 0 0 0 0 0 1 1 1 0 1 0 0 1 0 1 0 1 1 0 1 0 0 1 0 0 1 0 1 0 1 1 
0 0 0 1 0 0 0 0 0 0 1 0 0 0 0 0 1 1 1 0 1 0 0 1 0 1 0 1 1 0 1 0 0 1 0 0 1 0 1 1 0 1 

when changing bit 11: (channel 1 vs 2)
0 1 1 0 1 1 1 0 0 0 0 0 0 0 0 0 1 1 1 0 1 0 0 0 0 1 0 1 1 0 0 0 0 1 0 0 1 1 1 0 1 1 
0 1 1 0 1 1 1 0 0 0 0 1 0 0 0 0 1 1 1 0 1 0 0 0 0 1 0 1 1 0 0 0 0 1 0 0 1 1 1 1 0 0 

Kr.

merbanan, 2021-03-20:
Can you get some readings with known humidity? The previous samples look bogus.

merbanan, 2021-03-20:
Some samples where you push the button also.

merbanan, 2021-03-20:
Ok, figured out the humidity. TX-button might be bogus. And left is the integrity check. PR https://github.com/merbanan/rtl_433/pull/1683 opened.

merbanan, 2021-03-20:
@on1arf use this bitbench for integrity check work:

https://triq.net/bitbench?c=%7B42%7D%20f2%2000%20eb%206c%2040%2000%20%3A%2011110010%2000000000%2011101011%2001101100%2001000000%2000&c=%7B42%7D%20f2%2000%20ec%206c%2040%2040%20%3A%2011110010%2000000000%2011101100%2001101100%2001000000%2001&c=%7B42%7D%20f2%2000%20ec%206e%2040%20c0%20%3A%2011110010%2000000000%2011101100%2001101110%2001000000%2011&c=%7B42%7D%20f2%2000%20ec%2070%204d%2080%20%3A%2011110010%2000000000%2011101100%2001110000%2001001101%2010&c=%7B42%7D%20f2%2000%20ec%2070%204d%2080%20%3A%2011110010%2000000000%2011101100%2001110000%2001001101%2010&c=%7B42%7D%20f2%2000%20ec%2072%204e%2000%20%3A%2011110010%2000000000%2011101100%2001110010%2001001110%2000&c=%7B42%7D%20f2%2000%20ec%2072%204e%2000%20%3A%2011110010%2000000000%2011101100%2001110010%2001001110%2000&c=%7B42%7D%20f2%2000%20ec%2074%204e%2080%20%3A%2011110010%2000000000%2011101100%2001110100%2001001110%2010&c=%7B42%7D%20f2%2000%20ec%2076%204f%2000%20%3A%2011110010%2000000000%2011101100%2001110110%2001001111%2000&c=%7B42%7D%20f2%2000%20ef%207c%2041%2040%20%3A%2011110010%2000000000%2011101111%2001111100%2001000001%2001&c=%7B42%7D%2078%2000%20e7%205e%204e%20c0%20%3A%2001111000%2000000000%2011100111%2001011110%2001001110%2011&c=%7B42%7D%2078%2000%20e8%205c%204e%2080%20%3A%2001111000%2000000000%2011101000%2001011100%2001001110%2010&c=%7B42%7D%2078%2000%20ea%205c%204f%2000%20%3A%2001111000%2000000000%2011101010%2001011100%2001001111%2000&c=%7B42%7D%2078%2080%20ec%205c%2041%2080%20%3A%2001111000%2010000000%2011101100%2001011100%2001000001%2010&c=%7B42%7D%2078%2080%20ed%205c%2041%20c0%20%3A%2001111000%2010000000%2011101101%2001011100%2001000001%2011&c=%7B42%7D%2078%2080%20ee%205c%2042%2000%20%3A%2001111000%2010000000%2011101110%2001011100%2001000010%2000&c=&c=%7B42%7D%20ad%2010%20db%20a4%2040%2080%20%5B%20CH2%2C%2021.9C%2C%2082%25%20%5D&c=%7B42%7D%209a%2000%206a%208a%204e%2040%20%5B%20CH1%2C%2010.6C%2C%2069%25%20%5D&c=%7B42%7D%204d%200f%2051%2078%204e%2040%20%5B%20CH1%2C%20-8.1C%2C%2060%25%20%5D&c=%7B42%7D%2069%200f%205a%207c%2041%2000%20%5B%20CH1%2C%20-9.0C%2C%2062%25%20%5D&c=%7B42%7D%202f%2020%2088%20a0%204c%2040%20%5B%20CH3%2C%2013%2C6C%2C%2080%25%20%5D&c=%7B42%7D%20ad%2000%20f1%209c%2040%2000%20%5B%20CH1%2C%2024.1C%2C%2078%25%2C%20Batt.%20not%20changed%20%5D&c=%7B42%7D%20ad%2000%20f2%209a%204f%20c0%20%5B%20CH1%2C%2024.2C%2C%2077%25%2C%20Batt.%20not%20changed%20%5D&c=%7B42%7D%20ad%2000%20f2%20a6%204f%2000%20%5B%20CH1%2C%2024.2C%2C%2083%25%2C%20Batt.%20not%20changed%20%5D&f=ID%3A8b%20%3F%3Abb%20CH%3A2d%20TEMP_C%3A12d%20HUM%3A7d%20b%20FIX%3Abbbb%20CHK%3Abbbbbb&cw=4

merbanan, 2021-03-20:
@on1arf more messages would be nice, you have use the PR and run with:

rtl_433 2>log.txt
and the use:
cat log.txt | grep {42} | cut -b 6- | uniq | sort

to get nice output for bitbench.

on1arf, 2021-03-20:
that's what I am currently doing .. The device is currently outside so to have more different values while I leave it there for a night. (greater variation of temperature)
Tomorrow, I will make a small setup to automically do a power-off / power=on of the device  a few hunderd times,  so to get a number of messages with all different device-ids (bits 0 to 7)

merbanan, 2021-03-20:
It look like it was 6bit nibble sum.

merbanan, 2021-03-20:
Closed with 090f116.

closed 2021-03-20 by merbanan

on1arf, 2021-03-20:
Hi,

I  can confirm that it works!

 Thanks for the work.