merbanan/rtl_433 #1174: Bug in elantra tpms decoder

Mindavi, 2019-10-23:
Reproduced by: `rtl_433 -y '{480} 9c 55 5c 55 5c 55 5c 55 5c 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 55 c5 55 c5 55 c5 55 c5 55 c5 55 c5 55 c5 55 c5 55 c5 55 c5 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00'`.

The problem is that the decoder counts all the events together and returns that as 1 value. This shouldn't be a problem if 0 is nothing found and 1 is something found, but the new return values count up as a big negative number. This means that the elantra tpms is doing it wrong, but the events accounting is also not defensive enough.

zuckschwerdt, 2019-10-24:
I also want to simplify that two stage pattern (pattern find and then calling the real decoder).

zuckschwerdt, 2019-10-24:
`exit(1)` would also be fine, it's a condition that must not happen.

closed 2019-10-25 by Mindavi