merbanan/rtl_433 #1178: Issue with bitbuffer parsing

Mindavi, 2019-10-24:
This is an issue that (I'd guess) would never occur in the wild or could be crafted, but I bumped into it by accident.

Run `rtl_433 -y @test2.bb` to reproduce.
[test.zip](https://github.com/merbanan/rtl_433/files/3769016/test.zip)

```
==31658==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffc8da0cb20 at pc 0x000000512a18 bp 0x7ffc8da023d0 sp 0x7ffc8da023c8
READ of size 1 at 0x7ffc8da0cb20 thread T0
    #0 0x512a17 in bitbuffer_parse /home/rick/hobby/rtl-sdr/433mhz/rtl_433/src/bitbuffer.c:328:20
    #1 0x648bd0 in pulse_demod_string /home/rick/hobby/rtl-sdr/433mhz/rtl_433/src/pulse_demod.c:575:5
    #2 0x693454 in main /home/rick/hobby/rtl-sdr/433mhz/rtl_433/src/rtl_433.c:1258:22
    #3 0x7fcd84a1bb96 in __libc_start_main /build/glibc-OTsEL5/glibc-2.27/csu/../csu/libc-start.c:310
    #4 0x41c119 in _start (/usr/local/bin/rtl_433+0x41c119)

Address 0x7ffc8da0cb20 is located in stack of thread T0 at offset 34656 in frame
    #0 0x68da3f in main /home/rick/hobby/rtl-sdr/433mhz/rtl_433/src/rtl_433.c:1066

  This frame has 4 object(s):
    [32, 184) 'sigact' (line 1068)
    [256, 26200) 'r_devices' (line 1089)
    [26464, 34656) 'line' (line 1209) <== Memory access at offset 34656 overflows this variable
    [34912, 34920) 'e' (line 1229)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism, swapcontext or vfork
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow /home/rick/hobby/rtl-sdr/433mhz/rtl_433/src/bitbuffer.c:328:20 in bitbuffer_parse
Shadow bytes around the buggy address:
  0x100011b39910: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100011b39920: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100011b39930: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100011b39940: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100011b39950: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x100011b39960: 00 00 00 00[f2]f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2
  0x100011b39970: f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2 f2
  0x100011b39980: f2 f2 f2 f2 f8 f3 f3 f3 00 00 00 00 00 00 00 00
  0x100011b39990: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100011b399a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x100011b399b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==31658==ABORTING
```

zuckschwerdt, 2019-10-24:
Oops. I introduced that bug when adding the `@file` feature. Of course the null termination isn't guaranteed there.

zuckschwerdt, 2019-10-24:
Ah, wait, it actually is guaranteed by `fgets`, strange. Can't reproduce. Can you change line 1209 to `char line[INPUT_LINE_MAX] = {0};` and retest?

Mindavi, 2019-10-24:
Mm, I only saw it when the file was exactly 20.000 bytes long, so that was weird as well. I'll retest it with that change applied.

Mindavi, 2019-10-24:
@zuckschwerdt from which file?

Mindavi, 2019-10-24:
Found it, but that does not fix the problem. It still gives an error.

zuckschwerdt, 2019-10-24:
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L1209 -- I'm wondering if your fgets doesn't add the terminating `0`. Which libc is that?

zuckschwerdt, 2019-10-24:
Ok, does adding `line[INPUT_LINE_MAX - 1] = 0;` below the while-line 1223 help?

Mindavi, 2019-10-24:
Nope, that also does not seem to solve anything. Do you have ASAN enabled?

zuckschwerdt, 2019-10-24:
I can reproduce with glibc now. I'll dig deeper.

zuckschwerdt, 2019-10-24:
Got it. Parsing a length with no closing braces but instead end-of-string had the parsing skip the end. Nice puzzle, thanks!

closed 2019-10-24 by zuckschwerdt