merbanan/rtl_433 #1179: Division by zero in rtl_433/src/r_api.c:227:57

Mindavi, 2019-10-24:
[division_by_zero_samples.zip](https://github.com/merbanan/rtl_433/files/3769136/division_by_zero_samples.zip)

All these samples trigger a division by 0 in `r_api.c:227`.

```
rtl_433/src/r_api.c:227:57: runtime error: division by zero
SUMMARY: UndefinedBehaviorSanitizer: undefined-behavior rtl_433/src/r_api.c:227:57 in 
```

In `pulse_detect.c:423`, the -1 is probably introduced. Not sure how to fix this, besides the obvious check that could be introduced. I'm not sure what to do in that case however.

merbanan, 2019-10-24:
pulse_detect.c:423 calculates the estimate for the noise in a signal. While this is a synthetic signal (guessing) once in a billion years this will be triggered by a real signal. One fix could be that the estimator never could be less then zero (ie value clamping). If I understand things correctly this code uses some arbitrary logic to get the numbers in the right magnitude range.

Adding
s->ook_low_estimate = ((s->ook_low_estimate > 0) ? s->ook_low_estimate : 0);

Should fix the issue.

zuckschwerdt, 2019-10-24:
That's a really lucky hit there. I have a random +1 to counter div-0 and log(0) but getting a -1 is bad luck then. I guess I'll clamp both.

zuckschwerdt, 2019-10-24:
Fixed with clamping to `>0`.

closed 2019-10-24 by zuckschwerdt