merbanan/rtl_433 #1183: static code analysis

Mindavi, 2019-10-29:
Reminder for myself (or other that want to pick this up) that static analysis might help improve the quality of this code and find potential bugs. A quick run of cppcheck shows some possible places where problems might occur. Cppcheck does not need any (serious) setup, so can be easily run.

`cppcheck . --force --enable=all --std=c99 -i src/mongoose.c --xml --output-file=cppcheck.xml`
I deliberately ignore `mongoose` here, because it in itself generates some warnings we might not be that interested in. However, mongoose static checking gave one memory leak bug as well so it might be good to patch that upstream.

In the cppheck repository there is a tool to generate html output, See [the cppcheck repository](https://github.com/danmar/cppcheck). This may be handy for a quick view of the files and results.

Attached is such an html report (for a slightly older version, not master).
[cppcheck-report.zip](https://github.com/merbanan/rtl_433/files/3782773/cppcheck-report.zip)

merbanan, 2019-10-29:
Can you try CSA also ? I remember playing with that a few years ago. And maybe how to activate Coverity for the repo also.

merbanan, 2019-10-29:
I took a quick look into the report and it seems like there are a some real errors that needs to be fixed.

zuckschwerdt, 2019-10-29:
Some good hints in there. I'll fix the worst offenders asap.
Especially the out-of-scope pointer hint is valuable, it's just unspecified behaviour and works, but who knows if the compiler wants to reuse out-of-scope stack someday.

mnhauke, 2019-10-29:
> Can you try CSA also ? I remember playing with that a few years ago. And maybe how to activate Coverity for the repo also.

There's also some info available on [lgtm.com](https://lgtm.com/) (a service run by the company semmle (now part of github)) where merbanan/rtl_433 is listed since some time ...
https://lgtm.com/projects/g/merbanan/rtl_433/alerts/?mode=list

zuckschwerdt, 2019-10-29:
Indeed, lgtm caught two logic bugs the others didn't. I've started to address some warnings, but this will take some time to get right. The worst bugs should be fixed.

zuckschwerdt, 2019-11-03:
There is now CSA checker support in the CMake files.
To create a debug build and run full tests with the static analyzer:

    cmake -DBUILD_TESTING_ANALYZER=1 -DCMAKE_BUILD_TYPE=Debug .. && make CTEST_OUTPUT_ON_FAILURE=1 all test

Mindavi, 2020-01-31:
There's currently one warning open on lgtm. I've also seen some warnings about mongoose (😞 ) while building with hardened mode on NixOs. Will share these/PR when I have time.

gdt, 2023-09-25:
What's the status of this PR and the way forward?  Seems closable, but I'm not sure.

zuckschwerdt, 2023-09-26:
Yes, CSA is included and run on the CI.