merbanan/rtl_433 #1184: Missing header file when building?

ADeeds, 2019-10-31:
Hi,

I'm trying to build rtl_433 on my raspberry pi. When I run make, I get this error:

pulse_detect.c:14:10: fatal error: pulse_detect_fsk.h: No such file or directory
 #include "pulse_detect_fsk.h"

It doesn't look like that file exists in the repo. Am I missing some step that generates it?

zuckschwerdt, 2019-10-31:
Thanks for the heads-up. Sorry, the current master is broken by a recent commit. If you use Git please checkout 0f5e21c until this is fixed.

@merbanan pulse_detect_fsk.h and pulse_detect_fsk.c are missing from a784c2d

ADeeds, 2019-10-31:
No worries. I'll use the most recent tagged version, which is probably what I should've done from the start since I don't need any bleeding edge features :)

zuckschwerdt, 2019-10-31:
I've also reset the offending commits master HEAD is usable again. Good to use!
@merbanan the reset commits are retained in `master-new`.

merbanan, 2019-10-31:
I'll fix when I get home. I did test things but yeah the files exist for me so.

merbanan, 2019-10-31:
Fixed and compile tested.

closed 2019-10-31 by merbanan