merbanan/rtl_433 #1195: Flex decoder output not published to MQTT

Pihkal82, 2019-11-09:
Hi,

I'm using the flex decoder to capture the RF signal from my smoke alarms.
I've managed to setup the decoder just fine and it's working perfectly, until I try to publish it to MQTT.

My setup:

rtl_433 -F "mqtt://192.168.x.x:1883,user=xxx,pass=xxx,retain=0,devices=sensors/rtl_433" -X 'n=SMOKE_ALARM,m=OOK_PWM,s=868,l=1892,r=4152,g=0,t=0,y=3784,unique,rows=25,bits=18,match={18}0xfb77c'

I'm watching the topic sensors (with wild card) and get no data when I trigger the smoke alarm.
When I press the wireless doorbell the data of the doorbell gets published just fine.

If I change the above command to:

rtl_433 -F json -X 'n=SMOKE_ALARM,m=OOK_PWM,s=868,l=1892,r=4152,g=0,t=0,y=3784,unique,rows=25,bits=18,match={18}0xfb77c'

I consistently get the output below, as is expected.

time      : 2019-11-09 18:09:32
model     : SMOKE_ALARM  count     : 24            num_rows  : 25            len       : 18            data      : fb77c

I do notice when I press the doorbell only time and id gets published to MQTT, no model or data.
Since the smoke alarm doesn't have an id I'm expecting this to be an issue, but it still should be publishing time.

I'm a bit stuck now, am I doing something wrong or is it a bug?
Many thanks!

Kind regards,
Kenneth

zuckschwerdt, 2019-11-09:
First up: `g=0,t=0` are the defaults (disabled), you shouldn't list those.

Are you observing the right topic? Your config would give e.g. `sensors/rtl_433/time`. For a custom topic you should perhaps use `"devices=sensors/rtl_433[/model]"` or better the full `"devices=sensors/rtl_433[/type][/model][/subtype][/channel][/id]"` which will then also work with other sensors that have an id and/or channel etc.

zuckschwerdt, 2019-11-09:
S.a. https://triq.org/rtl_433/OPERATION.html#mqtt-format-strings for info on the topic format strings.

Pihkal82, 2019-11-09:
Hi,

adding "devices=sensors/rtl_433[/type][/model][/subtype][/channel][/id" to the command does make the flex decoder output show on MQTT.
Tested it with a modified flex decoder that triggers on almost anything, its data is showing.
I can't test the smoke alarms now as the kids are already in bed.
I'll update tomorrow but so far it's looking good.

Many thanks for your in input!

Kind regards

zuckschwerdt, 2019-11-09:
If you trigger the alarm once and record the signal with `-S all` you can play it back if you need to change the configuration. Might be easier and with less noise.

Pihkal82, 2019-11-10:
Thanks for the tip. :-)
I've just tested it with the smoke alarms and it's now working fine!
Many thanks for your help and for the excellent software.

Kind regards

closed 2023-09-25 by gdt