merbanan/rtl_433 #1196: Elster Power Meter

gcormier, 2019-11-10:
Someone has done quite a bit of work in GNU Radio to decode the packets from [Elster power meters](https://github.com/argilo/gr-elster) I'm no GNU Radio expert, but if I provide the sample captures would anyone be willing to write the decoder?

zuckschwerdt, 2019-11-10:
I don't expect the actual coding to be a problem. Can we get information on the protocol? There is something about frequency-hopping in there? Is that part of the protocol or just "alternative settings" to reduce device overlap?

zuckschwerdt, 2019-11-10:
I take it that https://github.com/argilo/gr-elster/blob/master/python/packetize.py has the main logic part. Doesn't look to involved.

gcormier, 2019-11-10:
Ok great. I picked up a LimeSDR mini, so I will do some wide band captures to see if it's actually hopping, or if it just repeats it across a few.

gcormier, 2020-01-06:
I looked up the FCC information on this one. It looks like the device is dumb, and it meets compliance merely by rotating through the 25 channels (https://fccid.io/QZC-R2EA/Test-Report/Test-Report-880526)

[image]

[image]

If I'm reading this correct the channels are ~397kHz apart. With 25 channels that's only one half of 902-928MHz. Pages 22-24 of the FCC doc talk about channels 0 through 63. So I'm not clear on what's going on.

I'll run a capture overnight on 902.40MHz. `rtl_433 -f 902.4M -d "" -g "" -s 1M -S unknown`

gcormier, 2020-01-06:
Found [this tidbit](https://www.reddit.com/r/RTLSDR/comments/7pgaga/how_to_demodulate_fcc_id_qzcr2ea_smart_meter/) on reddit as well,

`.. they are FSK modulated, 400khz channel spacing over 25 channels at either 35.5kbps or 142.2kbps data rate.`

gcormier, 2020-01-06:
I got around 900 packets overnight.

If we assume the above note is accurate on rates, we're looking for 28us or 7us pulses, which I don't see (except for three with a single 28us packet) from running `rtl_433 -r *` and grep'ing the output.

I could use some help on how to proceed.

(One of three useless single count 28us..)
````
Detected FSK package	@0.131516s
Analyzing pulses...
Total count:   30,  width: 5.27 ms		( 1318 S)
Pulse width distribution:
 [ 0] count:   24,  width:   40 us [40;48]	(  10 S)
 [ 1] count:    1,  width:  244 us [244;244]	(  61 S)
 [ 2] count:    2,  width:   84 us [80;88]	(  21 S)
 [ 3] count:    2,  width:  168 us [168;172]	(  42 S)
 [ 4] count:    1,  width:   28 us [28;28]	(   7 S)
Gap width distribution:
 [ 0] count:    2,  width:  660 us [616;704]	( 165 S)
 [ 1] count:   15,  width:   40 us [40;44]	(  10 S)
 [ 2] count:    4,  width:   76 us [76;88]	(  19 S)
 [ 3] count:    5,  width:  120 us [120;132]	(  30 S)
 [ 4] count:    2,  width:  204 us [204;204]	(  51 S)
 [ 5] count:    1,  width:  160 us [160;160]	(  40 S)
Pulse period distribution:
 [ 0] count:    2,  width:  704 us [664;748]	( 176 S)
 [ 1] count:   15,  width:   80 us [80;84]	(  20 S)
 [ 2] count:    3,  width:  332 us [292;376]	(  83 S)
 [ 3] count:    4,  width:  220 us [208;252]	(  55 S)
 [ 4] count:    3,  width:  120 us [120;124]	(  30 S)
 [ 5] count:    2,  width:  168 us [168;168]	(  42 S)
Level estimates [high, low]:   2050,    181
RSSI: -18.1 dB SNR: 21.1 dB Noise: -39.1 dB
Frequency offsets [F1, F2]:   -5282, -14762	(-20.1 kHz, -56.3 kHz)
Guessing modulation: No clue...
````

georgejung, 2021-10-19:
I'm interested in this. I just got my RTL-SDR yesterday and have found my water meter. Next is my electric meter which is an Elster R2S. Any updates?

merbanan, 2021-10-19:
We need signal recordings for this type of transmitter. The ones we got are of unknown quality.

@argilo the demod in rtl_433 is generic so it will be able to handle different frequency offsets in the tuned range. I do guess that it would only be able to handle 2-3 frequencies compared to the 6 that gr-elster can handle.

rtl_433 -f 902.4M -s 1M -S unknown should be able to record some signals. If you don't get anything of interest try adjusting the frequency and/or increase the sample rate.

rexbron, 2022-09-24:
I would also like to contribute to this. Has anyone look at https://github.com/shaunhey/ea_receiver

I seem to be able to use it to get the packets but the system doesn't decode them.

abstractvector, 2022-09-26:
We've just had a brand new meter installed and would like to help. Our meter is a Honeywell meter, but uses an Elster PCB I believe. At least, it's labeled as `TYPE R2SD` and says it contains `FCC ID: QZC-RX2EA4F` which is the Elster PCB: https://fccid.io/QZC-RX2EA4F

Where we're located we shouldn't have much, if any, interference on the 900MHz band and I have an antenna just a few feet from the meter so should be able to get clean signals.

cliv, 2022-09-27:
Looks like i might have a couple of interesting captures.

https://triq.org/pdv/#AAB0000401001C0038002500F2808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080828080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808082808080808280808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080908080808080808080808080808080808080808080808080808080808080808155+AAB00F0401001C0038002500F2808082A08155+AAB00C0401001C0038002500F2809155+AAB00E0401001C0038002500F29080808155+AAB00B0402001C0038002500F29155+AAB00F0401001C0038002500F2808080908155+AAB00B0408001C0038002500F29155+AAB00F0401001C0038002500F2808090808155+AAB00C0401001C0038002500F2809155+AAB00D0401001C0038002500F290808155+AAB0110401001C0038002500F28080809080808155+AAB00C0401001C0038002500F2908155+AAB00B0404001C0038002500F29155+AAB00F0401001C0038002500F2808090808155+AAB00B0401001C0038002500F29155+AAB00C0401001C0038002500F2908155+AAB00C0401001C0038002500F2809155+AAB00B0402001C0038002500F29155+AAB0110401001C0038002500F28080809080808155+AAB00D0402001C0038002500F280809155+AAB00B040B001C0038002500F29155+AAB00C0401001C0038002500F2908155+AAB00B0402001C0038002500F29155+AAB00C0401001C0038002500F2809155+AAB00D0401001C0038002500F290808155+AAB00C0401001C0038002500F2908155+AAB0100401001C0038002500F280908080808155+AAB00B0410001C0038002500F29155+AAB0110401001C0038002500F29080808080808155+AAB00F0401001C0038002500F2808090808155+AAB0100401001C0038002500F280808090808155+AAB00E0401001C0038002500F28080809155
https://triq.org/pdv/#AAB0000501001C000F0038002900F2808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808390808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808390A08080808080808080808080808080808080808080808080808080808080808255+AAB0110501001C000F0038002900F2808080A08255+AAB00E0501001C000F0038002900F280A255+AAB0100501001C000F0038002900F2A080808255+AAB00D0502001C000F0038002900F2A255+AAB0110501001C000F0038002900F2808080A08255+AAB00D0508001C000F0038002900F2A255+AAB0110501001C000F0038002900F28080A0808255+AAB00E0501001C000F0038002900F280A255+AAB0100501001C000F0038002900F2A080808255+AAB00D0501001C000F0038002900F2A255+AAB00E0502001C000F0038002900F280A255+AAB00F0501001C000F0038002900F28080A255+AAB00D0502001C000F0038002900F2A255+AAB00E0501001C000F0038002900F2A08255+AAB00F0501001C000F0038002900F2A0808255+AAB00F0501001C000F0038002900F280A08255+AAB00E0501001C000F0038002900F2A08255+AAB00F0501001C000F0038002900F2A0808255+AAB00E0501001C000F0038002900F2A08255+AAB00D0501001C000F0038002900F2A255+AAB0100501001C000F0038002900F28080A08255+AAB00F0502001C000F0038002900F28080A255+AAB00D050B001C000F0038002900F2A255+AAB00E0501001C000F0038002900F2A08255+AAB00D0502001C000F0038002900F2A255+AAB00E0501001C000F0038002900F280A255+AAB00F0501001C000F0038002900F2A0808255+AAB00E0501001C000F0038002900F2A08255+AAB0120501001C000F0038002900F280A08080808255+AAB00D0510001C000F0038002900F2A255+AAB0190501001C000F0038002900F2A080808080808080808080808255
https://triq.org/pdv/#AAB0000501002300370015000F000C82828282828282828282828282828282828282828282828282828282828282828282828282828282828382828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828283828282828282828282828282828282828282828282838282828282828282828282828282828282828282828282828282838283828282828282828282838282828382828282828282828384838282838282828282828282828283828282838282828282828282828282828282828382828282828282828282828283828282828282828282828282828282828283838282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828282828 [... cut ...] 0808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080A455
https://triq.org/pdv/#AAB1060000001C00150038002300008191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191919191A4919191919194A19191B19191919191919191919191A055

abstractvector, 2022-09-28:
I've been capturing this morning with `rtl_433 -f 902.4M -s 1M -S unknown` and the only interesting one so far is this:

* [g074_902.4M_1000k.cu8 (133585 samples, 393216 bytes)] [#1](https://triq.org/pdv/#AAB0000801001C00370014000A005500EB000E0024808080808080808080808080808080808080808080808080808080808080808080808080908080808080808080828080808080808080808080808080928080808080808080808080808080808080808080808080908090808080938690808080808080808080808080808080808080808080808080808080809280808080928080808080808092808080808080808080808080808080808080808080808090808080968080808080808080809080808080808083908080808080808080829080808080808080808080808080808080808080809680808080808080808082808080808096808080808082808080808080808080808080808080808080808080928080808080928080808080808082809680809682808080808080808080968080808080808080808080809080808080808082808080808080808082808080808080808080808080808080808080808080809280808080808080808080928082808080808080808080808080808080808080809280808080808080808080808092808080928080808080808080809380808080808080808080808080808080808080808080808080808080808080808080828090808092928096808080808080808080808080808080808080808080808080808080808080808080808080808082808080808080808080808082909380808080808080928080809280808080828080808082808080808080808080808080808080809280809680808080808080808155+AAB0350801001C00370014000A005500EB000E0024808080808080808080808080808086808080808080808080808080808280809080808155+AAB0150801001C00370014000A005500EB000E002492908155+AAB0130801001C00370014000A005500EB000E00249155+AAB0160801001C00370014000A005500EB000E00248080809155+AAB0130801001C00370014000A005500EB000E00249155+AAB0140801001C00370014000A005500EB000E0024809155+AAB0160801001C00370014000A005500EB000E00248080908155+AAB0130806001C00370014000A005500EB000E00249155+AAB0150801001C00370014000A005500EB000E002490808155+AAB0140801001C00370014000A005500EB000E0024809155+AAB0170801001C00370014000A005500EB000E0024809080808155+AAB01B0801001C00370014000A005500EB000E002490808080808080809155+AAB0140801001C00370014000A005500EB000E0024908155+AAB0130804001C00370014000A005500EB000E00249155+AAB0150801001C00370014000A005500EB000E002480809155+AAB0130801001C00370014000A005500EB000E00249155+AAB0160801001C00370014000A005500EB000E00248080908155+AAB0140801001C00370014000A005500EB000E0024908155+AAB0140801001C00370014000A005500EB000E0024809155+AAB0150801001C00370014000A005500EB000E002480809155+AAB0130801001C00370014000A005500EB000E00249155+AAB0140802001C00370014000A005500EB000E0024908155+AAB0160801001C00370014000A005500EB000E00248080908155+AAB0130814001C00370014000A005500EB000E00249155+AAB0190801001C00370014000A005500EB000E00248080808080809155+AAB0130801001C00370014000A005500EB000E00249155+AAB0150801001C00370014000A005500EB000E002490808155+AAB0130802001C00370014000A005500EB000E00249155+AAB0150801001C00370014000A005500EB000E002480809155+AAB0130807001C00370014000A005500EB000E00249155+AAB0160801001C00370014000A005500EB000E00248080908155)

gdt, 2023-09-30:
What's the status and path forward?  It looks like a fair bit of progress -- a PR would be a helpful step.

zuckschwerdt, 2024-08-01:
Elster FCC docs for QZC-REXUANZ (https://fcc.report/FCC-ID/QZC-REXUANZ) seem to indicate the same Frequency Hopping Spread Spectrum at 902.4 - 927.6 MHz.
You should be able to catch and investigate transmissions.

velaar, 2024-09-01:
> Elster FCC docs for QZC-REXUANZ (https://fcc.report/FCC-ID/QZC-REXUANZ) seem to indicate the same Frequency Hopping Spread Spectrum at 902.4 - 927.6 MHz. You should be able to catch and investigate transmissions.

Same can be said for  [QZC-GNIC](https://fcc.report/FCC-ID/QZC-GNIC/4602192.pdf) that just got installed by Enmax in Calgary / Alberta / Canada. 

As per Honeywell's roadmap all A4RES meters are using what they call REXU protocol [Source](https://www.meuw.org/files/Honeywell%20Roadmap%20-%20%20GB%20Meter%20School%20'23.pdf). The only mention of REXU I found was [here ](https://process.honeywell.com/us/en/products/utilities/electricity/residential-meters/rexuniversal-meter) and it clearly suggests that it is the same protocol that is used by elster.

greyltc, 2025-08-16:
Maybe I just got a good one? [triq.org link](https://triq.org/pdv/#AAB103001C0038072080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808090808080808080808080808080808080808080808080808080808080808080818080809081809190808081919080819080819190808191919191908190808190808081808080809080819081809191919180809180918080908081809190808180918090808190819081919191919191919180908080818090808180918080808090808081809191919191919191919191919191919191919191919191919191919191919191919191919191919191918091809190808191908081808091919191919191919180809191919081809080808190819081808090808080818090818080918255)
```
$ rtl_433 -f 902.4M -s 2M -S unknown
rtl_433 version 25.02 (2025-02-19) inputs file rtl_tcp RTL-SDR SoapySDR with TLS

New defaults active, use "-Y classic -s 250k" if you need the old defaults

Found Rafael Micro R820T tuner
[SDR] Using device 0: Nooelec, NESDR SMArt v5, SN: REDACTED, "Generic RTL2832U OEM"
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
*** Saving signal to file g001_902.4M_2000k.cu8 (350479 samples, 786432 bytes)
*** Saving signal to file g002_902.4M_2000k.cu8 (203791 samples, 524288 bytes)

*** Saving signal to file g003_902.4M_2000k.cu8 (77871 samples, 262144 bytes)
^CSignal caught, exiting!
$ rtl_433 -A g002_902.4M_2000k.cu8 
rtl_433 version 25.02 (2025-02-19) inputs file rtl_tcp RTL-SDR SoapySDR with TLS
[Input] Test mode active. Reading samples from file: g002_902.4M_2000k.cu8
Detected FSK package    @0.037368s
Analyzing pulses...
Total count:  780,  width: 49.98 ms             (99958 S)
Pulse width distribution:
 [ 0] count:  671,  width:   28 us [28;28]      (  55 S)
 [ 1] count:  109,  width:   56 us [56;56]      ( 112 S)
Gap width distribution:
 [ 0] count:  670,  width:   28 us [28;28]      (  56 S)
 [ 1] count:  109,  width:   56 us [56;56]      ( 112 S)
Pulse period distribution:
 [ 0] count:  643,  width:   56 us [56;56]      ( 112 S)
 [ 1] count:   54,  width:   84 us [84;84]      ( 168 S)
 [ 2] count:   82,  width:  112 us [112;113]    ( 225 S)
Pulse timing distribution:
 [ 0] count: 1341,  width:   28 us [28;28]      (  56 S)
 [ 1] count:  218,  width:   56 us [56;56]      ( 112 S)
 [ 2] count:    1,  width: 1824 us [1824;1824]  (3648 S)
Level estimates [high, low]:  15978,    640
RSSI: -0.1 dB SNR: 14.0 dB Noise: -14.1 dB
Frequency offsets [F1, F2]:   17032,   6541     (+519.8 kHz, +199.6 kHz)
Guessing modulation: Manchester coding
view at https://triq.org/pdv/#AAB103001C0038072080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808080808090808080808080808080808080808080808080808080808080808080808080818080809081809190808081919080819080819190808191919191908190808190808081808080809080819081809191919180809180918080908081809190808180918090808190819081919191919191919180908080818090808180918080808090808081809191919191919191919191919191919191919191919191919191919191919191919191919191919191918091809190808191908081808091919191919191919180809191919081809080808190819081808090808080818090818080918255
Attempting demodulation... short_width: 28, long_width: 0, reset_limit: 57, sync_width: 0
Unsupported
```