merbanan/rtl_433 #1198: Version 19.08-100-g9ad8c83 branch master at 201911091711 Input Changed/Missing

gary-hammer, 2019-11-15:
I've had an Ambient Weather/FineOffset WH25 and WH65B working well for about a year.
Today, I was adding a WH31E so updated my rtl_433 and rtl_sdr install via clone git and make/build.

After I am unable to see the WH25 using the same command as before. I am unable to see the WH25 running rtl_433 with all default decoders enabled.
Here is an example of the output. I note the WH24 is not in or around my location. Also, the pulse_FSK_detect() I have not seen before.
I have checked the USB, cable for the USB, restarted the server. Nothing changes.

```
rtl_433 -M time:utc -F json -f 914980000 -R -104

rtl_433 version 19.08-100-g9ad8c83 branch master at 201911091711 inputs file rtl_tcp RTL-SDR

Registered 111 out of 141 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-141 ]

Found Rafael Micro R820T tuner

Exact sample rate is: 1000000.026491 Hz

[R82XX] PLL not locked!

Sample rate set to 1000000 S/s.

Tuner gain set to Auto.

Tuned to 914.980MHz.

{"time" : "2019-11-15 02:29:28", "model" : "Fine Offset WH65B", "id" : 172, "battery" : "OK", "temperature_C" : -1.400, "humidity" : 79, "wind_dir_deg" : 97, "wind_speed_ms" : 0.191, "gust_speed_ms" : 0.510, "rainfall_mm" : 2295.144, "uv" : 0, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}

{"time" : "2019-11-15 02:30:00", "model" : "Fine Offset WH65B", "id" : 172, "battery" : "OK", "temperature_C" : -1.400, "humidity" : 79, "wind_dir_deg" : 97, "wind_speed_ms" : 0.000, "gust_speed_ms" : 0.000, "rainfall_mm" : 2295.144, "uv" : 0, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}

{"time" : "2019-11-15 02:30:16", "model" : "Fine Offset WH65B", "id" : 172, "battery" : "OK", "temperature_C" : -1.400, "humidity" : 79, "wind_dir_deg" : 97, "wind_speed_ms" : 0.000, "gust_speed_ms" : 0.000, "rainfall_mm" : 2295.144, "uv" : 0, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}

pulse_FSK_detect(): Maximum number of pulses reached!
{"time" : "2019-11-15 02:30:32", "model" : "Fine Offset WH24", "id" : 172, "battery" : "OK", "temperature_C" : -1.400, "humidity" : 79, "wind_dir_deg" : 97, "wind_speed_ms" : 0.000, "gust_speed_ms" : 0.000, "rainfall_mm" : 2710.800, "uv" : 0, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}

{"time" : "2019-11-15 02:30:48", "model" : "Fine Offset WH65B", "id" : 172, "battery" : "OK", "temperature_C" : -1.400, "humidity" : 79, "wind_dir_deg" : 97, "wind_speed_ms" : 0.000, "gust_speed_ms" : 0.000, "rainfall_mm" : 2295.144, "uv" : 0, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}

Mindavi, 2019-11-15:
Could you record some samples and add them to this issue? This might help finding the issue. Use `rtl_433 -S all [other flags]` for a couple of minutes and submit the samples. It might be that the device type is incorrectly extracted, but someone needs to look into that using some samples.

merbanan, 2019-11-15:
Try adding -Y classic. Another pulse recovery method has been added. It apparently needs some tweaks so a few samples to look at the signal would be nice.

gary-hammer, 2019-11-15:
The -Y classic results in the WH25 being seen again, but with a different ID.
I'm attaching samples.

[Samples.zip](https://github.com/merbanan/rtl_433/files/3852512/Samples.zip)

gary-hammer, 2019-11-15:
Don't know why it closed the issue, reopening.

Mindavi, 2019-11-17:
Hmm it's weird that these are decoded differently while the ones in rtl_433_tests seem to work perfectly. Will have to dig into this further, but not today.

Mindavi, 2019-11-25:
I don't know enough about the fsk decoding to solve it, but it seems like that goes wrong.

Using the classic decoder will give bitbuffers like these (g038_915M_1000k): `[00] {490} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 55 55 55 55 55 55 16 ea 74 89 34 8b 13 9b ec 24 d5 55 55 00 00`. It's clear there's data encoded in the buffer.

However, the `-Y minmax` option gives bitbuffers like `[00] {487} 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 aa aa aa aa aa 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00`, which clearly don't contain any data.

Somehow, it goes well for the rtl_433_tests data, and it doesn't seem like the data is encoded any different. Maybe it has to do with the sample rate/frequency?

merbanan, 2020-02-16:
This is an artifact of the new minmax fsk demodulator Will hopefully be fixed in the future by running both at the same time. Command line output has been added so the users can try the old defaults. Closing issue.

closed 2020-02-16 by merbanan