merbanan/rtl_433 #1200: WS2032 Weather station - How to decode it?

vampywiz17, 2019-11-15:
I bought this weather station:

https://www.banggood.com/WS2032-Digital-Weather-Station-RF-433MHz-Wireless-Meteorological-Instrument-Weather-Forecaster-with-Dual-Mount-Wind-Sensor-and-Direction-Sensor-p-1543891.html?rmmds=myorder&cur_warehouse=CN

Of course it not work out of the box...

the rtl_433 see this, but different device:

```
time      : 2019-11-15 18:21:13
model     : Smoke detector GS 558                  id        : 4267
unit      : 28           learn     : 0             Raw Code  : 32157c
*** signal_start = 6073788, signal_end = 6197158, signal_len = 123370, pulses_found = 345
Iteration 1. t: 337    min: 169 (342)    max: 506 (3)    delta 11026
Iteration 2. t: 337    min: 169 (342)    max: 506 (3)    delta 0
Pulse coding: Short pulse length 169 - Long pulse length 506

Short distance: 118, long distance: 377, packet distance: 851

p_limit: 337
bitbuffer:: Number of rows: 3
[00] {115} 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[01] {115} 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[02] {115} 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

Any idea how to decode it? (temperature, humidty, wind speed and direction)

vampywiz17, 2019-11-16:
I record it, i hope it help:

It mean:

temp:  12.2 celsius
humidity: 90%
wind direction: North
wind speed: 0.0 m/s

[g014_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/3854537/g014_433.92M_250k.cu8.zip)

zuckschwerdt, 2019-11-16:
The signal looks good. Running `rtl_433 -A g014_433.92M_250k.cu8` gives:
```
Pulse width distribution:
 [ 0] count:    6,  width:  248 us [240;264]	(  62 S)
 [ 1] count:    3,  width: 2012 us [2008;2016]	( 503 S)
 [ 2] count:   96,  width: 1000 us [968;1048]	( 250 S)
 [ 3] count:  240,  width:  532 us [484;560]	( 133 S)
Gap width distribution:
 [ 0] count:    6,  width: 1520 us [1520;1528]	( 380 S)
 [ 1] count:  336,  width:  484 us [476;496]	( 121 S)
 [ 2] count:    2,  width: 3412 us [3412;3416]	( 853 S)
```
So thats's OOK with PWM. Long = 1000 µs, short = 532 µs, gap = 484 µs.
The overlong and very short pulses are sync, see the Pulseview.

Now try:
```
rtl_433 -X 'n=WS2032,m=OOK_PWM,s=500,l=1000,g=750,r=4000' g014_433.92M_250k.cu8
```

Collect the data (`055ecc7fffc2d2ffffffff7fc65c0` here) and note the temp/hum/...
Last step: transfer the collected data to a [BitBench](http://triq.net/bitbench) and figure out the data packet :)

vampywiz17, 2019-11-16:
Thanks it is a big help, i try to figure out what is the right conversation. I'm never do this before, but i have to start somewhere...:)

vampywiz17, 2019-11-16:
I think i never do this alone... Somebody able to help me? if need, i able to send other .cu8 file and values.

zuckschwerdt, 2019-11-16:
Use
```
rtl_433 -X 'n=WS2032,m=OOK_PWM,s=500,l=1000,g=750,r=4000'
```
to collect data lines and write them down with the temp/hum/...
Then post that data here, we'll figure it out ;)

vampywiz17, 2019-11-17:
I collect some data:

`055ecc7fffcdd27fffffff7fd0b10`

temp: 10 celsius
hum: 91%
wind direction: north
wind speed: 0.0 m/s

`055ecc7fdfcb527fffffff7fae280`

temp: 10.5 celsius

hum: 91%

wind direction: east

wind speed:  0.0 m/s

`055ecc7fbfc2d0ffffffff7f846a0`

temp: 12.2 celsius

hum: 94%

wind direction: south

wind speed 0.0 m/s

`055ecc7fbfaa4e7fffffff7fe96a0`

temp 17.1 celsius
hum 99%
wind direction: south
wind speed: 0.0 m/s

`055ecc7f9fa0ce7ffeffff7fbe898`

temp 19 celsius
hum 99%
wind direction: west
wind speed 0.0 ms

`055ecc7f9f98cfffffffff7fb91e8`

temp: 20.6 celsius
hum: 96%
wind direction: west
wind speed 0.0 ms

`055ecc7fcf8ddd7fffffff7febba8`

temp: 22.8 celsius
hum: 69%
wind direction: SouthEast
wind speed 0.0 m/s

`055ecc7fcf8bde7fffffff7fea9e0`

temp 23.2 celsus
hum 67%
wind direction: SouthEast
wind speed 0.0 m/s

vampywiz17, 2019-11-17:
I think it a bit noisy, but it is the correct data right?

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:27:58
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:27:59
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:27:59
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:27:59
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:28:00
model     : WS2032       count     : 7             num_rows  : 7             rows      :
len       : 203          data      : fffffffffffffffffffff55757b31fefe2191fffffffdff7508,
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 113          data      : **055ecc7fbf88647fffffff7fdd420**,
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 142          data      : 055ecc7ffdfffd5eabebd3fffffffbfeea10
codes     : {203}fffffffffffffffffffff55757b31fefe2191fffffffdff7508, {1}8, {1}8, {113}**055ecc7fbf88647fffffff7fdd420**, {1}8, {1}8, {142}055ecc7ffdfffd5eabebd3fffffffbfeea10
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:28:00
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:28:01
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:28:01
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:28:01
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-17 11:28:02
model     : WS2032       count     : 1             num_rows  : 1             rows      :
len       : 123          data      : ffffffffffffffffffffffffffffffe
codes     : {123}ffffffffffffffffffffffffffffffe

zuckschwerdt, 2019-11-17:
Here is how the [data looks in a BitBench](http://triq.net/bitbench?c=055ecc7fffcdd27fffffff7fd0b10%20%5BT%3A%2010.0%20C%2C%20H%3A%2091%25%2C%20W%3A%20N%2C%200.0%20m%2Fs%5D&c=055ecc7fdfcb527fffffff7fae280%20%5BT%3A%2010.5%20C%2C%20H%3A%2091%25%2C%20W%3A%20E%2C%200.0%20m%2Fs%5D&c=055ecc7fbfc2d0ffffffff7f846a0%20%5BT%3A%2012.2%20C%2C%20H%3A%2094%25%2C%20W%3A%20S%2C%200.0%20m%2Fs%5D&c=055ecc7fbfaa4e7fffffff7fe96a0%20%5BT%3A%2017.1%20C%2C%20H%3A%2099%25%2C%20W%3A%20S%2C%200.0%20m%2Fs%5D&c=055ecc7f9fa0ce7ffeffff7fbe898%20%5BT%3A%2019.0%20C%2C%20H%3A%2099%25%2C%20W%3A%20W%2C%200.0%20ms%5D&c=055ecc7f9f98cfffffffff7fb91e8%20%5BT%3A%2020.6%20C%2C%20H%3A%2096%25%2C%20W%3A%20W%2C%200.0%20ms%5D&c=055ecc7fcf8ddd7fffffff7febba8%20%5BT%3A%2022.8%20C%2C%20H%3A%2069%25%2C%20W%3A%20SE%2C%200.0%20m%2Fs%5D&c=055ecc7fcf8bde7fffffff7fea9e0%20%5BT%3A%2023.2%20C%2C%20H%3A%2067%25%2C%20W%3A%20SE%2C%200.0%20m%2Fs%5D&f=8h%20&i=true). Now we can try to figure out where the bits are.

zuckschwerdt, 2019-11-17:
Ah, well.
- Too many `ff` in the data, likely inverted: Invert the data.
- Byte six look like Temp x5. We expect Temp x10: shift left (thus multiply by 2).
- Now we already have DIR, TEMP, and HUM :)
See this [partly decoded BitBench](http://triq.net/bitbench?c=055ecc7fffcdd27fffffff7fd0b10%20%5BT%3A%2010.0%20C%2C%20H%3A%2091%25%2C%20W%3A%20N%2C%200.0%20m%2Fs%5D&c=055ecc7fdfcb527fffffff7fae280%20%5BT%3A%2010.5%20C%2C%20H%3A%2091%25%2C%20W%3A%20E%2C%200.0%20m%2Fs%5D&c=055ecc7fbfc2d0ffffffff7f846a0%20%5BT%3A%2012.2%20C%2C%20H%3A%2094%25%2C%20W%3A%20S%2C%200.0%20m%2Fs%5D&c=055ecc7fbfaa4e7fffffff7fe96a0%20%5BT%3A%2017.1%20C%2C%20H%3A%2099%25%2C%20W%3A%20S%2C%200.0%20m%2Fs%5D&c=055ecc7f9fa0ce7ffeffff7fbe898%20%5BT%3A%2019.0%20C%2C%20H%3A%2099%25%2C%20W%3A%20W%2C%200.0%20ms%5D&c=055ecc7f9f98cfffffffff7fb91e8%20%5BT%3A%2020.6%20C%2C%20H%3A%2096%25%2C%20W%3A%20W%2C%200.0%20ms%5D&c=055ecc7fcf8ddd7fffffff7febba8%20%5BT%3A%2022.8%20C%2C%20H%3A%2069%25%2C%20W%3A%20SE%2C%200.0%20m%2Fs%5D&c=055ecc7fcf8bde7fffffff7fea9e0%20%5BT%3A%2023.2%20C%2C%20H%3A%2067%25%2C%20W%3A%20SE%2C%200.0%20m%2Fs%5D&f=32h%20DIR%3A4h%20TEMP%3A12d%20HUM%3A8d%2032h%208h%208h%208h%20&s=-1&i=true).

zuckschwerdt, 2019-11-17:
To proceed we need:
- one code with a negative temperature (to confirm suspected 2's complement encoding)
- 3-5 codes with different wind speeds
- 50 or more codes (just the codes, no readings) to figure out the checksum
- also if possible reset the device (replace the battery) to see where the ID is located.

vampywiz17, 2019-11-17:
OK i will do this, i will add this soon 

Negative temp... OK i put it to the refrigerator :D

One question. What it mean? 

"50 or more codes (just the codes, no readings) to figure out the checksum"

zuckschwerdt, 2019-11-17:
Just a long list (no temp/hum/...) like:
```
055ecc7fffcdd27fffffff7fd0b10
055ecc7fdfcb527fffffff7fae280
055ecc7fbfc2d0ffffffff7f846a0
055ecc7fbfaa4e7fffffff7fe96a0
055ecc7f9fa0ce7ffeffff7fbe898
055ecc7f9f98cfffffffff7fb91e8
055ecc7fcf8ddd7fffffff7febba8
055ecc7fcf8bde7fffffff7fea9e0
```

vampywiz17, 2019-11-17:
Ahh ok :)

Here the negative temp:

`055ecc7f83d9f5ffffffff7f84728`

temp:  -7.6 celsius
hum: 20%

`055ecc7f83d675ffffffff7f80ee8`
temp: -8.3 celsius
hum: 20%

zuckschwerdt, 2019-11-17:
✅ Ok, not 2's complement but a dedicated sign-bit, i.e. 1 bit sign, 11 bit temp.

vampywiz17, 2019-11-17:
the codes:

```
055ecc7f87fdf5ffffffff7fac468
055ecc7f87fdf5ffffffff7fac468
055ecc7f87eee07fffffff7f87d88
055ecc7f87eee07fffffff7f87d88
055ecc7f87e1dbffffffff7ff66c8
055ecc7f87e1dbffffffff7ff66c8
055ecc7f9fd55affff7fff7f80000
055ecc7f9fd55affff7fff7f80000
055ecc7f9fcd58ffffffff7ff6aa8
055ecc7f9fcd58ffffffff7ff6aa8
055ecc7f9fc7577fffffff7fef178
055ecc7f9fc7577fffffff7fef178
055ecc7f9fc255ffffffff7fe8d38
055ecc7f9fc255ffffffff7fe8d38
055ecc7f9fbc567fffffff7fe3770
055ecc7f9fbc567fffffff7fe3770
055ecc7f9fb9557fffffff7fdf018
055ecc7f9fb9557fffffff7fdf018
055ecc7f9fb4d47fffffff7fd9848
055ecc7f9fb4d47fffffff7fd9848
055ecc7f9fb0537fffffff7fd4040
055ecc7f9fb0537fffffff7fd4040
055ecc7f9fafd2ffffffff7fd3028
055ecc7f9fafd2ffffffff7fd3028
055ecc7f9fad51ffffffff7fcf800
055ecc7f9fad51ffffffff7fcf800
055ecc7f9faad17fffffff7fccd58
055ecc7f9faad17fffffff7fccd58
055ecc7f9fa852ffffffff7fcb8a8
055ecc7f9fa852ffffffff7fcb8a8
055ecc7f9fa7527fffffff7fca2b0
055ecc7f9fa7527fffffff7fca2b0
055ecc7f9fa5d1ffffffff7fc8198
055ecc7f9fa5d1ffffffff7fc8198
055ecc7f9fa651ffffffff7fc8e60
055ecc7f9fa651ffffffff7fc8e60
055ecc7f9fa4d17fffffff7fc6a38
055ecc7f9fa4d17fffffff7fc6a38
055ecc7f9f9255ffffffff7fb8988
055ecc7f9f90d67fffffff7fb7e60
055ecc7f9f90d67fffffff7fb7e60
055ecc7f9f90d6ffffffff7fb8240
055ecc7f9f90d6ffffffff7fb8240
055ecc7f9f90d77fffffff7fb8b98
055ecc7f9f90d77fffffff7fb8b98
055ecc7f9f90d7ffffffff7fb9648
055ecc7f9f90d7ffffffff7fb9648
055ecc7f9f90d87fffffff7fb9aa8
055ecc7f9f90d87fffffff7fb9aa8
055ecc7f9f8f5affffffff7fbafa8
```

vampywiz17, 2019-11-17:
wind speed:

`055ecc7fff8bdb7efaffff7f91f60`

0.9 m/s

`055ecc7fef865bfc76ffff7ff67a0`

3.1 m/s

`055ecc7fff845bfdf9ffff7f88ac8`

1.8 m/s

`055ecc7fef815cfd797fff7ff5d90`

2.2 m/s

vampywiz17, 2019-11-17:
@zuckschwerdt 

I replace the battery, here the result:

```
0535c47fbf815f7fffffff7fa0130
0535c47fbf815f7fffffff7fa0130
0535c47fbf815f7fffffff7fa0130
```

zuckschwerdt, 2019-11-17:
Wind speed could be x0.4 -3 -- but that doesn't look right.
See [this BitBench for suspected Avg and Gust](http://triq.net/bitbench?c=055ecc7fffcdd27fffffff7fd0b10%20%5BT%3A%2010.0%20C%2C%20H%3A%2091%25%2C%20W%3A%20N%2C%200.0%20m%2Fs%5D&c=055ecc7fdfcb527fffffff7fae280%20%5BT%3A%2010.5%20C%2C%20H%3A%2091%25%2C%20W%3A%20E%2C%200.0%20m%2Fs%5D&c=055ecc7fbfc2d0ffffffff7f846a0%20%5BT%3A%2012.2%20C%2C%20H%3A%2094%25%2C%20W%3A%20S%2C%200.0%20m%2Fs%5D&c=055ecc7fbfaa4e7fffffff7fe96a0%20%5BT%3A%2017.1%20C%2C%20H%3A%2099%25%2C%20W%3A%20S%2C%200.0%20m%2Fs%5D&c=055ecc7f9fa0ce7ffeffff7fbe898%20%5BT%3A%2019.0%20C%2C%20H%3A%2099%25%2C%20W%3A%20W%2C%200.0%20ms%5D&c=055ecc7f9f98cfffffffff7fb91e8%20%5BT%3A%2020.6%20C%2C%20H%3A%2096%25%2C%20W%3A%20W%2C%200.0%20ms%5D&c=055ecc7fcf8ddd7fffffff7febba8%20%5BT%3A%2022.8%20C%2C%20H%3A%2069%25%2C%20W%3A%20SE%2C%200.0%20m%2Fs%5D&c=055ecc7fcf8bde7fffffff7fea9e0%20%5BT%3A%2023.2%20C%2C%20H%3A%2067%25%2C%20W%3A%20SE%2C%200.0%20m%2Fs%5D&c=055ecc7fff8bdb7efaffff7f91f60%20%5B0.9%20m%2Fs%5D&c=055ecc7fef865bfc76ffff7ff67a0%20%5B3.1%20m%2Fs%5D&c=055ecc7fff845bfdf9ffff7f88ac8%20%5B1.8%20m%2Fs%5D&c=055ecc7fef815cfd797fff7ff5d90%20%5B2.2%20m%2Fs%5D&f=1x%2032h%20DIR%3A4h%20TEMP%3A12d%20HUM%3A8d%20AVG%3F8d%20GUST%3F8d%2016h8h%20CHK%3F8h8h%20TRAIL%3A8b&i=true) values.

zuckschwerdt, 2019-11-17:
Is there a readout for Gust speed and Average speed? Are these specs right (to get an indication what values ranges are expected):

- Outdoor temperature range: -40F to 140F (-40C to 60C)
- Temperature accuracy: +- 1.0 C
- Humidity range: 20% to 90%
- Humidity accuracy: +-5%
- Wind direction: E,S,W,N,SE,NE,SW,NW
- Wind direction accuracy: +- 10 deg
- Wind speed: 0 to 50m/s, Accuracy: 0.1 m/s

vampywiz17, 2019-11-17:
yes, it possible to set the gust speed and average speed.

I think the humidity is possible to higher (i measure sometimes higher than 90%)

zuckschwerdt, 2019-11-17:
Then the wind speed you listed above is the average speed? That works as AVG / 2.25:
```
AVG:2 GUST:10 [0.9 m/s] with 2/2.25 = 0.9, 10/2.25 = 4.4
AVG:7 GUST:18 [3.1 m/s] with 7/2.25 = 3.1, 18/2.25 = 8.0
AVG:4 GUST:12 [1.8 m/s] with 4/2.25 = 1.8, 12/2.25 = 5.3
AVG:5 GUST:13 [2.2 m/s] with 5/2.25 = 2.2, 13/2.25 = 5.8
```

vampywiz17, 2019-11-17:
@zuckschwerdt I think yes. I observed that it not change every refresh state, i think it use an average value (eg: it 0 at two times, it say "zero" next time)

vampywiz17, 2019-11-17:
I hope it help.

The "0.0" value place that send you.

[image]

vampywiz17, 2019-11-18:
@zuckschwerdt you need to ne any other data to help you?

vampywiz17, 2019-11-23:
Hi @zuckschwerdt Any news in this thing?

zuckschwerdt, 2019-11-23:
Yes, the last byte is a CRC-8, not sure if the byte before that makes it CRC-16, not sure where the preamble ends or if there is a sync word.
Could you collect more codes?

vampywiz17, 2019-11-23:
> Yes, the last byte is a CRC-8, not sure if the byte before that makes it CRC-16, not sure where the preamble ends or if there is a sync word.
> Could you collect more codes?

Yes of course, i Will send you (I'm not át home now)

vampywiz17, 2019-11-23:
I think i figure out how to see the gust and the avg wind speed.

Code: `050ee07fdfd955ff7effff7f822b0`

Avarge: 1.4 km/h

gust: 3.2 km/h

(i'm switch from m/s to km/h, because i already installed the device)

I Will push a next bunch of code. (about 10-20 if enought)

vampywiz17, 2019-11-23:
@zuckschwerdt

Here the new codes:

```
050ee07fdfda55fffeffff7f83ea8
050ee07fdfda55fffeffff7f83ea8
050ee07fdfda55ff7effff7f83598
050ee07fdfdad3fefdffff7f80028
050ee07fdfdad3ff7dffff7f80e68
050ee07fdfdad3fffeffff7f82738
050ee07fdfdad3ff7effff7f81938
050ee07fdfdad37f7e7fff7f80bf8
050ee07fdfdad37f7e7fff7f80bf8
050ee07fdfdad37f7e7fff7f80bf8
050ee07fdfdbd37f7dffff7f81728
050ee07fdfdbd37efcffff7fffa28
050ee07fffdbd37f7e7fff7fa19c0
050ee07fdfdbd37f7e7fff7f81c78
050ee07fdfdbd37ffeffff7f82d68
050ee07fffdbd37fffffff7fa3f78
050ee07fdfdbd37ffeffff7f82d68
050ee07fdfdbd37ffeffff7f82d68
050ee07fdfdbd37f7dffff7f81728
050ee07fdfdc537ffeffff7f830d0 
```

zuckschwerdt, 2019-11-23:
Interesting the "ID" changed. Did you reset the battery?
Decoded (skipping the first bit) it went from:
`f5 42 67 00` to
`f5 e2 3f 00`, i.e. the preamble is likely `f5`, the ID maybe the next 16 bit, and `00` is some empty data…

zuckschwerdt, 2019-11-23:
The raw avg/gust on the two codes above are
- `1` / `2` : 1.4 km/h / 3.2 km/h
- `1` / `3` : 0.4 m/s / 1.3 m/s

Which means we have

- 1 = 0.4 m/s
- 2 = 0.9 m/s
- 3 = 1.3 m/s

So a scale of 0.43 to 0.44 for raw to m/s.

vampywiz17, 2019-11-23:
> Interesting the "ID" changed. Did you reset the battery?
> Decoded (skipping the first bit) it went from:
> `f5 42 67 00` to
> `f5 e2 3f 00`, i.e. the preamble is likely `f5`, the ID maybe the next 16 bit, and `00` is some empty data…

Yes i reseted, when i install it to the final place.

vampywiz17, 2019-11-25:
Hello @zuckschwerdt

I collect more codes, i hope it help you. It need you more code+data lines? Now i able to send you the gust and avg value.

```
050ee07fdfdbd5ff7effff7f84a30
050ee07fdfdbd5fefdffff7f830c8
050ee07fdfdbd5ff7dffff7f83e88
050ee07fdfdbd5ff7dffff7f83e88
050ee07fdfdbd5ff7dffff7f83e88
050ee07fdfdad5ff7dffff7f82908
050ee07fdfdad5ff7dffff7f82908
050ee07fdfdad5ffff7fff7f84e38
050ee07fdfdad5ffffffff7f85238
050ee07fdfdad5ffffffff7f85238
050ee07fdfdad5ffffffff7f85238
050ee07fdfda577fffffff7f865b0
050ee07fdfda577fffffff7f865b0
050ee07fdfd9577fffffff7f854b8
050ee07fefd9577fffffff7f956f0
050ee07fdfd9577fffffff7f854b8
050ee07fdfd8577fffffff7f84338 
```

zuckschwerdt, 2019-11-25:
Please test #1208 
- Find out what `Flags?` means. Is there a value when the battery gets low?
- Find out what `Rain?` means. Is this an optional (and missing) rain value or something else?

vampywiz17, 2019-11-25:
The sensor is able to support rain sensor, but it is optional and not part of my package.

i will test it, thanks!!

vampywiz17, 2019-11-25:
I build thus branch but i only get a lot of noise.

Try simple rtl_433 command, i get this constantly:

```
time      : 2019-11-25 14:23:27
model     : WS2032       StationID : 0000
Temperature: 0.0 C       Humidity  : 0 %           Wind direction: 0.0       Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00
Rain?     : 000000       Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 14:23:28
model     : WS2032       StationID : 0000
Temperature: 0.0 C       Humidity  : 0 %           Wind direction: 0.0       Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00
Rain?     : 000000       Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 14:23:28
model     : WS2032       StationID : 0000
Temperature: 0.0 C       Humidity  : 0 %           Wind direction: 0.0       Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00
Rain?     : 000000       Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
```

If i use rtl_433 -X 'n=WS2032,m=OOK_PWM,s=500,l=1000,g=750,r=4000'

sometimes get this:

```
time      : 2019-11-25 14:25:31
model     : WS2032       count     : 4             num_rows  : 4             rows      :
len       : 45           data      : effff7ff3998,
len       : 1            data      : 8,
len       : 1            data      : 8,
len       : 113          data      : 050ee07fdfbe61fffeffff7ff3998
codes     : {45}effff7ff3998, {1}8, {1}8, {113}050ee07fdfbe61fffeffff7ff3998

```

I think it need to tune a bit right?

zuckschwerdt, 2019-11-25:
It's fixed now. (`git pull`)

vampywiz17, 2019-11-25:
@zuckschwerdt 

simple rtl_433:

```
bitbuffer:: Number of rows: 1
[00] {115} ff ff ff ff ff ff ff ff ff ff ff ff ff ff e0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {83} ff ff ff ff ff ff ff ff ff ff e0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {44} ff ff ff ff ff f0 : 11111111 11111111 11111111 11111111 11111111 1111
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {78} ff ff ff ff ff ff ff ff ff fc
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {35} ff ff ff ff e0 : 11111111 11111111 11111111 11111111 111
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
```

zuckschwerdt, 2019-11-25:
Oops, forgot the debug output in there. Does it work now?

vampywiz17, 2019-11-25:
@zuckschwerdt 

It is better now! 

```
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 10
[00] { 1} 80                                                       : 1
[01] { 1} 80                                                       : 1
[02] {146} 05 0e e0 7f cf c1 60 7f ff ff ff ff fe ff ff fd 7d 5d 40
[03] { 1} 80                                                       : 1
[04] { 1} 80                                                       : 1
[05] {113} 05 0e e0 7f cf c1 60 7f ff ff ff 7f e6 3a 80
[06] { 1} 80                                                       : 1
[07] { 1} 80                                                       : 1
[08] {113} 05 0e e0 7f cf c1 60 7f ff ff ff 7f e6 3a 80
[09] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 14:54:20
model     : WS2032       StationID : E23F
Temperature: 12.5 C      Humidity  : 63 %          Wind direction: 135.0     Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 0001            00
Integrity : CRC
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1
[00] {116} ff ff ff ff ff ff ff ff ff ff ff ff ff ff f0
bitbuffer:: Number of rows: 1

```
What do you think, it possible to filter a bit the tune? Now is to mach false hit.

zuckschwerdt, 2019-11-25:
Did you `git pull` and recompile? There should not be any `bitbuffer...` output.

vampywiz17, 2019-11-25:
> Did you `git pull` and recompile? There should not be any `bitbuffer...` output.

ehh forget to pull it, only re-compile.  Now is better, i get only the right datas!

but i get (decode) less times the data with the factory receiver.

```
vamp@ubuntu:~/rtl_433/build$ rtl_433 -R 142
rtl_433 version 19.08-105-g26a8245 branch feat-ws2032 at 201911251456 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/vamp/.config/rtl_433/rtl_433.conf"...
Reading conf from "/home/vamp/.config/rtl_433/rtl_433.conf".

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 2 out of 142 device decoding protocols [ 142 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:10:21
model     : WS2032       StationID : E23F
Temperature: 11.4 C      Humidity  : 69 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC

```

vampywiz17, 2019-11-25:
@zuckschwerdt 

About 13 minutes and no new data line.

zuckschwerdt, 2019-11-25:
Can you maybe change this line: https://github.com/merbanan/rtl_433/blob/26a82455775b15389e4b5a134c6c87c76a55f46e/src/devices/ws2032.c#L43
from
`    int row = bitbuffer_find_repeated_row(bitbuffer, 2, 14 * 8); // expected: 3 rows of 113 bits`
to
`    int row = bitbuffer_find_repeated_row(bitbuffer, 1, 14 * 8); // expected: 3 rows of 113 bits`
(`1` instead of `2`)

vampywiz17, 2019-11-25:
Mutch better now!

I get datas about 1 minutes

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:37:34
model     : WS2032       StationID : E23F
Temperature: 10.3 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:38:22
model     : WS2032       StationID : E23F
Temperature: 10.2 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:39:10
model     : WS2032       StationID : E23F
Temperature: 10.2 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:41:10
model     : WS2032       StationID : E23F
Temperature: 10.0 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:41:34
model     : WS2032       StationID : E23F
Temperature: 10.0 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:41:58
model     : WS2032       StationID : E23F
Temperature: 10.0 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:42:22
model     : WS2032       StationID : E23F
Temperature: 10.0 C      Humidity  : 74 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:44:22
model     : WS2032       StationID : E23F
Temperature: 9.8 C       Humidity  : 75 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-11-25 15:44:46
model     : WS2032       StationID : E23F
Temperature: 9.8 C       Humidity  : 75 %          Wind direction: 90.0      Wind avg speed: 0.0       Wind gust : 0.0           Flags?    : 00            Rain?     : 000100
Integrity : CRC

```

vampywiz17, 2019-11-25:
@zuckschwerdt 

Now i'm work now so i can not check the receiver datas, but now it is a bit weird that the wind avg and gust is 0 every data package. 

If i will go home, i will check it.

vampywiz17, 2019-11-27:
@zuckschwerdt 

I tested the values.

the temp/hum is OK, but the wind avg and gust is not accurate. I It possible to it is a conversion problem?
 
i record some data, i hope it help to investigate it.

avg: 1.4 km/h (receiver)
avg: 1.548 km/h (rtl_433)
------------------------------
max: 3.2 km/h (receiver)
max:  3,096 (rtl_433)

------------------------------
max: 6.4 km/h (receiver)
max: 6.192 km/h (rtl_433)

zuckschwerdt, 2019-11-27:
We need a wide range of wind value and the raw numbers from the codes to see what the conversion is.
S.a. https://github.com/merbanan/rtl_433/issues/1200#issuecomment-557816186

vampywiz17, 2019-11-27:
I see, OK i collect the raw values and wind avg and max. (i working now, but if i come come try to collect it)

vampywiz17, 2019-11-27:
@zuckschwerdt 

Here some avg wind value (it not change to much, so i only record 3. I hope it enought):

--AVG--

`050ee07fdfd950ff7e7fff7ffcbb8`

1.4 km/h

`050ee07fefd9507efdffff7f8b0e0`

3.2 km/h

`050ee07fdfd9507e7cffff7ff9938`

4.7 km/h

zuckschwerdt, 2019-11-27:
So
- 0 = 0 km/h
- 1 = 1.4 km/h
- 2 = 3.2 km/h
- 3 = 4.7 km/h

A quick Linear Regression shows a factor of

1.590 ± 0.052 (1.538 to 1.642)

A counts-to-m/s factor of 0.44 seems right. You'll only get the km/h values if you round twice!
I.e. count x 0.44, round, x 3.6, round. Which is wrong, the display is bad and we have the more accurate values in rtl_433 :)

vampywiz17, 2019-11-27:
@zuckschwerdt 

Good  to know :)

Anyway, i collect Gust values. Please check it:

`050ee07fffda4e7f7e7fff7f9b390`

4.7 km/h

`050ee07fffde4e7ffeffff7f9c748`

3.2 km/h

`050ee07fdfda4e7efdffff7ffa3f8`

6.5 km/h

zuckschwerdt, 2019-11-27:
Very good, 4 = 6.5 km/h also works with 0.44 factor!
4 x 0.44 = 1.76, round to 1.8, x 3.6 = 6.48, round to 6.5 km/h

I'll update and comment the code soon. The only thing missing is battery low :)

vampywiz17, 2019-11-27:
@zuckschwerdt 

Great! well, it no so easy to "emulate" the battery low state... :)

And again, thanks the lot of help!

vampywiz17, 2019-12-10:
@zuckschwerdt 

Hi there,

I tested it and working well! What do you think it possible to merge it to the master?

Hormold, 2019-12-29:
I don't know what happens, it was available at 433.690, but today it stops work (i think because of battery). 
 Now it works on 433.720, the indoor receiver starts works after reset.
ID changed from 31374 to 9137.
Check frequency after every battery replacement.

vampywiz17, 2019-12-29:
@Hormold the ID change is normal if you replace (or only put out/in) the battery. I replace sometime the battery (test it) but the frequency not change at all.

zuckschwerdt, 2020-01-11:
Support merged with #1208. Please test and report bugs. Also maybe find out about the rain sensor if some has that option.

closed 2020-01-11 by zuckschwerdt

axwax, 2021-01-25:
> Support merged with #1208. Please test and report bugs. Also maybe find out about the rain sensor if some has that option.

I just received a WS2032 and connected a generic rain sensor to its transmitter. 
rtl_433 is working perfectly with it and counts rain impulses correctly too. Many thanks!

zuckschwerdt, 2021-01-25:
@axwax 
So "maybe_rain" can be changed to "rain" :) Is there a fixed conversion factor from bucket tips to mm? Or is this custom (you say something about a generic rain sensor?)

Can you look into "maybe_flags", is there a battery flag in there? Perhaps try flat batteries or a variable power supply?

[3 more replies]