merbanan/rtl_433 #1204: LaCrosse TX29-IT no longer detected on 915000000

bronze51, 2019-11-23:
The version back in May (or prior), when running "rtl_433 -a -f 915000000", would detect a thermometer:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-11-22 18:06:56                    brand     : LaCrosse
model     : TX29-IT      id        : 8
Battery   : OK           NewBattery: 0             Temperature: 7.3 C
Integrity : CRC

However, this evening I re-built rtl_433 (version 19.08-102-g6b7f066) and it no longer seems to detect the above thermometer. I.e. I don't get the above output.

Mindavi, 2019-11-24:
You could try if the -Y classic flag helps. See also #1198 for information.

bronze51, 2019-11-24:
Adding "-Y classic" got it back! However the model I have states TX37Y-IT on the back (as I found out yesterday).
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2019-11-24 10:07:29                    brand     : LaCrosse
model     : TX29-IT      id        : 8
Battery   : OK           NewBattery: 0             Temperature: 11.7 C
Integrity : CRC

Mindavi, 2019-11-24:
If the protocol is the same it could just be an annotation in the documentation, otherwise it would be good to fix it. Also, some samples could help @merbanan improve the new decoder to also work for this device.

bronze51, 2019-12-21:
Noob question: how do I create a a file with some useful samples? I run "rtl_433 -f 915M -Y classic".

steepleian, 2019-12-21:
One way is to use the WeeWX SDR recipe.

Find it here: -

https://github.com/weewx/weewx/wiki/sdr-rpi-recipe

zuckschwerdt, 2019-12-21:
Likely you are looking for the `-S ` option:
```
  [-S none | all | unknown | known] Signal auto save. Creates one file per signal.
```

bronze51, 2019-12-28:
I used "rtl_433 -f 915M -R 76 -Y classic -S all" to capture signals from my TX37U-IT. 
The TX37U-IT is decoded by rtl_433 as TX29-IT.
[g00x_915M_1000k.tar.gz](https://github.com/merbanan/rtl_433/files/4007242/g00x_915M_1000k.tar.gz)

evilpete, 2020-02-05:
I have a TX29IT   (FCC ID OMO-TX29U) and detect it fine,

In fact it reports as both  LaCrosse-TX35DTHIT and LaCrosse-TX29IT
which is odd since the TX29IT uses a pause width of 55 and the TX35DTHIT uses a pause width of 105

Also note that the modulation is reported incorrectly ( I suspect is a side effect of one of the two FSK demodulators, but that is another issue)

```
 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-02-03 20:40:51                    Protocol  : 75            brand     : LaCrosse
model     : LaCrosse-TX35DTHIT                     id        : 8
Battery   : 1            NewBattery: 0             Temperature: 2.6 C        Integrity : CRC
Modulation: ASK          Freq      : 0.0 MHz
RSSI      : 0.0 dB       SNR       : 0.0 dB        Noise     : 0.0 dB
pulse_demod_pcm(): LaCrosse TX35DTH-IT, TFA Dostmann 30.3155 Temperature/Humidity sensor
bitbuffer:: Number of rows: 1
[00] {80} aa aa a8 b7 52 48 10 99 a8 04
LaCrosse TX29/35 detected, buffer is 80 bits length, device is TX29
{80} aa aa a8 b7 52 48 10 99 a8 04 : 10101010 10101010 10101000 10110111 01010010 01001000 00010000 10011001 10101000 00000100
lacrosse_it	buffer from 38	{40} 92 04 26 6a 01 : 10010010 00000100 00100110 01101010 00000001
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-02-03 20:40:51                    Protocol  : 76            brand     : LaCrosse
model     : LaCrosse-TX29IT                        id        : 8
Battery   : 1            NewBattery: 0             Temperature: 2.6 C        Integrity : CRC
Modulation: ASK          Freq      : 0.0 MHz
RSSI      : 0.0 dB       SNR       : 0.0 dB        Noise     : 0.0 dB
pulse_demod_pcm(): LaCrosse TX29IT Temperature sensor
bitbuffer:: Number of rows: 1
[00] {80} aa aa a8 b7 52 48 10 99 a8 04
LaCrosse TX29/35 detected, buffer is 80 bits length, device is TX35
{80} aa aa a8 b7 52 48 10 99 a8 04 : 10101010 10101010 10101000 10110111 01010010 01001000 00010000 10011001 10101000 00000100
```

zuckschwerdt, 2020-02-05:
They are both the same protocol but with a different PCM width. But the new PCM code locks onto the transmission anyway -- it's too good ;) I'll look into that, thanks.
Strange that it says ASK. It's FSK for sure.

evilpete, 2020-02-05:
@zuckschwerdt  it's not *that* good :-)  I am having a hell of a time trying to get it to accept Transmissions from my YardStick 1.  mostly false translations resulting in pulses having drops lasting only one or two samples 
[image]

[image]

Should I open a separate issue for this ?

zuckschwerdt, 2020-02-05:
I meant the PCM pulse to bitbuffer (it's not reliant on an exact bit rate anymore) "demod" -- perhaps that should be called "slicer". Those drops are in the FSK demod. @merbanan knows the minmax FSK demod better. You could try the older `-Y classic` demod.

zuckschwerdt, 2020-02-05:
The slicers are the first point where we known the expected mark/space widths, in principle we could low-pass filter there.

merbanan, 2020-02-06:
@evilpete IIRC when using sigrok-open scripts I think it always uses the classic demod.

zuckschwerdt, 2020-02-06:
Ah, yes. The script doesn't care about the frequency and thus the classic demod.

merbanan, 2020-02-06:
And yes the terminology is wrong, it should be the bit slicers. The actual demodulation is a frequency discriminator (arctan based). Depending on the chosen mode the classic code is used. This classic one is not suitable for "fast" bitrate protocols. The new min/max based one works better for that.

merbanan, 2020-02-06:
I think we should add one option to the script to force fsk demod (bitslicer).

merbanan, 2020-02-06:
@evilpete regarding short pulses, the classic version had code to remove them. We could introduce the same logic in the min/max version. But I need sample signals that trigger this. Generic fsk demod turned out to be a not so easy problem.

merbanan, 2020-02-06:
@evilpete and yes, open another issue for it.

merbanan, 2020-02-16:
The original issue has been explained and command line output has been added to inform users about there options. Closing issue.

closed 2020-02-16 by merbanan