merbanan/rtl_433 #1205: Auriol-HG02832 decoder also fits Rubicson 48957(Transmitter for 48956)

Net-time, 2019-11-24:
Hi.
Not really an issue .
More like a thumbs up!

zuckschwerdt, 2019-11-24:
The Rubicson 48659 is a thermometer, your Auriol HG02832 is a weather device, right?
Seem like an accident that it works then. Can you submit a captured sample? (`-S all`, verify with `-r`, zip and upload here.)

LiberationFrequency, 2019-11-25:
btw: I can confirm that the decoder also works for the newer version HG05124A-DCF, IAN 321304_1901, version 07/2019. However, the display occasionally shows 0. 1°C incorrectly, especially with odd values. But this is not an error of the evaluation of a single message, the sensor sends it this way. Perhaps the value is averaged in the station.
The tx-button is missing. This is also set via the channel. Those are my placement. Could be that varies.
            0=ch1=bat_ok
            1=ch2=bat_ok
            2=ch3=bat_ok
            4=ch1=tx_button&bat_ok
            5=ch2=tx_button&bat_ok
            6=ch3=tx_button&bat_ok
            8=ch1=bat_low
            9=ch2=bat_low
            a=ch3=bat_low
            c=ch1=tx_button&bat_low
            d=ch2=tx_button&bat_low
            e=ch3=tx_button&bat_low

Net-time, 2019-12-01:
The 48956 is a weatherstation, I dont have a Auriol or the 48956. I just bought the temp/hum sensor.
Started to decode it, realised I had not updated rtl_433 in a while, turns out my efforts was wasted.
I can however rule out that it's a true accident as channel,temp, humidity and crc works.
[00] {44} 70 34 02 02 e5 c0 : 01110000 00110100 00000010 00000010 11100101 1100
Unless you are concerned with the ID which I'm assuming is the first byte and a half.

zuckschwerdt, 2019-12-06:
Oh, I got that backwards. I was looking at the Rubicson 48659 meat thermometer.
But you actually mean the Auriol-HG02832 *decoder* supports the Rubicson-48957.
I'll add that to the docs.

commit 5a81eb39 (2019-12-06): Add TX-button to some decoders (closes #1205)
closed 2019-12-06 by zuckschwerdt