merbanan/rtl_433 #1209: dati sensore wind e rain hideki (tfa Sinus-bresser ...)

farmabeps, 2019-11-25:
Buonasera a tutti
Qualcuno potrebbe aiutarmi a risolvere un problema con i sensori pioggia e vento nascondi
il mio cielo attraverso il cielo passa a prendere tutti i sensori nascondi
esempio dando il comando sudo tail -f / var / log / syslog

24 nov 19:42:18 raspberrypi weewx [919]: sdr: MainThread: lines = []
24 nov 19:42:24 raspberrypi weewx [919]: sdr: MainThread: lines = ['{"time": "2019-11-24 18:42:18", "model": "HIDEKI TS04 sensor", " rc ": 1," channel ": 5," battery ":" OK "," temperature_C ": 11.300," umidità ": 84," mic ":" CRC "} \ n ',' {" time ":" 2019-11-24 18:42:18 "," modello ":" Sensore HIDEKI TS04 "," rc ": 1," canale ": 5," batteria ":" OK "," temperatura_C ": 11.300," umidità ": 84," mic ":" CRC "} \ n ',' {" time ":" 2019-11-24 18:42:18 "," model ":" HIDEKI TS04 sensor "," rc ": 1 , "canale": 5, "batteria": "OK","temperature_C": 11.300, "umidità": 84, "mic": "CRC"} \ n ',' {"tempo": "2019-11-24 18:42:20", "modello": "HIDEKI Wind sensor "," rc ": 3," channel ": 4," battery ":" OK "," temperature_C ": 11.200," wind_speed_mph ": 0.800," gust_speed_mph ": 0.000," wind_approach ": -1," wind_direction ": 225.000," mic ":" CRC "} \ n ',' {" time ":" 2019-11-24 18:42:20 "," model ":" HIDEKI Wind sensor "," rc ": 3 , "channel": 4, "battery": "OK", "temperature_C": 11.200, "wind_speed_mph": 0.800, "gust_speed_mph": 0.000, "wind_approach": -1, "wind_direction ": 225.000," mic ":" CRC "} \ n ',' {" time ":" 2019-11-24 18:42:20 "," model ":" HIDEKI Wind sensor "," rc ": 3, "canale": 4, "batteria": "OK", "temperatura_C": 11.200, "wind_speed_mph": 0.800, "gust_speed_mph": 0.000, "wind_approach": -1, "wind_direction": 225.000, "mic" : "CRC"} \ n ']000, "wind_approach": -1, "wind_direction": 225.000, "mic": "CRC"} \ n ']000, "wind_approach": -1, "wind_direction": 225.000, "mic": "CRC"} \ n ']

ma i Dati dei due vento Sensori e pioggia alla Partenza del Comando weeewx Sono
REC: 2019/11/24 19:45:00 CET (1574,6211 milioni) Altimetro: 29,6391,463737 millions, appTemp: Nessuno, barometro: 29,6550,231288 millions, base cumulo: 1485,29,012332 millions, dateTime: 1574621100,0 , punto di rugiada: 50,7616999571, ET: Nessuno, indice di calore: 53,024, humidex: 55,6811825747, in Punto di rugiada: 47,6495854633, in Umidità: 84,0, in Temp: 52,34, inTempBatteryStatus: 0,0, intervallo: 5, maxSolarRad: 0,0, outTempo 92: 53, outTempo: 92 : 0.0, pressione: 28.6202807991, pioggia: 0.0, rain_total: None, rainRate: 0.0, usUnits: 1, windBatteryStatus: 0.0, windchill: None, windDir: None, windGust: None, windGustDir: None, windrun: None, windSpeed: None
LOOP: 2019-11-24 19:45:42 CET (1574621142) altimetro: 29.6387270474, dateTime: 1574621142, inDewpoint: 47.6495854633, inHumidity: 84.0, inTemp: 52.34, inTempBatteryStatus: 0, maxSolarRad: 0.0, pressione: 28.619873285 0, usUnits: 1
purtroppo non ho dati vento e dai pioggia

farmabeps, 2019-11-25:
il mio sensor map e' giusto?
[SDR]
# This section is for the software-defined radio driver.

# The driver to use
driver = user.sdr
cmd = rtl_433 -M utc -F json -R 42
[[sensor_map]]
   outTempBatteryStatus = battery.1:5.HidekiTS04Packet
   outHumidity = humidity.1:5.HidekiTS04Packet
   outTemp = temperature.1:5.HidekiTS04Packet
   inTempBatteryStatus = battery.5:1.HidekiTS04Packet
   inHumidity = humidity.5:1.HidekiTS04Packet
   inTemp = temperature.5:1.HidekiTS04Packet
   windDir = wind_dir.4:3.HidekiWindPacket
   windSpeed = wind_speed.4:3.HidekiWindPacket
   windGust = wind_gust.4:3.HidekiWindPacket
   windBatteryStatus = battery.4:3.HidekiWindPacket
   rainBatteryStatus = battery.4.HidekiRainPacket   
   rain_total = rain_total.4:0.HidekiRainPacket

farmabeps, 2019-11-25:
dando il comando diretto
rtl_433 -M utc -F json -R 42
"time" : "2019-11-24 19:13:41", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.000, "wind_speed_mph" : 1.300, "gust_speed_mph" : 0.100, "wind_approach" : 1, "wind_direction" : 270.000, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:41", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.000, "wind_speed_mph" : 1.300, "gust_speed_mph" : 0.100, "wind_approach" : 1, "wind_direction" : 270.000, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:41", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.000, "wind_speed_mph" : 1.300, "gust_speed_mph" : 0.100, "wind_approach" : 1, "wind_direction" : 270.000, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:45", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.100, "humidity" : 84, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:45", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.100, "humidity" : 84, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:45", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.100, "humidity" : 84, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:52", "model" : "HIDEKI Rain sensor", "rc" : 0, "channel" : 4, "battery" : "OK", "rain_mm" : 274.400, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:52", "model" : "HIDEKI Rain sensor", "rc" : 0, "channel" : 4, "battery" : "OK", "rain_mm" : 274.400, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:52", "model" : "HIDEKI Rain sensor", "rc" : 0, "channel" : 4, "battery" : "OK", "rain_mm" : 274.400, "mic" : "CRC"}

merbanan, 2019-11-25:
Hi, we can only give support in English. And it seem you are using weeewx, so I suggest you contact them for support. From the rtl_433 output everything looks fine.

closed 2019-11-25 by merbanan