merbanan/rtl_433 #1210: sensor data wind and rain hideki (tfa Sinus-bresser ...)

farmabeps, 2019-11-25:
Good evening to all
I'm a beginner and I'm sorry for the previous message
Someone could help me solve a problem with rain and wind hideki sensors
my weewx through nesdr can take all the hideki sensors
example giving the command
sudo tail -f /var/log/syslog

Nov 24 19:42:18 raspberrypi weewx[919]: sdr: MainThread: lines=[]
Nov 24 19:42:24 raspberrypi weewx[919]: sdr: MainThread: lines=['{"time" : "2019-11-24 18:42:18", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.300, "humidity" : 84, "mic" : "CRC"}\n', '{"time" : "2019-11-24 18:42:18", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.300, "humidity" : 84, "mic" : "CRC"}\n', '{"time" : "2019-11-24 18:42:18", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.300, "humidity" : 84, "mic" : "CRC"}\n', '{"time" : "2019-11-24 18:42:20", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.200, "wind_speed_mph" : 0.800, "gust_speed_mph" : 0.000, "wind_approach" : -1, "wind_direction" : 225.000, "mic" : "CRC"}\n', '{"time" : "2019-11-24 18:42:20", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.200, "wind_speed_mph" : 0.800, "gust_speed_mph" : 0.000, "wind_approach" : -1, "wind_direction" : 225.000, "mic" : "CRC"}\n', '{"time" : "2019-11-24 18:42:20", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.200, "wind_speed_mph" : 0.800, "gust_speed_mph" : 0.000, "wind_approach" : -1, "wind_direction" : 225.000, "mic" : "CRC"}\n']

but the data of the two wind and rain sensors at the start of the weeewx command are

REC: 2019-11-24 19:45:00 CET (1574621100) altimeter: 29.6391463737, appTemp: None, barometer: 29.6550231288, cloudbase: 1485.29012332, dateTime: 1574621100.0, dewpoint: 50.7616999571, ET: None, heatindex: 53.024, humidex: 55.6811825747, inDewpoint: 47.6495854633, inHumidity: 84.0, inTemp: 52.34, inTempBatteryStatus: 0.0, interval: 5, maxSolarRad: 0.0, outHumidity: 92.0, outTemp: 53.024, outTempBatteryStatus: 0.0, pressure: 28.6202807991, rain: 0.0, rain_total: None, rainRate: 0.0, usUnits: 1, windBatteryStatus: 0.0, windchill: None, windDir: None, windGust: None, windGustDir: None, windrun: None, windSpeed: None
LOOP: 2019-11-24 19:45:42 CET (1574621142) altimeter: 29.6387270474, dateTime: 1574621142, inDewpoint: 47.6495854633, inHumidity: 84.0, inTemp: 52.34, inTempBatteryStatus: 0, maxSolarRad: 0.0, pressure: 28.6198732852, rainRate: 0, usUnits: 1
purtroppo non ho dati vento e dai pioggia

my sensor map is right?
[SDR]
# This section is for the software-defined radio driver.

# The driver to use
driver = user.sdr
cmd = rtl_433 -M utc -F json -R 42
[[sensor_map]]
   outTempBatteryStatus = battery.1:5.HidekiTS04Packet
   outHumidity = humidity.1:5.HidekiTS04Packet
   outTemp = temperature.1:5.HidekiTS04Packet
   inTempBatteryStatus = battery.5:1.HidekiTS04Packet
   inHumidity = humidity.5:1.HidekiTS04Packet
   inTemp = temperature.5:1.HidekiTS04Packet
   windDir = wind_dir.4:3.HidekiWindPacket
   windSpeed = wind_speed.4:3.HidekiWindPacket
   windGust = wind_gust.4:3.HidekiWindPacket
   windBatteryStatus = battery.4:3.HidekiWindPacket
   rainBatteryStatus = battery.4.HidekiRainPacket   
   rain_total = rain_total.4:0.HidekiRainPacket

farmabeps, 2019-11-25:
giving the direct command
rtl_433 -M utc -F json -R 42
"time" : "2019-11-24 19:13:41", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.000, "wind_speed_mph" : 1.300, "gust_speed_mph" : 0.100, "wind_approach" : 1, "wind_direction" : 270.000, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:41", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.000, "wind_speed_mph" : 1.300, "gust_speed_mph" : 0.100, "wind_approach" : 1, "wind_direction" : 270.000, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:41", "model" : "HIDEKI Wind sensor", "rc" : 3, "channel" : 4, "battery" : "OK", "temperature_C" : 11.000, "wind_speed_mph" : 1.300, "gust_speed_mph" : 0.100, "wind_approach" : 1, "wind_direction" : 270.000, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:45", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.100, "humidity" : 84, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:45", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.100, "humidity" : 84, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:45", "model" : "HIDEKI TS04 sensor", "rc" : 1, "channel" : 5, "battery" : "OK", "temperature_C" : 11.100, "humidity" : 84, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:52", "model" : "HIDEKI Rain sensor", "rc" : 0, "channel" : 4, "battery" : "OK", "rain_mm" : 274.400, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:52", "model" : "HIDEKI Rain sensor", "rc" : 0, "channel" : 4, "battery" : "OK", "rain_mm" : 274.400, "mic" : "CRC"}
{"time" : "2019-11-24 19:13:52", "model" : "HIDEKI Rain sensor", "rc" : 0, "channel" : 4, "battery" : "OK", "rain_mm" : 274.400, "mic" : "CRC"}

farmabeps, 2019-11-25:
please, help me,.... don't close my question..

merbanan, 2019-11-25:
This is still a question for the weeewx forum. rtl_433 seems to decode everything fine thus there is no problem as I see it.

farmabeps, 2019-11-25:
even if I take the sensors I don't have on the graphs and on weewx no data on wind and rain
so you advise me to transfer the application?

zuckschwerdt, 2019-11-25:
- http://www.weewx.com/support.html
- https://github.com/weewx/weewx/issues

farmabeps, 2019-11-25:
grazie mille
chiudo

closed 2019-11-25 by farmabeps