merbanan/rtl_433 #1212: Informations for correct Rain decodage Bresser 5in1

yves-jose, 2019-11-26:
https://github.com/merbanan/rtl_433/blob/4e11edf9f0fb1051441975998bb15f51864e8a71/src/devices/bresser_5in1.c#L28-L41

time      : 2019-11-26 09:51:15
model     : Bresser-5in1 id        : 164
Temperature: 9.7 C       Humidity  : 95     Wind Gust : 0.0 m/s      Wind Speed: 0.4 m/s       Direction : 112.5 °      Rain : 1002.4 mm
Integrity : CHECKSUM
pulse_demod_pcm(): Bresser Weather Center 5-in-1
bitbuffer:: Number of rows: 1
[00] {285} 00 aa aa aa aa aa 2d d4  
eb 5b -7f -ff -af -fb ef -68 -ef -6a db -ff -ff -14 a4 -80 00 50 - 04 - 10 97 10 -95 -24 00 00 00 00
CC CC CC CC CC CC CC CC CC CC CC CC CC uu  I I   --G GG DW WW -- TT -RT HH RR  RR   t

* u = unknown (data changes from packet to packet, but meaning is still unknown)
uu (14 = 20) is the number of bit at 1 in next 12 bytes
* R = rain in mm, BCD coded, RxRRxRR =  => 1002.4 mm 

correction :     
int rain_raw = (msg[23] & 0x0f) + ((msg[23] & 0xf0) >> 4) * 10 + (msg[24] & 0x0f) * 100 + ((msg[24] & 0xf0) >> 4) * 1000 + ((msg[21] & 0xf0) >> 4) * 10000;
    float rain = rain_raw * 0.1f;

devZer0, 2020-02-19:
just stumbled on this.

WTF !?!?!

This is known since 26.11.2019 and i put hours on finding out the same?  🙈
 ( https://groups.google.com/forum/#!topic/rtl_433/G0dou0HomGI )

zuckschwerdt, 2020-02-19:
Oh, I never read that note as "Popcount". Ah well, two independant confirmations makes a checksum extra solid, right? ;)

devZer0, 2020-02-19:
yes :)

thanks for merging

steff107, 2020-07-17:
Is the rain now correctly decoded?
I get 2.4mm instead of 0.0mm...

peteakalad, 2020-07-17:
Yes just note it’s cumulative since the station was reset.

steff107, 2020-07-17:
Oh, that's right, thanks!

Another question: How can I reset this value?

closed 2020-09-28 by zuckschwerdt