merbanan/rtl_433 #1213: Trying to decode Lacrosse TX141W

tvass, 2019-11-27:
Hi there,
I bought this weather station:
https://www.lacrossetechnology.com/tx141w

I try to add the support for it in rtl_433.

```
$ rtl_433 -r g001_433.92M_250k.cu8 -A
Detected OOK package	@0.068516s
Analyzing pulses...
Total count:  828,  width: 642.38 ms		(160594 S)
Pulse width distribution:
 [ 0] count:  559,  width:  256 us [248;272]	(  64 S)
 [ 1] count:   47,  width:  748 us [736;756]	( 187 S)
 [ 2] count:  222,  width:  500 us [492;512]	( 125 S)
Gap width distribution:
 [ 0] count:   48,  width:  712 us [700;716]	( 178 S)
 [ 1] count:  556,  width:  472 us [464;516]	( 118 S)
 [ 2] count:  222,  width:  228 us [220;276]	(  57 S)
 [ 3] count:    1,  width: 1884 us [1884;1884]	( 471 S)
Pulse period distribution:
 [ 0] count:  779,  width:  732 us [720;960]	( 183 S)
 [ 1] count:   47,  width: 1460 us [1444;1472]	( 365 S)
 [ 2] count:    1,  width: 2136 us [2136;2136]	( 534 S)
Level estimates [high, low]:  15904,      3
RSSI: -0.1 dB SNR: 37.2 dB Noise: -37.4 dB
Frequency offsets [F1, F2]:   -1027,      0	(-3.9 kHz, +0.0 kHz)

```
Then:
```
$ rtl_433 -X 'n=lacrose-tx141w,m=OOK_PWM,s=256,l=500,r=1888,g=0,t=0,y=748' old/g001_433.92M_250k.cu8

time      : @0.068508s
model     : lacrose-tx141w                         count     : 13            num_rows  : 13            rows      : 
len       : 1            data      : 8, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befedc8faeb78, 
len       : 65           data      : f6a9befdfd6fff2d8, 
len       : 65           data      : f6a9befdfd6fff2d8, 
len       : 65           data      : f6a9befdfd6fff2d0, 
len       : 65           data      : f6a9befdfd6fff2d8, 
len       : 65           data      : f6a9befdfd6fff2d8

```
Values are supposed to be:

> Humidity:81%
> Temp:6.7
> Wind:4.1

I am confused how to move forward.
Any idea how to decode it? (temperature, humidity, wind speed). 
65 bits seems to be more other Lacrosse devices I can see here.

I send the file in this PR:
https://github.com/merbanan/rtl_433_tests/pull/315

But i guess an archive here might be a better place to get some help.
[tx141w.zip](https://github.com/merbanan/rtl_433/files/3898925/tx141w.zip)

Thanks !

zuckschwerdt, 2019-11-28:
The demod looks good. Seems to roughly match the existing LaCrosse 141's. Collect more samples with Hum/Temp/Wind annotations and [stick it in a BitBench](http://triq.net/bitbench?c=%7B65%7Df6a9befedc8faeb78%20%5B81%25%206.7C%204.1%20km%2Fh%3F%5D&c=%7B65%7Df6a9befdfd6fff2d8%20%5B81%25%206.7C%204.1%20km%2Fh%3F%5D&f=8h%20%0A&i=true).

tvass, 2019-11-28:
Thanks @zuckschwerdt 
I started to fill with more [data](http://triq.net/bitbench?c=%7B65%7Df6a9befedc8faeb78%20%5B81%25%20%206.7C%204.1%20km%2Fh%5D&c=%7B65%7Df6a9befdfd6fff2d8%20%5B81%25%20%206.7C%204.1%20km%2Fh%5D&c=%7B65%7Df6a9befed31fd8158%20%5B39%25%2021.8C%200.0%20km%2Fh%5D&c=%7B65%7Df6a9befdffffffec8%20%5B36%25%20-5.4C%200.0%20km%2Fh%5D&c=%7B65%7Df6a9befee3afb5ae8%20%5B74%25%20-4.7C%200.0%20km%2Fh%5D&f=8h%20%0A&i=true) ..

I added negative value.

zuckschwerdt, 2019-11-28:
Great. Use this format to see the values
`PRE+ID? 8h 8h 8h TYPE8h TEMP_WIND12d ?4h HUM8d CHK?8h 1b`.
- type 1 has temp+hum (temp is offset 500 and scale 10)
- type 2 has the wind

tvass, 2019-11-28:
Nice !
I reset the station to change ID a few times. Here are the latest [values](http://triq.net/bitbench?c=%7B65%7D%20f6%20a9%20be%20fe%20dc%208f%20ae%20b7%208%20%5B81%25%20%206.7C%20%204.1%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20fd%206f%20ff%202d%208%20%5B81%25%20%206.7C%20%204.1%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fe%20d3%201f%20d8%2015%208%20%5B39%25%2021.8C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20ff%20ff%20ff%20ec%208%20%5B36%25%20-5.4C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20ff%20ff%20ff%20ec%208%20%5B74%25%20-4.7C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fe%20e3%20af%20b5%20ae%208%20%5B74%25%20-4.7C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fe%20e9%20ef%20bd%2040%208%20%5B66%25%20-14.7C%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20ff%20ff%20ff%20ec%208%20%5B66%25%20-14.7C%200.0%20km%2Fh%5D&c=%7B65%7D%20f0%2040%204f%20fe%20da%20df%209e%202d%208%20%5Bchange%20bat%5D&c=%7B65%7D%20f0%2040%204f%20fd%20ff%20ff%20ff%20f1%208&c=%7B65%7D%20f0%2040%204f%20be%20da%20cf%209d%20f0%208&c=%7B65%7D%20f0%2040%204f%20bd%20ff%20ff%20ff%2011%208&c=%7B65%7D%20f0%2040%204f%20fe%20d4%204f%20d1%2084%208%20&c=%7B65%7D%20f0%2040%204f%20fd%20ff%20ff%20ff%20f1%208&c=%7B65%7D%20f0%2040%204f%20fe%20d4%203f%20d2%20ec%208%20&c=%7B65%7D%20f0%2040%204f%20fd%20ff%20ff%20ff%20f1%208&c=%7B65%7D%20f7%20e2%204e%20be%20d3%204f%20d3%20b2%208%20%5Bchange%20bat%5D&c=%7B65%7D%20f7%20e2%204e%20bd%20ff%20ff%20ff%2046%208&f=PRE%2BID%2028h%20TYPE%204h%20TEMP_WIND%2012d%204h%20HUM%208d%20CHK%208h%201b&i=true).

I am trying to figure the PREFIX and ID now.
(btw I moved the type on 4H instead 8H. Not sure this is good). 

`PRE+ID 28h TYPE 4h TEMP_WIND 12d 4h HUM 8d CHK 8h 1b`

Next step will be to trigger the low battery flag + CRC i guess.

zuckschwerdt, 2019-11-28:
Usual IDs are 8 bits. But 16 bit seems plausible. The first 4 bits then would be a preamble and the byte before the type maybe flags?

I just dumped your collected codes into `revsum` from [revdgst](https://github.com/triq-org/revdgst/) and the CRC found is the very usual 0x31:

`Found: crc8(&b[0], 7, 0x31, 0x00) ^ b[7] == 0; // (100.0%)`

tvass, 2019-11-28:
Ack. I am now looking for a low battery to trigger this alert ... This is the only missing piece.

It's still unclear for me what other flags could be, as there is nothing else to decode [[spec](https://www.lacrossetechnology.com/ca87092)] ?

zuckschwerdt, 2019-11-29:
Maybe Power-on Sync-mode.

tvass, 2019-12-02:
Hi @zuckschwerdt 
I added new values that include Battery alert [[here](http://triq.net/bitbench?c=%7B65%7D%20f6%20a9%20be%20fe%20dc%208f%20ae%20b7%208%20%5B81%25%20%206.7C%20%204.1%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20fd%206f%20ff%202d%208%20%5B81%25%20%206.7C%20%204.1%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fe%20d3%201f%20d8%2015%208%20%5B39%25%2021.8C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20ff%20ff%20ff%20ec%208%20%5B36%25%20-5.4C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20ff%20ff%20ff%20ec%208%20%5B74%25%20-4.7C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fe%20e3%20af%20b5%20ae%208%20%5B74%25%20-4.7C%20%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fe%20e9%20ef%20bd%2040%208%20%5B66%25%20-14.7C%200.0%20km%2Fh%5D&c=%7B65%7D%20f6%20a9%20be%20fd%20ff%20ff%20ff%20ec%208%20%5B66%25%20-14.7C%200.0%20km%2Fh%5D&c=%7B65%7D%20f0%2040%204f%20fe%20da%20df%209e%202d%208%20%5Bchange%20bat%5D&c=%7B65%7D%20f0%2040%204f%20fd%20ff%20ff%20ff%20f1%208&c=%7B65%7D%20f0%2040%204f%20be%20da%20cf%209d%20f0%208&c=%7B65%7D%20f0%2040%204f%20bd%20ff%20ff%20ff%2011%208&c=%7B65%7D%20f0%2040%204f%20fe%20d4%204f%20d1%2084%208%20&c=%7B65%7D%20f0%2040%204f%20fd%20ff%20ff%20ff%20f1%208&c=%7B65%7D%20f0%2040%204f%20fe%20d4%203f%20d2%20ec%208%20&c=%7B65%7D%20f0%2040%204f%20fd%20ff%20ff%20ff%20f1%208&c=%7B65%7D%20f7%20e2%204e%20be%20d3%204f%20d3%20b2%208%20%5Bchange%20bat%5D&c=%7B65%7D%20f7%20e2%204e%20bd%20ff%20ff%20ff%2046%208&f=PRE%205h%20ID%2023h%20TYPE%204h%20TEMP_WIND%2012d%204h%20HUM%208d%20CHK%208h%201b&i=true)]

`PRE5h ID23h TYPE4h TEMP_WIND12d 4h HUM8d CHK7h BAT1b 1b`

I am not sure about the checksum anymore. If if you have a minute to check it. Thanks!

zuckschwerdt, 2019-12-02:
It's unlikely that the CRC isn't the last thing. And it will be 8 bit wide. The format string should end with `CHK8h 1x`.
ID will at most be 16 bit. Perhaps you can reset the battery a few times and see which of the first 24 bits really change and which are always the same?

tvass, 2019-12-02:
Thanks for your help.
I reset the sensor a few times....

 ```
{65} f6 a9 be fe dc 8f ae b7 8 [NEW ID, H=81%,T=6.7C, BAT OFF]
{65} f0 40 4f fe da df 9e 2d 8 [NEW ID BAT OFF]
{65} f7 e2 4e be d3 4f d3 b2 8 [NEW ID BAT OFF]
{65} f6 6b 2a 7e d3 5f e0 f5 8 [NEW ID + BAT ON]
{65} f6 6b 2a fe d3 6f e0 b6 8 [SAME ID + BAT OFF]
{65} f2 6d 0d fe d3 3f e1 f7 8 [NEW ID BAT OFF]
{65} f3 97 75 fe d3 3f e1 35 8 [NEW ID BAT OFF] 
{65} f4 b3 d3 fe d3 3f e1 5d 8 [NEW ID BAT OFF]
{65} f3 04 64 fe d3 3f e1 0a 8 [NEW ID BAT OFF]
{65} f1 86 e6 fe d3 3f e1 8f 8 [NEW ID BAT OFF]

```

As far as i can [see](http://triq.net/bitbench?c=%7B65%7D%20f6%20a9%20be%20fe%20dc%208f%20ae%20b7%208%20%5BNEW%20ID%2C%20H%3D81%25%2CT%3D6.7C%2C%20BAT%20OFF%5D&c=%7B65%7D%20f0%2040%204f%20fe%20da%20df%209e%202d%208%20%5BNEW%20ID%20BAT%20OFF%5D&c=%7B65%7D%20f7%20e2%204e%20be%20d3%204f%20d3%20b2%208%20%5BNEW%20ID%20BAT%20OFF%5D&c=%7B65%7D%20f6%206b%202a%207e%20d3%205f%20e0%20f5%208%20%5BNEW%20ID%20%2B%20BAT%20ON%5D&c=%7B65%7D%20f6%206b%202a%20fe%20d3%206f%20e0%20b6%208%20%5BSAME%20ID%20%2B%20BAT%20OFF%5D&c=%7B65%7D%20f2%206d%200d%20fe%20d3%203f%20e1%20f7%208%20%5BNEW%20ID%20BAT%20OFF%5D&c=%7B65%7D%20f3%2097%2075%20fe%20d3%203f%20e1%2035%208%20%5BNEW%20ID%20BAT%20OFF%5D%20&c=%7B65%7D%20f4%20b3%20d3%20fe%20d3%203f%20e1%205d%208%20%5BNEW%20ID%20BAT%20OFF%5D&c=%7B65%7D%20f3%2004%2064%20fe%20d3%203f%20e1%200a%208%20%5BNEW%20ID%20BAT%20OFF%5D&c=%7B65%7D%20f1%2086%20e6%20fe%20d3%203f%20e1%208f%208%20%5BNEW%20ID%20BAT%20OFF%5D&f=PRE24b%20BAT1b%20%3F1b%202h%20TYPE4h%20TEMP_WIND12d%204h%20HUM8d%20CHK8h%201x%0A%0A%0A&i=true) Prefix is  5b, ID is 19b.

zuckschwerdt, 2019-12-02:
If you just use `v` as format string you can easily see that you are right. `5h 19h` : 19 bit ID is really long (and i would have expected maybe 20 but not 19).

I'll try to add a full decoder soon.

tvass, 2019-12-04:
Great. Let me know if I can help with it.
Thks

zuckschwerdt, 2020-01-11:
TX141W is now supported with latest master. Please test and report any bugs ;)

closed 2020-01-11 by zuckschwerdt

tvass, 2020-01-13:
Thanks a lot @zuckschwerdt ! It works but I dont think it decodes packet with type 2.

zuckschwerdt, 2020-01-14:
The code is there, it should work. Perhaps you can record a sample of txpe 2 and upload a zip here?

tvass, 2020-01-14:
ok.

```
pi@pi3sdr:~/work/rtl433/build/src $ ./rtl_433 -S all
rtl_433 version 19.08-139-g2736df3 branch master at 202001111915 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 116 out of 145 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-104 108-116 119 121 124-128 131-145 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-14 14:57:44
model     : LaCrosse-TX141W                        Sensor ID : 6c40a
Channel   : 0            Battery level: 1          Temperature: -6.50 C      Humidity  : 81 %          Test?     : 0
*** Saving signal to file g001_433.92M_250k.cu8 (195863 samples, 393216 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-14 14:58:15
model     : LaCrosse-TX141W                        Sensor ID : 6c40a
Channel   : 0            Battery level: 1          Temperature: -6.50 C      Humidity  : 81 %          Test?     : 0
*** Saving signal to file g002_433.92M_250k.cu8 (195861 samples, 393216 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-01-14 14:58:47
model     : LaCrosse-TX141W                        Sensor ID : 6c40a
Channel   : 0            Battery level: 1          Temperature: -6.40 C      Humidity  : 81 %          Test?     : 0
*** Saving signal to file g003_433.92M_250k.cu8 (195862 samples, 393216 bytes)
^CSignal caught, exiting!
```

[TX141W.zip](https://github.com/merbanan/rtl_433/files/4059788/TX141W.zip)

zuckschwerdt, 2020-01-14:
I see, the repeat count on the type 2 packet is lower (only 5). I've reduced the required repeat minimum, should work now.

tvass, 2020-01-14:
It works like a charm now. I also checked the value with the station screen and it is correct.
Thanks Christian.