merbanan/rtl_433 #1214: Add Protocol Weather Station Bresser 6in1

f4gqk, 2019-11-29:
Data of Weather Station Bresser 6in1
https://www.bresser.de/en/Weather-Time/Weather-Center/BRESSER-PC-Weather-station-with-6-in-1-outdoor-sensor.html

decoding,pictures.... here
[Bresser6in1.zip](https://github.com/merbanan/rtl_433_tests/files/3902193/Bresser6in1.zip)

[image]

f4gqk, 2019-12-01:
[image]

f4gqk, 2019-12-03:
Hello,

Bresser 6in1 was added yesterday in rtl433_tests.
[https://github.com/merbanan/rtl_433_tests/tree/master/tests/bresser_6in1](url)

How will the protocol be added in rtl_433?

Need some information? If so why ?

Or do we have to wait a while to add other protocols in the update?

I am ready to do the required manipulations if necessary.

Thanks
Seb

zuckschwerdt, 2019-12-03:
Can you run `rtl_433 -s 1M -X 'n=Bresser-6in1,m=FSK_PCM,s=100,l=100,r=2000'`
and send us:

```
{240}d5552aaaaa859ea17d54184000260e30fd21ffdc8f6402182203fffc0f20 [Hum 10% Temp 12.3 C Wind 4.2 m/s ...]
{246}5555552aaa859ea2bd9086100009838c3f4867f3918f87fffffffff00b33fc [Hum 20% Temp 7.8 C Wind 2.3 m/s ...]
```

f4gqk, 2019-12-03:
{"time" : "2019-12-03 14:29:58", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 192, "data" : "9555552a9542cf50cfe503080004c1c61f00000000000000"}], "codes" : ["{192}9555552a9542cf50cfe503080004c1c61f00000000000000"]}
{"time" : "2019-12-03 14:29:58", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 17, "data" : "00000"}], "codes" : ["{17}00000"]}
{"time" : "2019-12-03 14:30:10", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 169, "data" : "aaaa552aaa859ea0000000000000000000000000000"}], "codes" : ["{169}aaaa552aaa859ea0000000000000000000000000000"]}
{"time" : "2019-12-03 14:30:10", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 45, "data" : "000000000000"}], "codes" : ["{45}000000000000"]}
{"time" : "2019-12-03 14:30:10", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 17, "data" : "00000"}], "codes" : ["{17}00000"]}
{"time" : "2019-12-03 14:30:22", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 245, "data" : "aa552a954aa167a83e5443080004c1c61fe67bff18eb0fffffffffe01147f8"}], "codes" : ["{245}aa552a954aa167a83e5443080004c1c61fe67bff18eb0fffffffffe01147f8"]}
{"time" : "2019-12-03 14:30:34", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 246, "data" : "54aa55552a859ea1e40806100009838c3ff7bffe31d9003d887bfffc0663fc"}], "codes" : ["{246}54aa55552a859ea1e40806100009838c3ff7bffe31d9003d887bfffc0663fc"]}
{"time" : "2019-12-03 14:30:46", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 244, "data" : "55555554aa167a8fcb32184000260e30ffdefff8c7443fffffffff8031eff"}], "codes" : ["{244}55555554aa167a8fcb32184000260e30ffdefff8c7443fffffffff8031eff"]}
{"time" : "2019-12-03 14:30:58", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 248, "data" : "54aa552a9542cf50322fc3080004c1c61fe799ff9908801ec43e1ffe0108ff"}], "codes" : ["{248}54aa552a9542cf50322fc3080004c1c61fe799ff9908801ec43e1ffe0108ff"]}

[Hum 78% Temp 7.7 C Wind 0.8 m/s ...]

{"time" : "2019-12-03 14:33:22", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 248, "data" : "aa552a954aa167a8432210c20001307187f7bdff4020400f7a1ecfff00fcff"}], "codes" : ["{248}aa552a954aa167a8432210c20001307187f7bdff4020400f7a1ecfff00fcff"]}

[Hum 77% Temp 7.8 C Wind 2.2 m/s ...]

{"time" : "2019-12-03 14:35:46", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 246, "data" : "952a95554aa167a8c20f2184000260e30fef61fe1963003e087cfff8011ff8"}], "codes" : ["{246}952a95554aa167a8c20f2184000260e30fef61fe1963003e087cfff8011ff8"]}
{"time" : "2019-12-03 14:35:58", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 244, "data" : "54aa55552a859ea3f84006100009838c3ff633ff32230fffffffffe015eff"}], "codes" : ["{244}54aa55552a859ea3f84006100009838c3ff633ff32230fffffffffe015eff"]}
{"time" : "2019-12-03 14:36:10", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 248, "data" : "54aa552a9542cf5108467184000260e30ff32cffc045801f043e1ffe0119ff"}], "codes" : ["{248}54aa552a9542cf5108467184000260e30ff32cffc045801f043e1ffe0119ff"]}
{"time" : "2019-12-03 14:36:22", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 244, "data" : "5554aa552a859ea0075f86100009838c3fcacffc657c3fffffffff80467fe"}], "codes" : ["{244}5554aa552a859ea0075f86100009838c3fcacffc657c3fffffffff80467fe"]}

[Hum 79% Temp 7.9 C Wind 1.5 m/s ...]

f4gqk, 2019-12-03:
{"time" : "2019-12-03 15:36:34", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 241, "data" : "5554000000000000000000000000000000000000000000000000000000000"}], "codes" : ["{241}5554000000000000000000000000000000000000000000000000000000000"]}
{"time" : "2019-12-03 15:36:46", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 244, "data" : "aa55552a9542cf507d7f06100009838c3ff63ffe311f0fffffffffe017f00"}], "codes" : ["{244}aa55552a9542cf507d7f06100009838c3ff63ffe311f0fffffffffe017f00"]}

[Hum 76% Temp 8.0 C Wind 0.9 m/s ...]

{"time" : "2019-12-03 15:38:34", "model" : "bresser", "count" : 1, "num_rows" : 1, "rows" : [{"len" : 243, "data" : "5554aaaaaa167a81f620184000260e30ff32cffc0a88020043d9fffc1f9fe"}], "codes" : ["{243}5554aaaaaa167a81f620184000260e30ff32cffc0a88020043d9fffc1f9fe"]}

[Hum 76% Temp 8.0 C Wind 1.2 m/s ...]

zuckschwerdt, 2019-12-03:
Your data in a BitBench [looks like this](https://triq.net/bitbench?c=%7B245%7Daa552a954aa167a83e5443080004c1c61fe67bff18eb0fffffffffe01147f8%20%5BHum%2078%25%20Temp%207.7%20C%20Wind%200.8%20m%2Fs%5D&c=%7B246%7D54aa55552a859ea1e40806100009838c3ff7bffe31d9003d887bfffc0663fc%20%5BHum%2078%25%20Temp%207.7%20C%20Wind%200.8%20m%2Fs%5D&c=%7B244%7D55555554aa167a8fcb32184000260e30ffdefff8c7443fffffffff8031eff%20%5BHum%2078%25%20Temp%207.7%20C%20Wind%200.8%20m%2Fs%5D&c=%7B248%7D54aa552a9542cf50322fc3080004c1c61fe799ff9908801ec43e1ffe0108ff%20%5BHum%2078%25%20Temp%207.7%20C%20Wind%200.8%20m%2Fs%5D&c=&c=%7B248%7Daa552a954aa167a8432210c20001307187f7bdff4020400f7a1ecfff00fcff%20%5BHum%2077%25%20Temp%207.8%20C%20Wind%202.2%20m%2Fs%5D&c=&c=%7B246%7D952a95554aa167a8c20f2184000260e30fef61fe1963003e087cfff8011ff8%20%5BHum%2079%25%20Temp%207.9%20C%20Wind%201.5%20m%2Fs%5D&c=%7B244%7D54aa55552a859ea3f84006100009838c3ff633ff32230fffffffffe015eff%20%5BHum%2079%25%20Temp%207.9%20C%20Wind%201.5%20m%2Fs%5D&c=%7B248%7D54aa552a9542cf5108467184000260e30ff32cffc045801f043e1ffe0119ff%20%5BHum%2079%25%20Temp%207.9%20C%20Wind%201.5%20m%2Fs%5D&c=%7B244%7D5554aa552a859ea0075f86100009838c3fcacffc657c3fffffffff80467fe%20%5BHum%2079%25%20Temp%207.9%20C%20Wind%201.5%20m%2Fs%5D&c=&c=%7B244%7Daa55552a9542cf507d7f06100009838c3ff63ffe311f0fffffffffe017f00%20%5BHum%2076%25%20Temp%208.0%20C%20Wind%200.9%20m%2Fs%5D&c=%7B243%7D5554aaaaaa167a81f620184000260e30ff32cffc0a88020043d9fffc1f9fe%20%5BHum%2076%25%20Temp%208.0%20C%20Wind%201.2%20m%2Fs%5D&f=8h%20&a=Preamble&m=0). I can't see the values.

zuckschwerdt, 2019-12-03:
Oh, it's actually (120µs not 100µs):
`rtl_433 -s 1M -X 'n=Bresser-6in1,m=FSK_PCM,s=120,l=120,r=2500'`

zuckschwerdt, 2019-12-03:
Can you again get data? Should look much better, like this:
```
{204}d55555555516ea2f550c4001618c7d47fd93b408c240fff8390
{208}555555555516ea57688c4001618c7d46fd933c7fffffffc05b7f
{207}d55555555516ea7c170c4001618c7e637e931c08c240fff83400
{208}555555555516ea623e8c4001618c7e3c7e94947fffffffc072ff
{208}555555555516ea140f0c4001618c7ddbfe132c7fffffffc0587f
{207}d55555555516ea10f40c4001618c7dce7e198408c240fff85bfe
{208}d55555555516ea41d70c4001618c7e3c7e14947fffffffc07300
{207}555555555516ea2e720c4001618c7ddd7e134c08c240fff80b00
{208}555555555516ea685e8c4001618c7cd6fd13247fffffffc05fff
```
But we need: [Hum 76% Temp 8.0 C Wind 1.2 m/s] :)

f4gqk, 2019-12-03:
2019-12-03 18:57:35
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 206          data      : 55555555545ba83e803100058631ff11fe6611ffffffff01cc00
codes     : {206}55555555545ba83e803100058631ff11fe6611ffffffff01cc00

[Hum 96% Temp 3.8 C Wind 0.7 m/s ...]

f4gqk, 2019-12-03:
time      : 2019-12-03 20:34:00
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 206          data      : 55555555545ba8b4503100058631fffffe66d1ffffffff01f5f8
codes     : {206}55555555545ba8b4503100058631fffffe66d1ffffffff01f5f8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-03 20:34:12
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 205          data      : 55555555545ba999263100058631fffffe66d006092bffe0cff8
codes     : {205}55555555545ba999263100058631fffffe66d006092bffe0cff8

[Hum 95% Temp 3.0 C Wind 0.0 m/s ...]

time      : 2019-12-03 20:35:48
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 199          data      : 55555555545ba840523100058631ff77fe668000495fff0bbe
codes     : {199}55555555545ba840523100058631ff77fe668000495fff0bbe

[Hum 95% Temp 3.0 C Wind 0.4 m/s ...]

f4gqk, 2019-12-03:
time      : 2019-12-03 20:44:12
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 205          data      : 55555555545ba94d063100058631fffffe665006092bffe14ff8
codes     : {205}55555555545ba94d063100058631fffffe665006092bffe14ff8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-03 20:44:24
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 206          data      : 55555555545ba860703100058631fffffe6651ffffffff0135fc
codes     : {206}55555555545ba860703100058631fffffe6651ffffffff0135fc

[Hum 95% Temp 3.0 C Wind 0.0 m/s ...]

f4gqk, 2019-12-03:
time      : 2019-12-03 22:37:01
model     : Bresser-6in1 count     : 1             num_rows  : 1             rows      :
len       : 205          data      : 55555555545ba98be83100058631fffffe6130050929ffe17800
codes     : {205}55555555545ba98be83100058631fffffe6130050929ffe17800

[Hum 95% Temp 2.8 C Wind 0.8 m/s ...]

zuckschwerdt, 2019-12-04:
Here [is a BitBench](https://triq.net/bitbench?c=%7B206%7D55555555545ba83e803100058631ff11fe6611ffffffff01cc00%20%5BHum%2096%25%20Temp%203.8%20C%20Wind%200.7%20m%2Fs%5D&c=%7B205%7D55555555545ba999263100058631fffffe66d006092bffe0cff8%20%5BHum%2095%25%20Temp%203.0%20C%20Wind%200.0%20m%2Fs%5D&c=%7B199%7D55555555545ba840523100058631ff77fe668000495fff0bbe%20%5BHum%2095%25%20Temp%203.0%20C%20Wind%200.4%20m%2Fs%5D&c=%7B205%7D55555555545ba94d063100058631fffffe665006092bffe14ff8&c=%7B206%7D55555555545ba860703100058631fffffe6651ffffffff0135fc%20%5BHum%2095%25%20Temp%203.0%20C%20Wind%200.0%20m%2Fs%5D&c=%7B205%7D55555555545ba924d23100058631ff99fe68b004e92dffe073f8%20%5BHum%2096%25%20Temp%202.7%20C%20Wind%200.4%20m%2Fs%5D&c=%7B202%7D55555555545ba813403100058631ff77fe6810050929ffe1180%20%5BHum%2094%25%20Temp%202.8%20C%20Wind%200.4%20m%2Fs%5D&c=%7B205%7D55555555545ba98be83100058631fffffe6130050929ffe17800%20%5BHum%2095%25%20Temp%202.8%20C%20Wind%200.8%20m%2Fs%5D&f=SYNC16h%20CHK%3F8h8h%20ID%3F8h8h8h8h8h%208h8h8h%208h%208h%20TEMP8h.4h%20%3F4h%20HUM8h%208h%208h%208h%208h%0A&a=Sync&m=2dd4) with the data collected.

Try out the PR #1225 for Temp and Hum so far.

f4gqk, 2019-12-04:
I do not understand.

Do you have the information you were looking for or do you want something else?

Do you want me to enter data in BITCHEN format?

Thanks,

zuckschwerdt, 2019-12-04:
All is well. The BitBench is just to play and understand.
Try https://github.com/merbanan/rtl_433/archive/feat-bresser6in1.zip

f4gqk, 2019-12-04:
The link allows me what?

Can I decode the frames of this weather station Bresser 6-in-1?

With which syntax ?

Thank you

f4gqk, 2019-12-08:
hello 

Little question 

Will the Bresser 6in1 protocol add to the list of protocols so that I can use the weather station with weewx.

Thanks

zuckschwerdt, 2019-12-08:
Yes. But it will take some time.

Do you use Windows or Linux? Can you test the code? Use #1225 or download the zip, then compile and run.

f4gqk, 2019-12-08:
Ok for the developpement . I use Linux on Raspberry. I will test the code. 

Thanks

nkpap, 2019-12-27:
HI,  

any news with this device?
Do you need data? How can we help?

BR
Nikos

f4gqk, 2019-12-29:
The solution but it's simple for me.

Would make RTL433 compatible with this weather station so that I can send the data on WEEWX because it is not recognized by the cable.

Thanks

nkpap, 2019-12-30:
Hi,

I also need the RTL decoding to feed the WEEWX.
BR
Nikos

level20peon, 2020-01-28:
I also own this weather station (the display is a colored one though). Can I do anything to help implementing it?

zuckschwerdt, 2020-02-26:
@f4gqk your recent data looks good in a [BitBench](http://triq.net/bitbench?c=%7B164%7D5ba938663100058631f99ff84e100908efffe14c0&c=%7B165%7D5ba965383100058631f99ff84e500928efffe0ec00&c=%7B165%7D5ba8c9623100058631f99ff84e31fffffffe030ff0&c=%7B162%7D5ba80e7c3100058631f9effa4e300928efffe0bbc&c=%7B165%7D5ba8d4383100058631fb33fa4e91fffffffe0317f0&c=%7B165%7D5ba852aa3100058631fbb5fc4e900928efffe09000&c=%7B165%7D5ba877ae3100058631fbb7fc4e91fffffffe029000&c=%7B164%7D5ba9deac3100058631fdbffc4e900928efffe0840&c=%7B165%7D5ba9981a3100058631fd77fc4f11fffffffe024ff0&f=8h%20&s=1)
Shift by one bit and the expected sync word `2dd4` is visible.

The branch from #1225 does work, yes? We just need to figure out where the wind etc. data is.

We need annotated codes like this:
```
2dd4  30 38 18 80 02 c3 18 ff ff ff 33 28 ff ff ff ff 80 9a 7f [Hum 95% Temp 3.0 C Wind 0.0 m/s]
```
or this
```
{147} 5e aa 18 80 02 c3 18 fa 8f fb 27 68 11 84 81 ff f0 72 00  [Temp 11.8 C  Hum 81%]
```

f4gqk, 2020-02-26:
with which command line? because for now I only have code

zuckschwerdt, 2020-02-26:
```
git clone https://github.com/merbanan/rtl_433.git
cd rtl_433
git checkout feat-bresser6in1
mkdir build ; cd build ; cmake .. && make
./src/rtl_433
```

danrhjones, 2020-03-20:
I am also hoping to get this working for the bresser 5-in-1 model 7002580, which I believe is the same as the 6-in-1

If there is any thing I can do to help, please let me know

nkpap, 2020-08-15:
I installed the latest version
I get 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-15 09:50:30
model     : Bresser-6in1 Temperature: 28.8 C       Humidity  : 36            Integrity : CHECKSUM
bresser_6in1_callback: {147} 31 48 18 b0 08 87 18 ff ff ff 26 58 ff ff ff ff 01 18 00 
bresser_6in1_callback: {149} 86 ee 18 b0 08 87 18 ff ff ff 26 58 28 80 36 ff c0 78 00 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

except of Temperature and Humidity no other readings are decoded. 
Did I miss something?

Nikos

zuckschwerdt, 2020-08-15:
Work in progress. So far nobody found other values in that data.

zuckschwerdt, 2020-08-15:
@nkpap can you log some of that data and annotate with the expected readings (see my 26 Feb comment)?
Perhaps we can spot and add more fields.

nkpap, 2020-08-20:
Sure,

here are some data. Do you need more?
Temperature: 29.5 C
Humidity  52%
Wdir 262
Wind 0
UV 0 
Baro 990
Rain 0
_ _ __ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-20 09:40:03
model     : Bresser-6in1 Temperature: 29.5 C       Humidity  : 52            Integrity : CHECKSUM
bresser_6in1_callback: {149} 74 d4 18 b0 08 87 18 ff ff ff 26 28 ff ff ff ff 01 48 00 
bresser_6in1_callback: {150} 7a 29 18 b0 08 87 18 ff ff ff 26 28 29 54 52 ff f0 87 f8 
_ _

Temperature: 29.5 C
Humidity  52%
Wdir 359
Wind 1.1
UV 0 
Baro 990
Rain 0
time      : 2020-08-20 09:43:39
model     : Bresser-6in1 Temperature: 29.5 C       Humidity  : 52            Integrity : CHECKSUM
bresser_6in1_callback: {150} 7a 05 18 b0 08 87 18 ff ff ff 35 88 ff ff ff ff 01 d9 f8 
bresser_6in1_callback: {149} 63 9e 18 b0 08 87 18 fb ae fe 35 98 29 54 52 ff f0 5e 00 

Temperature: 29.5 C
Humidity  52%
Wdir 84
Wind 0.9
UV 0 
Baro 990
Rain 0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-20 09:46:03
model     : Bresser-6in1 Temperature: 29.5 C       Humidity  : 52            Integrity : CHECKSUM
bresser_6in1_callback: {149} 34 3d 18 b0 08 87 18 fd b6 ff 08 48 ff ff ff ff 01 91 f8 
bresser_6in1_callback: {149} 66 e9 18 b0 08 87 18 fe ff fe 08 48 29 54 53 ff f0 86 00 

Temperature: 29.6 C
Humidity  53%
Wdir 269
Wind 0
UV 2.1
Baro 990
Rain 0.7

_ _ _ _ _ _ _ _ _ _ _ _ _ __ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-20 09:49:18
model     : Bresser-6in1 Temperature: 29.6 C       Humidity  : 53            Integrity : CHECKSUM
bresser_6in1_callback: {150} 22 d9 18 b0 08 87 10 ff ff ff 28 68 ff ff fd ab 01 64 00 
bresser_6in1_callback: {149} da bd 18 b0 08 87 10 ff ff ff 28 48 29 64 53 fd e0 6e 00 

Temperature: 29.7 C
Humidity  53%
Wdir 269
Wind 0
UV 2.1
Baro 990
Rain 1.2
_ _ _ _ _ _ _ _ _ _ _ _ _ __ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-20 09:53:18
model     : Bresser-6in1 Temperature: 29.7 C       Humidity  : 53            Integrity : CHECKSUM
bresser_6in1_callback: {150} 99 10 18 b0 08 87 10 ff ff ff 26 68 ff ff ed 8f 01 92 00 
bresser_6in1_callback: {150} 43 53 18 b0 08 87 10 ff ff ff 26 78 29 84 53 fd e0 20 00 

N.-

nkpap, 2020-09-20:
Hi all,

is there any progress on the Bresser 6in1?
Here https://www.mail-archive.com/search?l=weewx-user@googlegroups.com&q=subject:%22%5C%5Bweewx%5C-user%5C%5D+Re%5C%3A+Bresser+6in1%22&o=newest&f=1
they claim to have a solution for weather stations with PC connected consoles.
But what about radio?

BR
Nikos

zuckschwerdt, 2020-09-20:
I could find wind dir and perhaps UV in the data, see [this BitBench](https://triq.net/bitbench?c=%7B149%7D%2074%20d4%2018%20b0%2008%2087%2018%20ff%20ff%20ff%2026%2028%20ff%20ff%20ff%20ff%2001%2048%2000%20%5BTemp%2029.5%20C%20Humidity%2052%25%20Wdir%20262%20Wind%200%20UV%200%20Baro%20990%20Rain%200%5D&c=%7B150%7D%207a%2029%2018%20b0%2008%2087%2018%20ff%20ff%20ff%2026%2028%2029%2054%2052%20ff%20f0%2087%20f8&c=%7B150%7D%207a%2005%2018%20b0%2008%2087%2018%20ff%20ff%20ff%2035%2088%20ff%20ff%20ff%20ff%2001%20d9%20f8%20%5BTemp%2029.5%20C%20Humidity%2052%25%20Wdir%20359%20Wind%201.1%20UV%200%20Baro%20990%20Rain%200%5D&c=%7B149%7D%2063%209e%2018%20b0%2008%2087%2018%20fb%20ae%20fe%2035%2098%2029%2054%2052%20ff%20f0%205e%2000&c=%7B149%7D%2034%203d%2018%20b0%2008%2087%2018%20fd%20b6%20ff%2008%2048%20ff%20ff%20ff%20ff%2001%2091%20f8%20%5BTemp%2029.5%20C%20Humidity%2052%25%20Wdir%2084%20Wind%200.9%20UV%200%20Baro%20990%20Rain%200%5D&c=%7B149%7D%2066%20e9%2018%20b0%2008%2087%2018%20fe%20ff%20fe%2008%2048%2029%2054%2053%20ff%20f0%2086%2000&c=%7B150%7D%2022%20d9%2018%20b0%2008%2087%2010%20ff%20ff%20ff%2028%2068%20ff%20ff%20fd%20ab%2001%2064%2000%20%5BTemp%2029.6%20C%20Humidity%2053%25%20Wdir%20269%20Wind%200%20UV%202.1%20Baro%20990%20Rain%200.7%5D&c=%7B149%7D%20da%20bd%2018%20b0%2008%2087%2010%20ff%20ff%20ff%2028%2048%2029%2064%2053%20fd%20e0%206e%2000&c=%7B150%7D%2099%2010%2018%20b0%2008%2087%2010%20ff%20ff%20ff%2026%2068%20ff%20ff%20ed%208f%2001%2092%2000%20%5BTemp%2029.7%20C%20Humidity%2053%25%20Wdir%20269%20Wind%200%20UV%202.1%20Baro%20990%20Rain%201.2%5D&c=%7B150%7D%2043%2053%2018%20b0%2008%2087%2010%20ff%20ff%20ff%2026%2078%2029%2084%2053%20fd%20e0%2020%2000&f=CHK%3F8h8h%20ID%3F8h8h8h8h8h%208h8h8h%20WDIR%3A12h%204h%20TEMP8h.4h%20%3F4h%20HUM8h%20UV%3F~12h%204h%208h8h).
`CHK?8h8h ID?8h8h8h8h8h 8h8h8h WDIR:12h 4h TEMP8h.4h ?4h HUM8h UV?~12h 4h 8h8h`
The barometric pressure is likly not measured remote. Haven't spotted Wind speed or rain yet. Can you perhaps add more codes with interesting readings for wind and rain (paste here or link to a BitBench)?

zuckschwerdt, 2020-09-20:
Rain is likely measured in "bucket tips" and accumulated, we are looking for a field with some value that increases when rain is reported.

zuckschwerdt, 2020-09-22:
Ideally to find the missing values you need to get the sensor indoors then tip the rain bucket by hand and observe which number in the codes increases. Slowly add single bucket tips to confirm. Maybe then pour a measured amount of water in and see how much water is one bucket tip or compare a counted number of tips to the console readout.
For wind speed maybe rotate the sensor a different fixed speeds and observe where there is a matching number in the codes. I expect there to be two fields: gust (maximum) speed and average speed.

RebeckaJohansson, 2020-09-23:
Still trying to learn how to read the code on BitBench. 
But like an earlyer comment will use this for my weewx

RebeckaJohansson, 2020-09-27:
For weewx there is an extension and Bob that decode this has this information in py file there is some codes to read in this .py file 

> bin/user/ws6in1.py

RebeckaJohansson, 2020-09-28:
I have find wind speed and wind gust in the code.
I have dubblecheck it to. 

I have ventus w835 6in1 and is a copy of Bresser 6in1 

In bitbench i used:

SYNC16h CHK?8h8h ID?8h8h8h8h8h SOME VALUE? 8h8h8h WIND DIR 8h4h ?4h TEMP8h.4h ?4h HUM8h RAIN?8h WIND SPEED / GUST 8h 8h 8h

The result

zuckschwerdt, 2020-09-28:
Thanks! I'Ve updated your format string to `SYNC16h DIGEST:8h8h ID?8h8h8h8h FLAGS:4h BATT:1b CH:3d ?8h8h8h WIND DIR 8h.4h ?4h TEMP8h.4h ?4h HUM8h UV/RAIN?8h WIND/GUST 8h 8h TRAILER:8h` ("trailer" is invalid, the 6-in-1 message is 18 bytes long)

Can you see what the last bytes (wind speed/gust) actually mean?

RebeckaJohansson, 2020-09-28:
> Thanks! I'Ve updated your format string to `SYNC16h DIGEST:8h8h ID?8h8h8h8h FLAGS:4h BATT:1b CH:3d WSPEED:~8h~4h ~4h~8h WDIR 8h.4h ?4h TEMP8h.4h ?4h HUM8h UV/RAIN?8h8h CHKSUM:8h TRAILER:8h`.
> "Trailer" is invalid, the 6-in-1 message is 18 bytes long, the last byte is a checksum, the last 16-bit before the checksum is known to be UV (and maybe rain?).
> 
> The wind speed is in byte 7-9 but inverted! So we known most of the fields now :)

I updated for what is correct for my code and i find specifik WSPEED and WGUST and added it i bitbench 

The rain or uv is hard for me i have no uv recorded today because of dark and rainy weather in sweden so my uv is only 0 my rain today is 0,5 in mm as rain rate so i cant se if its uv or rain value because i only have 0 in the code or 0,1

But i added this in bitbench and synced it and for my code this is correct for me

SYNC16h DIGEST:8h8h ID?8h8h8h8h FLAGS:4h BATT:1b CH:3d WSPEED:~8h,~4h / WGUST ~4h,~8h WDIR 8h4h ?4h TEMP8h.4h ?4h HUM8h RAIN/UV? ~8h,8h CHKSUM:8h TRAILER:~8h

zuckschwerdt, 2020-09-28:
I think it's `WGUST:~8h,~4h / WSPEED ~4h,~8h` not `WSPEED:~8h,~4h / WGUST ~4h,~8h`. Otherwise looks good!
You can try #1225 which now includes these fields.

RebeckaJohansson, 2020-09-28:
> I think it's `WGUST:~8h,~4h / WSPEED ~4h,~8h` not `WSPEED:~8h,~4h / WGUST ~4h,~8h`. Otherwise looks good!
> You can try #1225 which now includes these fields.

I tested it but the wind value goes wrong and for me the windspeed calculate gust and wind gust calculate windspeed 🤔

zuckschwerdt, 2020-09-28:
Strange. Gust is the bigger value and Average Speed the smaller one, right? Like in:
```
model     : Bresser-6in1 id        : 411042499
channel   : 0            Battery   : 0             Temperature: 11.8 C       Humidity  : 81            Wind Gust : 4.4 m/s       Wind Speed: 3.5 m/s
Direction : 269          Rain?     : 165           Flags     : 1             Integrity : CRC
```
and
```
model     : Bresser-6in1 id        : 411042499
channel   : 0            Battery   : 0             Wind Gust : 6.5 m/s       Wind Speed: 5.2 m/s       Direction : 264           Rain      : 0.0 mm
UV        : 165.0        Flags     : 1             Integrity : CRC
```

RebeckaJohansson, 2020-09-28:
> Strange. Gust is the bigger value and Average Speed the smaller one, right? Like in:
>
> ```
> model     : Bresser-6in1 id        : 411042499
> channel   : 0            Battery   : 0 [...]

Strange. In bitbench for my codeing for Ventus 6in1 the value and wind check is avg speed first and then gust 🤔 

I can check tomorrow again. 
But before i tested when i touch the wind pointer outside and same time decode it gave me the value like i wrote in bitbench 1,1 wspeed and 9,3 wgust in m/s mabey i need to change something in the bitbench code.

RebeckaJohansson, 2020-09-28:
> Strange. Gust is the bigger value and Average Speed the smaller one, right? Like in:
>
> ```
> model     : Bresser-6in1 id        : 411042499
> channel   : 0            Battery   : 0 [...]

Check my codeing here,  if i change to gust first and then avg speed it gives wrong value for me. 
[image]

zuckschwerdt, 2020-09-28:
I see. the digits in the transmission are in a strange order: if you see `022 900` you need to read that as 02,2 and 00,9 -- that's what the decoder currently does.

RebeckaJohansson, 2020-09-28:
> I see. the digits in the transmission are in a strange order: if you see `022 900` you need to read that as 02,2 and 00,9 -- that's what the decoder currently does.

Ohh i see, I was confused at first but now i understand how i need to read it. So i change it to this now like what you described before with the wind. 

I hope this is right now :)
[image]

zuckschwerdt, 2020-09-29:
The protocol named "6-in-1" (18 bytes data, digest+checksum) is used by multiple sensors. At least:
- new 5-in-1
- 6-in-1
- 7-in-1 indoor
- SM60020 Soil moisture

We need to figure out if there are flags in the data indicating the sensor type.
Can everybody please report each of your exact sensor models with the ID from current #1225 output.
ID should be like ID "1880 02c3 18"
Exact model should include the branded name a product code (some longer number) and what type of sensor it is.

You can also analyze the data yourself, there are two different messages, one with temp/hum and one with rain:
```
bresser_6in1_decode: {144} 83 ae 18 80 02 c3 18 fc 78 fc 29 28 ff ff ff ff 01 cc
bresser_6in1_decode: {144} 5c e4 18 80 02 c3 18 fb ba fc 26 98 11 84 81 ff f0 16
                           ^^^^^digest          ^-.. wind ..-^ ^^^^^^^^ temp/hum or 0xff,rain
                                 ^^^^^^^^^^^^^^ id and flags            ^^^^^ UV or unknown
```

RebeckaJohansson, 2020-09-29:
I also checked wich stations are 6in1 and those who has same outdata:

Ventus w835
My Ventus w835 has data for 
Barometer
Winddir 360°
Windspeed
Windgust
Beaufort
Temp
Hum
Dewpoint
Heatindex
Windchill
UV
Rain with raintipper
Solarpanel/power for battery
And my Ventus transmissions interval:
12 seconds for vind data 
24 seconds for temperatur, hum, UV and rain dat

The other station with or close same output is
Youshiko YC9388
Bresser 6in1
Garni 935PC

RebeckaJohansson, 2020-09-29:
> The protocol named "6-in-1" (18 bytes data, digest+checksum) is used by multiple sensors. At least:
>
> * new 5-in-1
> * 6-in-1
> * 7-in-1 indoor
> * SM60020 Soil moisture
>
> We need to figure out if there are [...]

I still figuring out uv and rain

My uv is on 0 now and rainrate is 0 for specific today. 
My total rain is 5,8 mm and highest rain rait is 4,1 mm/hour

RebeckaJohansson, 2020-09-29:
Hmm i dont know how to find the specifik productcode 

I checked my box that my station come with and it only have VENTUS W835 as name and what the station can do. Cant find any other number more then this number from the brand site

(https://www.nshnordic.com/english/products/produkter/ventus/vaderstationer/ventus-professionell-vaderstation-w835-med-pc-mjukvara?AreaID=51&LanguageID=LANG1)

RebeckaJohansson, 2020-09-29:
Sorry for spamming. Bur i mabey figuring out the last in the code. My rain rate check is 4,1 and the code shows when seperated uv and rain i got 014 in code.

Check my picture. Mabey is right mabey is wrong. 
[image]

zuckschwerdt, 2020-09-29:
It's `UV:8h 8h CHK:8h` -- your code is too long.
Rain is encoded in the temp/hum field.
- with temp/hum `179466` : 17.9 C, 66%
- with temp/hum `ffedcb` : `edcb ^ ffff = 1234` 123.4 mm

RebeckaJohansson, 2020-09-29:
> It's `UV:8h 8h CHK:8h` -- your code is too long.
> Rain is encoded in the temp/hum field.
> 
> * with temp/hum `179466` : 17.9 C, 66%
> * with temp/hum `ffedcb` : `edcb ^ ffff = 1234` 123.4 mm

Thanks, yes UV is 0 here, then i remove rainrate 👍

RebeckaJohansson, 2020-09-29:
> It's `UV:8h 8h CHK:8h` -- your code is too long.
> Rain is encoded in the temp/hum field.
> 
> * with temp/hum `179466` : 17.9 C, 66%
> * with temp/hum `ffedcb` : `edcb ^ ffff = 1234` 123.4 mm

Then it should be like this i think 
[image]

zuckschwerdt, 2020-09-30:
@nkpap @danrhjones can you test the code in #1225 and report your exact device name/product number with the ID+Status from the output?

mvdgrift, 2020-10-01:
Hi @zuckschwerdt,

I've tested the code in #1225 in combination with my Wi-Fi Colour Weather Station with 5in1 Sensor and it seems to work. Please ignore the UV-sensor value (the weather station is not equipped with a UV-sensor).

The output:
```
time      : 2020-10-01 12:51:47
model     : Bresser-6in1 id        : 2030088d
channel   : 0            Battery   : 0             Temperature: 16.2 C       Humidity  : 81            Wind Gust : 0.6 m/s       Wind Speed: 0.6 m/s       Direction : 338           Unknown   : 165           Flags     : 1             Integrity : CRC
MSG: {200} 4f cf 20 30 08 8d 18 ff 77 ff 33 88 ff fa ff ff 01 da 00 00 00 00 00 00 00 
bresser_6in1_decode: {144} 4f cf 20 30 08 8d 18 ff 77 ff 33 88 ff fa ff ff 01 da 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-01 12:51:59
model     : Bresser-6in1 id        : 2030088d
channel   : 0            Battery   : 0             Wind Gust : 0.8 m/s       Wind Speed: 0.8 m/s       Direction : 338           Rain      : 50.0 mm       UV        : 165.0         Flags     : 1             Integrity : CRC
```

mvdgrift, 2020-10-01:
> @mvdgrift what command do you run to get that? every thing I have tried has not returned any values

`rtl_433 -f 868.3M -s 1M -R 119`

RebeckaJohansson, 2020-10-01:
I dont get any values more then callback and sometimes hum and temp for my 6in1 ventus

danrhjones, 2020-10-01:
`_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-01 15:37:49
model     : Bresser-6in1 id        : 19700d57
channel   : 0            Battery   : 0             Wind Gust : 0.3 m/s
Wind Speed: 0.3 m/s      Direction : 180           Rain      : 441.6 mm
UV        : 165.0        Flags     : 1             Integrity : CRC
MSG: {200} 91 7b 19 70 0d 57 18 ff cc ff 18 08 16 22 54 ff f0 95 00 00 00 00 00 00 00
bresser_6in1_decode: {144} 91 7b 19 70 0d 57 18 ff cc ff 18 08 16 22 54 ff f0 95

Please report your exact sensor model and this ID "1970 0d57 18"
at https://github.com/merbanan/rtl_433/pull/1214
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-01 15:38:01
model     : Bresser-6in1 id        : 19700d57
channel   : 0            Battery   : 0             Temperature: 16.2 C       Humidity  : 54            Wind Gust : 0.3 m/s       Wind Speed: 0.3 m/s       Direction : 180
Unknown   : 165          Flags     : 1             Integrity : CRC
MSG: {200} 68 78 19 70 0d 57 18 ff dd ff 00 08 ff bb e9 ff 01 74 00 00 00 00 00 00 00
bresser_6in1_decode: {144} 68 78 19 70 0d 57 18 ff dd ff 00 08 ff bb e9 ff 01 74

Please report your exact sensor model and this ID "1970 0d57 18"
at https://github.com/merbanan/rtl_433/pull/1214
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-10-01 15:38:13
model     : Bresser-6in1 id        : 19700d57
channel   : 0            Battery   : 0             Wind Gust : 0.2 m/s       Wind Speed: 0.2 m/s       Direction : 0             Rain      : 441.6 mm      UV        : 165.0
Flags     : 1            Integrity : CRC`

[40 more replies]

closed 2023-09-26 by gdt