merbanan/rtl_433 #1217: 315 mhz 5 button fan remote

beau76, 2019-12-01:
hi I like to add this remote FCC ID A25-TX005R
[5 button remote.zip](https://github.com/merbanan/rtl_433/files/3908519/5.buttion.remote.zip)

zuckschwerdt, 2019-12-02:
Step 1: view the Spectrogram https:/triq.net/iqs to verify good samples.

[image]

- g004 is empty
- g002 & g007 are something else (TPMS?)

zuckschwerdt, 2019-12-02:
Step 2: use `rtl_433 -A -r YOURFILE.cu8` to get an idea of the timings:
```
Pulse width distribution:
 [ 0] count:   13,  width:  424 us [424;444]	( 106 S)
 [ 1] count:   12,  width:  880 us [880;888]	( 220 S)
Gap width distribution:
 [ 0] count:   12,  width:  892 us [888;896]	( 223 S)
 [ 1] count:   12,  width:  444 us [440;448]	( 111 S)
Pulse period distribution:
 [ 0] count:   15,  width: 1320 us [1316;1336]	( 330 S)
 [ 1] count:    5,  width:  868 us [868;876]	( 217 S)
 [ 2] count:    4,  width: 1776 us [1776;1780]	( 444 S)
```
The suggestion of a demodulation could be wrong, is it really manchester?
```
Guessing modulation: Manchester coding
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=424,l=0,r=900'
```
Verify with `sigrok-open.sh YOURFILE.cu8`
[image]
Does not look like Manchester. Just plain long gap+short pulse and short gap+long pulse.

zuckschwerdt, 2019-12-02:
Step 3: Use the gathered information about modulation and timings to answer the following:
- What does a "short" symbol look like?
- What does a "long symbol look like?
- Is there a "sync" symbol, what does it look like?
- Are there multiple repeats in one transmission? What is the maximum gap within one repeat?
- What is the maximum gap in the transmission (the "reset" value)?

Now try to construct a simple flex decoder.

beau76, 2019-12-02:
can I do all of this on a windows pc?

zuckschwerdt, 2019-12-02:
Sigrok Pulseview is available for Windows: https://sigrok.org/wiki/Downloads

Not sure if you can find a way to run the converter script `sigrok-open.sh`.
Perhaps some other Windows user can help out here?

merbanan, 2020-01-12:
No answer for a while. Closing issue.

closed 2020-01-12 by merbanan