merbanan/rtl_433 #1218: Receiving the same sensor under 433.92 and 868.3 MHz? Cannot receive Bresser signal

giddyhup, 2019-12-02:
I am trying to receive the signal from a Bresser-5-in-1 weather center. It used to work until a week ago. When it stopped working I replaced the batteries and updated rtl_433. Now I can still receive the data from 10 meters away under Windows with an older rtl_433 version but not with 19.08-104-g3d4aff0 under Linux (2 meters away).

Strangely though, I can receive a 433 MHz sensor also under 868 MHz:
```
$ rtl_433 -f 868300000 -f 433920000 -H 40                                                                                master
rtl_433 version 19.08-104-g3d4aff0 branch master at 201911301241 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

        Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
        A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 112 out of 141 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-104 108-116 119 121 124-128 131-141 ]
rtl_433: warning: 104 "Wireless M-Bus, Mode C&T, 100kbps (-f 868950000 -s 1200000)" does not support CSV output
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-02 15:47:25
model     : Prologue sensor                        id        : 5
rid       : 117          Channel   : 3             Battery   : OK            Temperature: 2.90 C       Button    : 0             Humidity  : 13 %

Tuned to 433.920MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-02 15:48:00
model     : Prologue sensor                        id        : 5
rid       : 117          Channel   : 3             Battery   : OK            Temperature: 2.90 C       Button    : 0             Humidity  : 13 %
^CSignal caught, exiting!
```

What is going on?

merbanan, 2019-12-10:
Try with -Y classic.

giddyhup, 2019-12-10:
@merbanan Thanks. That works and gives the expected results.

closed 2019-12-17 by giddyhup