merbanan/rtl_433 #1220: Ambient Weather being decoded as 2 devices

slvrscoobie, 2019-12-02:
Hi! 
Im having trouble with my weewx, and found that it might be partly due to the parsing not working correctly. I have a WS2902a Array that for almost a year has been decoded as a WH65Packet. Today I changed the batteries in the array as I was have trouble connecting to it (but only it) and in watching the tail log I saw this :

slvrscoobie, 2019-12-02:
parsed: {'humidity.10.FOWH24Packet': 97.0, 'uv_index.10.FOWH24Packet': 0.0, 'light.10.FOWH24Packet': 2418.0, 'dateTime': 1575317711, 'rain_total.10.FOWH24Packet': 1790.1, 'temperature.10.FOWH24Packet': 0.5, 'wind_gust.10.FOWH24Packet': 1.12, 'battery.10.FOWH24Packet': 0, 'wind_dir.10.FOWH24Packet': 293.0, 'usUnits': 17, 'wind_speed.10.FOWH24Packet': 0.7}
out:[u'{"time" : "2019-12-02 20:15:25", "model" : "AmbientWeather-WH31E", "id" : 161, "channel" : 3, "battery" : "OK", "temperature_C" : 5.200, "humidity" : 57, "data" : "7f00000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:15:27", "model" : "Fine Offset WH24", "id" : 10, "battery" : "OK", "temperature_C" : 0.500, "humidity" : 97, "wind_dir_deg" : 342, "wind_speed_ms" : 0.420, "gust_speed_ms" : 1.120, "rainfall_mm" : 1790.100, "uv" : 22, "uvi" : 0, "light_lux" : 2464.000, "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:15:30", "model" : "AmbientWeather-WH31E", "id" : 211, "channel" : 1, "battery" : "OK", "temperature_C" : 1.000, "humidity" : 95, "data" : "ce00000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:15:30", "model" : "AmbientWeather-WH31E", "id" : 211, "channel" : 1, "battery" : "OK", "temperature_C" : 1.000, "humidity" : 95, "data" : "ce00000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n']
parsed: {'channel.161.AmbientWH31EPacket': 3, 'snr.161.AmbientWH31EPacket': 0.0, 'battery.161.AmbientWH31EPacket': 0, 'dateTime': 1575317725, 'temperature.161.AmbientWH31EPacket': 5.2, 'humidity.161.AmbientWH31EPacket': 57.0, 'noise.161.AmbientWH31EPacket': 0.0, 'rssi.161.AmbientWH31EPacket': 0, 'usUnits': 17}
parsed: {'humidity.10.FOWH24Packet': 97.0, 'uv_index.10.FOWH24Packet': 0.0, 'light.10.FOWH24Packet': 2464.0, 'dateTime': 1575317727, 'rain_total.10.FOWH24Packet': 1790.1, 'temperature.10.FOWH24Packet': 0.5, 'wind_gust.10.FOWH24Packet': 1.12, 'battery.10.FOWH24Packet': 0, 'wind_dir.10.FOWH24Packet': 342.0, 'usUnits': 17, 'wind_speed.10.FOWH24Packet': 0.42}
parsed: {'channel.211.AmbientWH31EPacket': 1, 'snr.211.AmbientWH31EPacket': 0.0, 'noise.211.AmbientWH31EPacket': 0.0, 'dateTime': 1575317730, 'rssi.211.AmbientWH31EPacket': 0, 'battery.211.AmbientWH31EPacket': 0, 'temperature.211.AmbientWH31EPacket': 1.0, 'humidity.211.AmbientWH31EPacket': 95.0, 'usUnits': 17}
parsed: {'channel.211.AmbientWH31EPacket': 1, 'snr.211.AmbientWH31EPacket': 0.0, 'noise.211.AmbientWH31EPacket': 0.0, 'dateTime': 1575317730, 'rssi.211.AmbientWH31EPacket': 0, 'battery.211.AmbientWH31EPacket': 0, 'temperature.211.AmbientWH31EPacket': 1.0, 'humidity.211.AmbientWH31EPacket': 95.0, 'usUnits': 17}
out:[u'{"time" : "2019-12-02 20:15:43", "model" : "Fine Offset WH65B", "id" : 10, "battery" : "OK", "temperature_C" : 0.500, "humidity" : 97, "wind_dir_deg" : 16, "wind_speed_ms" : 0.446, "gust_speed_ms" : 1.020, "rainfall_mm" : 1515.618, "uv" : 21, "uvi" : 0, "light_lux" : 2418.000, "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n']
parsed: {'rain_total.10.FOWH65BPacket': 1515.618, 'uv.10.FOWH65BPacket': 21.0, 'wind_dir.10.FOWH65BPacket': 16.0, 'uv_index.10.FOWH65BPacket': 0.0, 'light.10.FOWH65BPacket': 2418.0, 'battery.10.FOWH65BPacket': 0, 'wind_gust.10.FOWH65BPacket': 1.02, 'temperature.10.FOWH65BPacket': 0.5, 'dateTime': 1575317743, 'humidity.10.FOWH65BPacket': 97.0, 'usUnits': 17, 'wind_speed.10.FOWH65BPacket': 0.446}
out:[u'{"time" : "2019-12-02 20:15:57", "model" : "AmbientWeather-WH31E", "id" : 160, "channel" : 7, "battery" : "OK", "temperature_C" : 20.200, "humidity" : 45, "data" : "4d00000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:15:57", "model" : "AmbientWeather-WH31E", "id" : 160, "channel" : 7, "battery" : "OK", "temperature_C" : 20.200, "humidity" : 45, "data" : "4d00000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:15:59", "model" : "Fine Offset WH24", "id" : 10, "battery" : "OK", "temperature_C" : 0.500, "humidity" : 97, "wind_dir_deg" : 76, "wind_speed_ms" : 1.820, "gust_speed_ms" : 2.240, "rainfall_mm" : 1790.100, "uv" : 21, "uvi" : 0, "light_lux" : 2418.000, "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:16:00", "model" : "AmbientWeather-WH31E", "id" : 116, "channel" : 2, "battery" : "OK", "temperature_C" : 5.300, "humidity" : 80, "data" : "0000000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n', u'{"time" : "2019-12-02 20:16:00", "model" : "AmbientWeather-WH31E", "id" : 116, "channel" : 2, "battery" : "OK", "temperature_C" : 5.300, "humidity" : 80, "data" : "0000000000", "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n']
parsed: {'battery.160.AmbientWH31EPacket': 0, 'rssi.160.AmbientWH31EPacket': 0, 'humidity.160.AmbientWH31EPacket': 45.0, 'temperature.160.AmbientWH31EPacket': 20.2, 'channel.160.AmbientWH31EPacket': 7, 'snr.160.AmbientWH31EPacket': 0.0, 'noise.160.AmbientWH31EPacket': 0.0, 'dateTime': 1575317757, 'usUnits': 17}
[... cut ...] "uv" : 21, "uvi" : 0, "light_lux" : 2464.000, "mic" : "CRC", "mod" : "ASK", "freq" : 0.000, "rssi" : 0.000, "snr" : 0.000, "noise" : 0.000}\n']
parsed: {'humidity.10.FOWH24Packet': 97.0, 'uv_index.10.FOWH24Packet': 0.0, 'light.10.FOWH24Packet': 2464.0, 'dateTime': 1575317775, 'rain_total.10.FOWH24Packet': 1790.1, 'temperature.10.FOWH24Packet': 0.5, 'wind_gust.10.FOWH24Packet': 2.24, 'battery.10.FOWH24Packet': 0, 'wind_dir.10.FOWH24Packet': 48.0, 'usUnits': 17, 'wind_speed.10.FOWH24Packet': 1.54}

zuckschwerdt, 2019-12-03:
Please capture a few samples (`rtl_433 -S all`) and upload here as zip.

slvrscoobie, 2019-12-03:
[g014_914975M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/3918542/g014_914975M_1000k.cu8.zip)
I saved a couple. I saw it decode as both WH65 and WH24 a few times before I cancelled. 
Thanks!!!!

slvrscoobie, 2019-12-12:
Added some more files with json data here so you know what you're looking at (if you didnt already) 

sudo rtl_433 -f 914979000 -S all
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 112 out of 141 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-104 108-116 119 121 124-128 131-141 ]
rtl_433: warning: 104 "Wireless M-Bus, Mode C&T, 100kbps (-f 868950000 -s 1200000)" does not support CSV output
usb_claim_interface error -6
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 914.979MHz.
*** Saving signal to file g001_914.979M_1000k.cu8 (86290 samples, 262144 bytes)
*** Saving signal to file g002_914.979M_1000k.cu8 (86287 samples, 262144 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-12 16:38:31
model     : Fine Offset WH65B                      ID        : 10
Battery   : OK           Temperature: -1.0 C       Humidity  : 59 %          Wind direction: 287       Wind speed: 0.0 m/s       Gust speed: 0.0 m/s       Rainfall  : 1572.3 mm     UV        : 3
UVI       : 0            Light     : 0.0 lux       Integrity : CRC
*** Saving signal to file g003_914.979M_1000k.cu8 (44844 samples, 131072 bytes)
*** Saving signal to file g004_914.979M_1000k.cu8 (86280 samples, 262144 bytes)
*** Saving signal to file g005_914.979M_1000k.cu8 (86336 samples, 262144 bytes)
*** Saving signal to file g006_914.979M_1000k.cu8 (61316 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-12 16:38:47
model     : Fine Offset WH24                       ID        : 10
Battery   : OK           Temperature: -1.0 C       Humidity  : 58 %          Wind direction: 287       Wind speed: 0.0 m/s       Gust speed: 0.0 m/s       Rainfall  : 1857.0 mm     UV        : 2
UVI       : 0            Light     : 0.0 lux       Integrity : CRC
*** Saving signal to file g007_914.979M_1000k.cu8 (44833 samples, 131072 bytes)
*** Saving signal to file g008_914.979M_1000k.cu8 (48631 samples, 131072 bytes)
*** Saving signal to file g009_914.979M_1000k.cu8 (49786 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-12 16:38:52
model     : AmbientWeather-WH31E                   id        : 94
Channel   : 8            Battery   : OK            Temperature: -17.1 C      Humidity  : 62 %          Extra Data: 0900000000    Integrity : CRC
*** Saving signal to file g010_914.979M_1000k.cu8 (208385 samples, 524288 bytes)
*** Saving signal to file g011_914.979M_1000k.cu8 (78740 samples, 262144 bytes)
*** Saving signal to file g012_914.979M_1000k.cu8 (35298 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-12 16:39:03
model     : Fine Offset WH24                       ID        : 10
Battery   : OK           Temperature: -1.0 C       Humidity  : 58 %          Wind direction: 287       Wind speed: 0.0 m/s       Gust speed: 0.0 m/s       Rainfall  : 1857.0 mm     UV        : 3
UVI       : 0            Light     : 0.0 lux       Integrity : CRC
*** Saving signal to file g013_914.979M_1000k.cu8 (44822 samples, 131072 bytes)
*** Saving signal to file g014_914.979M_1000k.cu8 (86351 samples, 262144 bytes)
*** Saving signal to file g015_914.979M_1000k.cu8 (49814 samples, 131072 bytes)
*** Saving signal to file g016_914.979M_1000k.cu8 (44316 samples, 131072 bytes)
*** Saving signal to file g017_914.979M_1000k.cu8 (44841 samples, 131072 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-12 16:39:28
model     : AmbientWeather-WH31E                   id        : 160
Channel   : 7            Battery   : OK            Temperature: 20.3 C       Humidity  : 55 %          Extra Data: bc00000000    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-12-12 16:39:28
model     : AmbientWeather-WH31E                   id        : 160
[... cut ...]
Battery   : OK           Temperature: -1.1 C       Humidity  : 61 %          Wind direction: 287       Wind speed: 0.0 m/s       Gust speed: 0.0 m/s       Rainfall  : 1857.0 mm     UV        : 3
UVI       : 0            Light     : 0.0 lux       Integrity : CRC
*** Saving signal to file g033_914.979M_1000k.cu8 (44833 samples, 131072 bytes)
*** Saving signal to file g034_914.979M_1000k.cu8 (86353 samples, 262144 bytes)
[g001_914.979M_1000k_12_12_19.zip](https://github.com/merbanan/rtl_433/files/3957946/g001_914.979M_1000k_12_12_19.zip)

ryan-kramer, 2020-02-02:
Just did a side-by-side with 19.08-157 and 18.12-207   Very obvious difference, it seems like almost every other packet is decoded as WH24.  (19 also seems inconsistent with decoding it in general, it will sometimes miss the packet entirely while 18 picks it up.

(output from two different pi's running concurrently)

root@weatherpi2:~#  rtl_433 -M utc -F json -R 78 -f 915m -d0
rtl_433 version 19.08-157-g4cab9ff branch master at 202001271734 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R820T tuner

{"time" : "2020-02-02 00:58:46", "model" : "Fine Offset WH24", "id" : 128, "battery" : "OK", "temperature_C" : 8.800, "humidity" : 53, "wind_dir_deg" : 281, "wind_speed_ms" : 13.860, "gust_speed_ms" : 19.040, "rainfall_mm" : 846.900, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}
{"time" : "2020-02-02 00:59:02", "model" : "Fine Offset WH65B", "id" : 128, "battery" : "OK", "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 258, "wind_speed_ms" : 7.395, "gust_speed_ms" : 8.670, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}
{"time" : "2020-02-02 00:59:34", "model" : "Fine Offset WH24", "id" : 128, "battery" : "OK", "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 275, "wind_speed_ms" : 15.540, "gust_speed_ms" : 19.040, "rainfall_mm" : 846.900, "uv" : 1, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC"}
{"time" : "2020-02-02 00:59:50", "model" : "Fine Offset WH65B", "id" : 128, "battery" : "OK", "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 269, "wind_speed_ms" : 7.459, "gust_speed_ms" : 8.670, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "mic" : "CRC

------------------------------------------------------------

root@weatherpi:~#  rtl_433 -M utc -F json -R 78 -f 915m -d0
rtl_433 version 18.12-207-g82d0f55 branch master at 201905061642 inputs file rtl_tcp RTL-SDR
Found Rafael Micro R820T tuner

{"time" : "2020-02-02 00:58:46", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.800, "humidity" : 53, "wind_dir_deg" : 281, "wind_speed_ms" : 6.311, "gust_speed_ms" : 8.670, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 00:59:02", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 258, "wind_speed_ms" : 7.395, "gust_speed_ms" : 8.670, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 00:59:18", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 273, "wind_speed_ms" : 5.865, "gust_speed_ms" : 8.160, "rainfall_mm" : 717.042, "uv" : 1, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 00:59:34", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 275, "wind_speed_ms" : 7.076, "gust_speed_ms" : 8.670, "rainfall_mm" : 717.042, "uv" : 1, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 00:59:50", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.800, "humidity" : 52, "wind_dir_deg" : 269, "wind_speed_ms" : 7.459, "gust_speed_ms" : 8.670, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}

ryan-kramer, 2020-02-02:
one more quick test, built 18.12-207 on weatherpi2 (the pi that had 19 above), just to eliminate any hardware differences.   Works as expected, picks up 100% of packets and decodes as WH65B

root@weatherpi2:~#  rtl_433 -M utc -F json -R 78 -f 915m -d0
rtl_433 version 18.12-207-g82d0f55 branch master at 201905061642 inputs file rtl_tcp RTL-SDR
{"time" : "2020-02-02 01:08:38", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.200, "humidity" : 57, "wind_dir_deg" : 275, "wind_speed_ms" : 7.012, "gust_speed_ms" : 7.650, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 01:08:54", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 8.000, "humidity" : 58, "wind_dir_deg" : 274, "wind_speed_ms" : 7.076, "gust_speed_ms" : 7.650, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 01:09:10", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 7.900, "humidity" : 58, "wind_dir_deg" : 277, "wind_speed_ms" : 7.076, "gust_speed_ms" : 7.650, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 01:09:26", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 7.800, "humidity" : 57, "wind_dir_deg" : 282, "wind_speed_ms" : 6.885, "gust_speed_ms" : 7.650, "rainfall_mm" : 717.042, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}
{"time" : "2020-02-02 01:09:42", "model" : "Fine Offset WH65B", "id" : 128, "temperature_C" : 7.700, "humidity" : 57, "wind_dir_deg" : 257, "wind_speed_ms" : 6.375, "gust_speed_ms" : 7.140, "rainfall_mm" : 717.042, "uv" : 1, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC"}

zuckschwerdt, 2020-02-02:
Did you try `-Y classic`?

ryan-kramer, 2020-02-02:
Just tried classic, it no longer shows WH24 model, but it still drops packets.  3 captures, all 3 missed at least one if not multiple packets.  Maybe just luck of the draw, but it is fairly consistent at dropping the second packet.  not every time but most.     (the pi is about 10 feet away from the station, and v18 never drops an 16 second interval)

root@weatherpi2:~/rtl_433/build/src# ./rtl_433 -Y classic -M utc -F json -R 78 -f 915m -d0
rtl_433 version 19.08-184-g401008d branch master at 202002021734 inputs file rtl_tcp RTL-SDR
{"time" : "2020-02-02 17:14:15",
{"time" : "2020-02-02 17:14:31",
<MISSING DATA HERE> 
{"time" : "2020-02-02 17:15:03",
{"time" : "2020-02-02 17:15:19",
<MISSING DATA HERE> 
<MISSING DATA HERE> 
{"time" : "2020-02-02 17:16:07",

root@weatherpi2:~/rtl_433/build/src# ./rtl_433 -Y classic -M utc -F json -f 915m -d0
rtl_433 version 19.08-184-g401008d branch master at 202002021734 inputs file rtl_tcp RTL-SDR
{"time" : "2020-02-02 17:16:23",
<MISSING DATA HERE> 
{"time" : "2020-02-02 17:16:55",
{"time" : "2020-02-02 17:17:11",
{"time" : "2020-02-02 17:17:27", 
{"time" : "2020-02-02 17:17:43", 
{"time" : "2020-02-02 17:17:59", 

root@weatherpi2:~/rtl_433/build/src# ./rtl_433 -Y classic -M utc -F json -f 915m -d0
rtl_433 version 19.08-184-g401008d branch master at 202002021734 inputs file rtl_tcp RTL-SDR
{"time" : "2020-02-02 17:18:15",
<MISSING DATA HERE> 
{"time" : "2020-02-02 17:18:47",
{"time" : "2020-02-02 17:19:03",
{"time" : "2020-02-02 17:19:19", "

ryan-kramer, 2020-02-02:
Just for the fun of it, forced the detection code to default to WH65B and it works consistently without dropping.  So something isn't quite happy in that.

   // Classification heuristics
    if (bitbuffer->bits_per_row[0] - bit_offset - sizeof(b) * 8 < 8)
        if (bit_offset < 61)
            model = MODEL_WH65B; // nominal 3 bits postamble
        else
            model = MODEL_WH65B;

zuckschwerdt, 2020-02-02:
Good find. If you happen to record samples where the classification would fail we can maybe look into improving it. Otherwise we could perhaps offer an override. Best case though would be an improved understanding what fails or could fail and then make it more robust overall.

slvrscoobie, 2020-09-01:
So what should I do - you're saying built 18.12-207 works perfectly? how would I get that build if I only have the WH65

zuckschwerdt, 2020-09-01:
That's a custom change (made locally and then rebuild the program).
We still don't know why the WH65 sometimes looks like a WH24. Usually the WH65B sends the same data with a slightly longer preamble and postamble. That might be missing or not picked up in some transmissions.

slvrscoobie, 2020-09-01:
do my recordings above offer any insights? how could I build the other version, or set in the override?
Im also having to use frequencies far 'off' from what I would expect, such as 914.69mhz in stead of 915m...

zuckschwerdt, 2020-09-01:
In your last zip of samples the `g033_914.979M_1000k.cu8` decodes as WH24 (using `-Y minmax`, the default for `-f 914.979M`) but when using `-Y classic` it shows as WH65B as expected.
I guess using `-Y classic` will "just work", but we might need to explore why the demod from minmax is different.

slvrscoobie, 2020-09-01:
Ill try it once I get my signal back, it worked a bit this AM but now Im having trouble with it finding the WH65 at all

zuckschwerdt, 2020-09-01:
Can you try CubicSDR, Gqrx, or SDRangel to visualize the live spectrum and find the signal there?

slvrscoobie, 2020-09-01:
ive tried but Im not exactly sure how to use them, in the sense that I see blips, but when I center the freq on them it does no better at picking up

zuckschwerdt, 2020-09-01:
exactly centering isn't the best option, you likely only want to get near the signal, say 50k off.

wauswaus, 2021-03-04:
hey all
I'm also a new owner of a weather station and also experience that the W65B is on average detected as being a WH24.

I'm running the RTL_433 in Home Assistant (using https://github.com/pbkhrv/rtl_433-hass-addons/tree/main/rtl_433) and I am quite new in using the whole rtl library.
What is now the easiest way to get this fixed?

merbanan, 2021-03-07:
@wauswaus use -S known and get some signals that are decoded correctly and some that are decoded incorrectly. That way we can debug the issue in a proper way.

merbanan, 2021-03-07:
@zuckschwerdt I think we need a command line option like the no-wh5. Does that sound like a good solution. The heuristics can only go so far. We should remove it and then add the command line option.

wauswaus, 2021-03-11:
Well I tuned my frequency a bit and tried again the classic pulse_detect setting and she purrs like a kitten :)
I have made logs with the pulse_detect set to auto and there all the received messages where decoded as coming from a WH64 while it is a WH65B. I also made logs with the pulse detect set to classic and there it all goes fine.

If you need me to provide more logs, please let me know.

[good logs with classic pulse detect.zip](https://github.com/merbanan/rtl_433/files/6124869/good.logs.with.classic.pulse.detect.zip)
[WH65 decoded as WH24 using auto pulse detect.zip](https://github.com/merbanan/rtl_433/files/6124870/WH65.decoded.as.WH24.using.auto.pulse.detect.zip)

slvrscoobie, 2021-03-11:
How do you do the classic vs auto pulse detect?

wauswaus, 2021-03-11:
I have a .conf file. In this file there is the following:
```
# as command line option:
#   [-Y auto | classic | minmax] FSK pulse detector mode.
pulse_detect auto
```
There you can set the values "classic" or "auto" (no clue what minmax does).
Via command line you would use the -Y option.

wauswaus, 2021-11-08:
It appears that my config wont run in classic mode anymore. Running the latest version (rtl_433 version 21.05).
Config file has:
sample_rate   1024k
pulse_detect classic

in the log it says "New defaults active, use "-Y classic -s 250k" for the old defaults!"
when I use 
sample_rate   250k
it still comes up with "New defaults active, use "-Y classic -s 250k" for the old defaults!"

Result is that it reduces the number of good decodes considerably .

What todo?

zuckschwerdt, 2021-11-08:
"New defaults" is triggered when you set a frequency (`-f`)  above 800M. Order is important, you can return to `-Y classic` if you have that option later (in the conf or on the command line). Ignore the message then.

gdt, 2023-09-25:
@slvrscoobie Where are we on this issue?  Is it still valid with today's git master?

slvrscoobie, 2023-09-25:
Ive left it on -Classic for a while and do have random decodes as the other device, but on -classic its been better, probably about 75-85% decoded as the correct device. 
If you're willing to investigate, I can update and send more data

gdt, 2023-09-26:
TBH I am ticket gardening; I don't remotely have time to dig into your issue.  A report from fresh code is useful; I don't expect anyone to want to debug about old code.   And, the new code might fix your issue - I have no idea, but there's a lot of progress over time.  Better yet is if you could dig in, fix and send a PR.

zuckschwerdt, 2024-05-25:
This is strange, both protocols have the same timing and message checksum, but the WH24 has a first byte (model code) of `24` and none of the Ambient Weather type share that code. No Ambient Weather message should start with `24` and the decoders check that.

Do you see both decodes at the same time? Or do they alternate in some way? Are you sure this is not some neighbours device?

Is there a chance to record a sample file?

zuckschwerdt, 2024-05-25:
For a log: run the decoders verbose to also get the raw data they receive with `-vv`